esp-arduino-libs/ESP32_Display_Panel #11: Use 24-bit LCD panel

DrTron, 2024-01-14:
My panel has an integrated ESP32-S3 (ESP32-8048S050, ST7262 driver chip) and as far as I can make out from the documentation, addresses the panel in 24-bit mode. At least 24 GPIO's are mentioned in the documentation, R0-7, G0-7 and B0-7.
Can I use this panel/EPS combination with the ESP32_Display_Panel library? I've seen that the definitions for the GPIO's both in the config file as in ESP_Panel.cpp only list 16 GPIO's.

Thanks!

Lzw655, 2024-01-30:
Hi @DrTron, sorry for the late. 

Fundamentally, the ESP32-S3 SoC can only support 16-bit RGB565 and 8-bit RGB888. So, the ESP32_Display_Panel library is the same. 

If you want to drive a 24-bit RGB LCD, hardware processing is required. In simple terms, this involves discarding the low bits of each element in RGB, such as R [0:2], G [0:1], and B [0:2]. Please refer to the [documentation ](https://docs.espressif.com/projects/esp-iot-solution/en/latest/display/lcd/rgb_lcd.html#color-formats) for more details.

DrTron, 2024-02-01:
Thanks, I'll have a look!