esp-arduino-libs/ESP32_Display_Panel #12: Copying the correct files across to Square Line Porting

Gert69, 2024-02-04:
When I copy my UI exported file from Square Line across to the Square Line porting files of ESP_Panel_Display and replace the UI files it gives me a UI.H error no such files in directory but when I use Square line expoted files it compiles correctly but I need to use the ESP panel libs because of my s3 development board .https://www.waveshare.com/esp32-s3-touch-lcd-4.3.htm
[image]

Can someone please explain to me what I am doing wrong?

LHYhorion, 2024-02-04:
> When I copy my UI exported file from Square Line across to the Square Line porting files of ESP_Panel_Display and replace the UI files it gives me a UI.H error no such files in directory but when I use Square line expoted files it compiles correctly but I need to use the ESP panel libs because of my s3 development board .https://www.waveshare.com/esp32-s3-touch-lcd-4.3.htm ![UI H error](https://private-user-images.githubusercontent.com/64138360/302075205-59f93d27-6505-46e5-8a34-23eab777749e.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDcwMjI0NjAsIm5iZiI6MTcwNzAyMjE2MCwicGF0aCI6Ii82NDEzODM2MC8zMDIwNzUyMDUtNTlmOTNkMjctNjUwNS00NmU1LThhMzQtMjNlYWI3Nzc3NDllLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjA0VDA0NDkyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ4NGQ0MmE4ZjAyZDU2MTIzYWIzNjM4YjcxNWExMjQxNThkZTdjYmJlZTk5OWVhODQ3OGRmZDdlMzRkNGU4YTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.rfo4_c8f1sha9GBZ_ayYYILTG-xIEkoEsPbadSP7lEI)
> 
> Can someone please explain to me what I am doing wrong?

Hi, thank you for your question. The reason you are encountering a compilation error is likely due to the incorrect placement of the 'ui' folder. You need to place the 'ui' folder at the same level as the 'ESP32_Display_Panel' directory. Alternatively, you can refer to https://github.com/esp-arduino-libs/ESP32_Display_Panel#port-the-squareline-project, which provides detailed instructions on how to port the Squareline project and the correct directory placement. If you have any further questions, feel free to ask.

Gert69, 2024-02-04:
Thank you for your response I am truly grateful for the assistance and that you have taken the time to read my post this has solved all my issues and I am able to move forward with my project.

LHYhorion, 2024-02-04:
You're welcome. Feel free to ask if you have any further questions.

closed 2024-02-04 by LHYhorion

Gert69, 2024-02-04:
Thanks a million Just battling with the button I add in square Line after adding buttons and assigning event to click to go to the next screen it works but as soon as you reset the device and try to click to go to the next screen it doesn't work.

Lzw655, 2024-02-05:
Hi @Gert69, did you reset the device by hardware (set EN pin to low)? Could you please provide a detailed description of the issue? For example, is the touch behaving abnormally, causing the buttons to be unresponsive?

Gert69, 2024-02-05:
Hi @Lzw655 I reloaded the the program this morning on the UART port and it works fine when i plug it out and repower it the screen it doesn't work at all after pushing the reset button a few times and then powering the USB that solves the issue after that you can plug it in and out as you please no issues at all. Thanks for your kind response below I post a link to the Development board.[https://www.waveshare.com/esp32-s3-touch-lcd-4.3.htm](url)

Gert69, 2024-02-05:
@Lzw655 It's really a very cool board absolutely love it I want to use expansion board for my relays and Temperature Sensors to create a mobile one small unit for easy plug and play so it can measure temps and has the ability to switch on and off if you design a nice case for it it will look great when it's installed.

Greatly appreciate the replies and advice you guys are the best @Lzw655 @LHYhorion

Lzw655, 2024-02-05:
@Lzw655 Sure, and I'm restructuring this library now. The new library will support more interfaces of LCDs (like QSPI). For RGB LCD, it can also support the anti-tearing function when using LVGL.

Afterward, since it is not easy for me to purchase this development board. Could I trouble you to contribute your adaptation code to this repository?

Gert69, 2024-02-05:
@Lzw655 Sure I will be happy to contribute the useful code I get and write to the library from my side.

Gert69, 2024-02-05:
@Lzw655  Do you have any idea what causes the screen tearing when using the development boards with LGVL, as everything I load has screen tearing I tweak all the settings in the files as required still standing Images do very well as soon as you insert moving objects and background Images the issue starts I was told by the development board manufacturers that it's considered to be firmware problem/issue

Lzw655, 2024-02-06:
@Gert69 Is it that the screen is drifting and cannot be restored to normal? Can you take a photo?

Gert69, 2024-02-06:
https://github.com/esp-arduino-libs/ESP32_Display_Panel/assets/64138360/37da1b5b-af9b-40ea-b617-46ee91f25372

Gert69, 2024-02-06:
@Lzw655 That the best I could get the video so you can make out that is having that tearing affect when going to the next page if I remove the background images it's okay then the screen tearing is much less.

Lzw655, 2024-02-06:
@Gert69 I think the reason is the insufficient bandwidth of the PSRAM. If you use a big background image, it takes a lot of bandwidth of the PSRAM, which will cause the RGB unable to output data promptly. You can check this problem in the [document](https://docs.espressif.com/projects/esp-faq/en/latest/software-framework/peripherals/lcd.html#why-do-i-get-drift-overall-drift-of-the-display-when-esp32-s3-is-driving-an-rgb-lcd-screen).

I'm updating the library to solve this problem. For now, I think you can try to decrease the PCLK frequency, like 10MHz.

Gert69, 2024-02-07:
@Lzw655  Thanks a million man I tried finding these settings since yesterday but battling where to find the setting to change it thaanks a million for the info @Lzw655

Lzw655, 2024-02-18:
@Gert69 Sorry for the late. The setting is [ESP_PANEL_LCD_RGB_CLK_HZ](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/2622eeceaa3d6ab7e05bacfe5698be82009aea1c/ESP_Panel_Conf_Template.h#L93) in your `ESP_Panel_Conf.h` file.

Gert69, 2024-02-18:
@Lzw655 No stress at all I did try this a while ago if I take it to 10- MHz the screen seems to load the RGB colors White, Red, Blue, Green.

Lzw655, 2024-02-18:
Yeah, the screen may display abnormally if the PCLK is too low...

Gert69, 2024-02-20:
@Lzw655  I got it going since im not using any large images and only displaying static values most of the screen tearing is all out I will assist in getting the lib updated as soon as the project is out of all the libraries available this library is the best and seems to integrate very well with this display if we sort out the screen tearing it's 100%.

Concentration on back light and sleep mode to save on the backup battery if screen is not used.

Once again I humbelly thank you for all the advise and assistance and for the awesome library you have put together.

Lzw655, 2024-02-21:
@Gert69 Thank you very much for your high praise. 😆 I will strive to update this library as soon as possible, and I anticipate completing it within this month. 

BTW, just to mention, the updated library will require dependency on Arduino-ESP32 version 3.0.0-alpha3.

Gert69, 2024-02-24:
It's my pleasure @Lzw655 just a question would one be able to integrate a SD card library for future projects that will take up somewhat space and move for example the UI files generated to the SD card and have your code read it from the SD Card?

Lzw655, 2024-02-25:
@Gert69 Okay, we will consider adding an example that uses an SD card to store images. However, compared to directly using bitmaps, the slower read/write speed of the file system may result in a lower FPS.

Gert69, 2024-02-25:
@Lzw655 I have taken the time to cod an example if you agree I can add it in the repository thanks for considering it, I have loaded my UI but I am limited on the space so when I add the WIFI Bluetooth and SD card functions to my code Sketch becomes to Large.

Gert69, 2024-02-25:
@Lzw655 I will add the SD Test file under .ino whoever uses it should change the path in the code to load images etc.

The test file will serve as a testing file to make sure the card mounts and is functional reads & writes files etc.

Lzw655, 2024-02-25:
@ReacherS3 Great, thank you! I will notify you when the library is updated, and then you can make a PR for it.

ReacherS3, 2024-02-26:
@Lzw655 Got it thanks I did add a test file on my side in the meantime.

ReacherS3, 2024-03-04:
@Lzw655  I trust you are doing well.

would you happen to know how to setup the Backlight function for these displays?

I have changed the 0 to 1 that needed to be changed but I am having difficulties with the code structure in the front end INO. file. 

`/*-------------------------------- Backlight Related --------------------------------*/
#define ESP_PANEL_USE_BL                    (1)
#if ESP_PANEL_USE_BL
/* IO num of backlight pin */
#define ESP_PANEL_LCD_IO_BL                 (45)

/* If the backlight is on when high level, set to 1; otherwise to 0 */
#define ESP_PANEL_LCD_BL_ON_LEVEL           (1)

/* Set to 1 if use PWM for backlight brightness control. */
#define ESP_PANEL_LCD_BL_USE_PWM            (1)
#if ESP_PANEL_LCD_BL_USE_PWM
/**
 *  Backlight LEDC Parameters.
 *
 *  Please refer to https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/ledc.html for details.
 */
    #define ESP_PANEL_LCD_BL_PWM_TIMER          (0)
    #define ESP_PANEL_LCD_BL_PWM_CHANNEL        (0)
    #define ESP_PANEL_LCD_BL_PWM_RESOLUTION     (10)
    #define ESP_PANEL_LCD_BL_PWM_FREQ_HZ        (5000)
#endif /* ESP_PANEL_LCD_BL_USE_PWM */
#endif /* ESP_PANEL_USE_BL */
`

Lzw655, 2024-03-05:
Hi @ReacherS3, did you change the `ESP_PANEL_LCD_IO_BL` macro to your actual pin num which controls the backlight? And the `ESP_PANEL_LCD_BL_ON_LEVEL` too.

BTW, I'm about to complete the new library. I still need to organize the README and other documentation. It should be released within this week. You can check MR #13 .

Lzw655, 2024-03-07:
Hi @ReacherS3, the [new version](https://github.com/esp-arduino-libs/ESP32_Display_Panel/pull/13) is ready now. Can you help me to add your board and example to it?

ReacherS3, 2024-03-07:
@Lzw655 I will help you anytime to add the board and example I will add them as soon as I get a moment @Lzw655

Lzw655, 2024-03-07:
@ReacherS3 Sure, it's okay.

Lzw655, 2024-03-08:
@ReacherS3 Yes, thank you very much!

ReacherS3, 2024-03-11:
@Lzw655 

Can't get all the files upload can i create a online folder docs dropbox etc. 

If okay then you can have access to most of the files I worked on for this device.

Lzw655, 2024-03-12:
Hi @ReacherS3, could you please confirm if your example includes many images? I would prefer the examples in the library to be as simple as possible without being too large. Would you mind simplifying it?

Lzw655, 2024-03-13:
Hi @ReacherS3, I have to release the new version this week. Could you please submit the modifications on time? Please don't feel any pressure, you can make the changes later. We just need to release a new version.

ReacherS3, 2024-03-15:
@Lzw655  No problem at all I will work on them and upload them.

How do we get the new release files loaded?

Lzw655, 2024-03-18:
@ReacherS3 Sorry, which files do you mean by 'the new release files'?

ReacherS3, 2024-03-22:
@Lzw655 Apologies been busy the last few days will update as soon as I can.

On power off and power on I get these errors as well!

`.WiFi connected
IP Address: 192.168.0.123
HTTP server started
Hello Squareline!V8.3.8
I am ESP32_Display_Panel
E (593) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(128): i2c transaction failed
E (593) GT911: touch_gt911_read_cfg(262): GT911 read error!
E (594) GT911: esp_lcd_touch_new_i2c_gt911(122): GT911 init failed
E (600) GT911: Error (0xffffffff)! Touch controller GT911 initialization failed!
E (607) GT911: [GT911.cpp:41] begin(): Check error -1 (ESP_FAIL)
Starting LVGL task
E (632) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(128): i2c transaction failed
E (632) GT911: esp_lcd_touch_gt911_read_data(148): I2C read error!
E (634) LcdTouch: [ESP_PanelLcdTouch.cpp:74] readData(): Check error -1 (ESP_FAIL)
Setup done
E (787) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(128): i2c transaction failed
E (787) GT911: esp_lcd_touch_gt911_read_data(148): I2C read error!
E (789) LcdTouch: [ESP_PanelLcdTouch.cpp:74] readData(): Check error -1 (ESP_FAIL)
E (817) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(128): i2c transaction failed
E (817) GT911: esp_lcd_touch_gt911_read_data(148): I2C read error!
E (819) LcdTouch: [ESP_PanelLcdTouch.cpp:74] readData(): Check error -1 (ESP_FAIL)
E (847) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(128): i2c transaction failed
E (847) GT911: esp_lcd_touch_gt911_read_data(148): I2C read error!
E (849) LcdTouch: [ESP_PanelLcdTouch.cpp:74] readData(): Check error -1 (ESP_FAIL)
E (877) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(128): i2c transaction failed
E (877) GT911: esp_lcd_touch_gt911_read_data(148): I2C read error!
E (879) LcdTouch: [ESP_PanelLcdTouch.cpp:74] readData(): Check error -1 (ESP_FAIL)
E (907) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(128): i2c transaction failed`

ReacherS3, 2024-03-24:
> @ReacherS3 Sorry, which files do you mean by 'the new release files'?

Still working on these files my apology for them being late! @Lzw655

Lzw655, 2024-03-25:
Hi @ReacherS3, there are two I2C addresses for GT911 and it depends on the initial level of the INT pin when powered on. You can change the address by modifying [macro](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/bd59067b8b1318f54b8a1b558b3d58ed5dcc377f/ESP_Panel_Board_Custom.h#L239) instead of using the default one.

ReacherS3, 2024-03-26:
[image]
`c:\Users\esp\Desktop\sqaure2\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp: In member function 'bool ESP_Panel::init()':
C:\Users\esp\Desktop\sqaure2\libraries\ESP32_Display_Panel\examples\SquareLine\v8\Porting/ESP_Panel_Board_Custom.h:32:37: error: 'ST7262_PANEL_IO_3WIRE_SPI_CONFIG' was not declared in this scope; did you mean 'ST7701_PANEL_IO_3WIRE_SPI_CONFIG'?
   32 | #define ESP_PANEL_LCD_NAME          ST7262
      |                                     ^~~~~~
c:\Users\esp\Desktop\sqaure2\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:40:57: note: in definition of macro '_LCD_PANEL_IO_3WIRE_SPI_CONFIG'
   40 |                                                         name##_PANEL_IO_3WIRE_SPI_CONFIG(line_config, scl_active_edge)
      |                                                         ^~~~
c:\Users\esp\Desktop\sqaure2\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:213:9: note: in expansion of macro 'LCD_PANEL_IO_3WIRE_SPI_CONFIG'
  213 |         LCD_PANEL_IO_3WIRE_SPI_CONFIG(ESP_PANEL_LCD_NAME, line_config, ESP_PANEL_LCD_3WIRE_SPI_SCL_ACTIVE_EDGE);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\esp\Desktop\sqaure2\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:213:39: note: in expansion of macro 'ESP_PANEL_LCD_NAME'
  213 |         LCD_PANEL_IO_3WIRE_SPI_CONFIG(ESP_PANEL_LCD_NAME, line_config, ESP_PANEL_LCD_3WIRE_SPI_SCL_ACTIVE_EDGE);
      |                                       ^~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: 'ST7262_PANEL_IO_3WIRE_SPI_CONFIG' was not declared in this scope; did you mean 'ST7701_PANEL_IO_3WIRE_SPI_CONFIG'?`

On the update if you configure macros it's automatically on SPI if you revert to RGB like we use to in the old files it has a compilation error is there a spesific way or do we use the SPI on the updated version?? @Lzw655