esp-arduino-libs/ESP32_Display_Panel #15: "esp_memory_utils.h"

ReacherS3, 2024-03-24:
Good Day Guys when loading the Squareline porting example on the new update I am experiencing the following error not sure why and there's no Library for it?

`c:\Users\esp\Desktop\sqaure2\libraries\ESP32_Display_Panel\src\lcd\ESP_PanelLcd.cpp:13:10: fatal error: esp_memory_utils.h: No such file or directory
 #include "esp_memory_utils.h"
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1`

Lzw655, 2024-03-25:
Hi @ReacherS3, the version of `arduino-esp32` should be v3.0.0-alpha3 as shown in [README](https://github.com/esp-arduino-libs/ESP32_Display_Panel/tree/master#dependencies-and-versions).

ReacherS3, 2024-03-25:
> Hi @ReacherS3, the version of `arduino-esp32` should be v3.0.0-alpha3 as shown in [README](https://github.com/esp-arduino-libs/ESP32_Display_Panel/tree/master#dependencies-and-versions).

Thanks @Lzw655 I have no idea how to get that running.

Lzw655, 2024-03-26:
@ReacherS3 It's a development release. You can follow the [docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#installing-using-arduino-ide) to install it. BTW, should use `https://espressif.github.io/arduino-esp32/package_esp32_dev_index.json` url.

ReacherS3, 2024-03-26:
@Lzw655 Thank you the mistake was on my side I never entered the index URL that's why it never registered in the IDE under ESP32 to update to v3 thank for the assistance and always being available to assist you are the best!!

I will upload the files that I promised as soon as I can get them to fit on Github

Lzw655, 2024-03-26:
Yeah, also thank you for using this library. 😉

ReacherS3, 2024-03-27:
[image]
The previous version supported the CH422G Io expansion chip the new release does not contain the expansion chip files so as soon as I enable #define ESP_PANEL_USE_TOUCH         (0)         // 0/1 it interferes with the display is there a way to fix this?

I believe if I cant enable the touch I will not be able to use #define ESP_PANEL_USE_EXPANDER          (0)         // 0/1 as well?

ReacherS3, 2024-03-27:
If I update accordingly I get the following errors!

`ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29d8
entry 0x403c9880
Squareline porting example start
Initialize panel device
E (19) tca95xx_8bit: write_direction_reg(147): Write direction reg failed
E (20) tca95xx_8bit: reset(164): Write dir reg failed
E (24) tca95xx_8bit: esp_io_expander_new_i2c_tca95xx_8bit(96): Reset failed
E (31) tca95xx_8bit: [TCA95xx_8bit.cpp:63] begin(): Check error -1 (ESP_FAIL)
E (60) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed
E (60) GT911: touch_gt911_read_cfg(362): GT911 read error!
E (61) GT911: esp_lcd_touch_new_i2c_gt911(128): GT911 init failed
E (67) GT911: Error (0xffffffff)! Touch controller GT911 initialization failed!
Initialize LVGL
Create UI
Squareline porting example end`

ReacherS3, 2024-03-27:
Here is my updated file as well!

`/*
 * SPDX-FileCopyrightText: 2023-2024 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#pragma once

// *INDENT-OFF*

/* Set to 1 if using a custom board */
#define ESP_PANEL_USE_CUSTOM_BOARD       (1)         // 0/1

#if ESP_PANEL_USE_CUSTOM_BOARD

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
//////////////////////////// Please update the following macros to configure the LCD panel /////////////////////////////
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
/* Set to 1 when using an LCD panel */
#define ESP_PANEL_USE_LCD           (1)     // 0/1

#if ESP_PANEL_USE_LCD
/**
 * LCD Controller Name. Choose one of the following:
 *      - GC9A01, GC9B71, GC9503
 *      - ILI9341
 *      - NV3022B
 *      - SH8601
 *      - SPD2010
 *      - ST7262, ST7701, ST7789, ST7796, ST77916, ST77922
 */
#define ESP_PANEL_LCD_NAME          ST7262

/* LCD resolution in pixels */
#define ESP_PANEL_LCD_WIDTH         (800)
#define ESP_PANEL_LCD_HEIGHT        (480)

/* LCD Bus Settings */
/**
 * If set to 1, the bus will skip to initialize the corresponding host. Users need to initialize the host in advance.
 * It is useful if other devices use the same host. Please ensure that the host is initialized only once.
 */
#define ESP_PANEL_LCD_BUS_SKIP_INIT_HOST    (1)     // 0/1
/**
 * LCD Bus Type. Choose one of the following:
 *      - ESP_PANEL_BUS_TYPE_I2C (not ready)
 *      - ESP_PANEL_BUS_TYPE_SPI
 *      - ESP_PANEL_BUS_TYPE_QSPI
 *      - ESP_PANEL_BUS_TYPE_I80 (not ready)
 *      - ESP_PANEL_BUS_TYPE_RGB (only supported for ESP32-S3)
 */
#define ESP_PANEL_LCD_BUS_TYPE      (ESP_PANEL_BUS_TYPE_RGB)
/**
 * LCD Bus Parameters.
 *
 * Please refer to https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/lcd.html and
 * https://docs.espressif.com/projects/esp-iot-solution/en/latest/display/lcd/index.html for more details.
 *
 */
#if ESP_PANEL_LCD_BUS_TYPE == ESP_PANEL_BUS_TYPE_SPI

    #define ESP_PANEL_LCD_BUS_HOST_ID           (1)     // Typically set to 1
    #define ESP_PANEL_LCD_SPI_IO_CS             (5)
#if !ESP_PANEL_LCD_BUS_SKIP_INIT_HOST
    #define ESP_PANEL_LCD_SPI_IO_SCK            (7)
    #define ESP_PANEL_LCD_SPI_IO_MOSI           (6)
    #define ESP_PANEL_LCD_SPI_IO_MISO           (-1)    // -1 if not used
#endif
[... 243 lines cut ...]
    #define ESP_PANEL_EXPANDER_I2C_SCL_PULLUP   (0)     // 0/1
    #define ESP_PANEL_EXPANDER_I2C_SDA_PULLUP   (0)     // 0/1
    #define ESP_PANEL_EXPANDER_I2C_IO_SCL       (9)
    #define ESP_PANEL_EXPANDER_I2C_IO_SDA       (8)
#endif
#endif /* ESP_PANEL_USE_EXPANDER */

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
///////////////////////////// Please utilize the following macros to execute any additional code if required. //////////
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
// #define ESP_PANEL_BEGIN_START_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_EXPANDER_START_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_EXPANDER_END_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_LCD_START_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_LCD_END_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_TOUCH_START_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_TOUCH_END_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_BACKLIGHT_START_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_BACKLIGHT_END_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_END_FUNCTION( panel )

#endif /* ESP_PANEL_USE_CUSTOM_BOARD */

// *INDENT-OFF*
`

ReacherS3, 2024-03-27:
With Debug enabled.

`Squareline porting example start
Initialize panel device
D (19) ESP_Panel: Panel init start
D (19) ESP_Panel: Use LCD
D (19) ESP_Panel: Use RGB bus
D (21) ESP_Panel: Create LCD bus
D (24) ESP_Panel: Create LCD device
D (27) ESP_Panel: Use touch
D (29) ESP_Panel: Use I2C bus
D (32) ESP_Panel: Create touch bus
D (35) ESP_Panel: Create touch device
D (39) ESP_Panel: Initialize host
D (42) ESP_Panel: Panel init end
D (45) ESP_PanelHost: Destory
D (47) ESP_Panel: Panel begin start
D (50) ESP_Panel: Begin LCD
D (75) ST7262_CPP: LCD panel @0x3fc9f3ac created
D (75) ESP_PanelLcd: Begin start
D (76) ESP_PanelLcd: Begin end
D (76) ESP_Panel: Begin touch
D (77) ESP_PanelBus_I2C: Panel IO @0x3fcecf54 created
D (82) ESP_PanelTouch: Disable interruption
E (86) i2c: i2c_master_cmd_begin(1490): i2c driver not installed
E (91) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed
E (98) GT911: touch_gt911_read_cfg(362): GT911 read error!
E (103) GT911: esp_lcd_touch_new_i2c_gt911(128): GT911 init failed
E (109) GT911: Error (0x103)! Touch controller GT911 initialization failed!
E (116) GT911_CPP: [ESP_ERR_INVALID_STATE] begin(43): New driver failed
E (122) ESP_Panel: begin(516): Begin touch failed`

Lzw655, 2024-03-28:
Hi @ReacherS3, the ESP32_IO_Expander library didn't support `CH422G`. Are you using a third-party one? If so, can you provide it to me? Then I can add `CH422G` into the `ESP32_IO_Expander`.

[image]

Lzw655, 2024-03-28:
Oh, I realize the reason for your issues, please check the below points:
1. For the touch, set `ESP_PANEL_TOUCH_I2C_ADDRESS` correctly.
[image]
2. For the IO expander, set `ESP_PANEL_EXPANDER_I2C_ADDRESS` correctly. Since some expanders can be assigned various I2C addresses, users should adjust ESP_PANEL_EXPANDER_I2C_ADDRESS based on the actual circumstances.

ReacherS3, 2024-03-28:
Hi @Lzw655 
[image]

Above is the board that I am using it's definately using CH422G as the previous version of Display panel and IO expander supported CH422G CHIP.

[https://files.waveshare.com/wiki/ESP32-S3-Touch-LCD-4.3/S3-4.3-libraries.zip](url)   

This link will download the required files that I used before including the CH422G Chip files

Thanks @Lzw655

ReacherS3, 2024-04-02:
`#include "FS.h"
#include "SD.h"
#include "SPI.h"
#include <ESP_IOExpander_Library.h>

// Extend IO Pin define
#define TP_RST 1
#define LCD_BL 2
#define LCD_RST 3
#define SD_CS 4
#define USB_SEL 5

// I2C Pin define
#define I2C_MASTER_NUM 0
#define I2C_MASTER_SDA_IO 8
#define I2C_MASTER_SCL_IO 9

#define SD_MOSI 11
#define SD_CLK  12
#define SD_MISO 13
#define SD_SS -1
`

`Serial.println("Initialize IO expander");
    /* Initialize IO expander */
    ESP_IOExpander *expander = new ESP_IOExpander_CH422G(I2C_MASTER_NUM, ESP_IO_EXPANDER_I2C_CH422G_ADDRESS_000, I2C_MASTER_SCL_IO, I2C_MASTER_SDA_IO);
    // ESP_IOExpander *expander = new ESP_IOExpander_CH422G(I2C_MASTER_NUM, ESP_IO_EXPANDER_I2C_CH422G_ADDRESS_000);
    expander->init();
    expander->begin();
    expander->multiPinMode(TP_RST | LCD_BL | LCD_RST | SD_CS | USB_SEL, OUTPUT);
    expander->multiDigitalWrite(TP_RST | LCD_BL | LCD_RST, HIGH);

    // use extend GPIO for SD card
    expander->digitalWrite(SD_CS, LOW);

    // Turn off backlight
     expander->digitalWrite(LCD_BL, LOW);

    // When USB_SEL is HIGH, it enables FSUSB42UMX chip and gpio19,gpio20 wired CAN_TX CAN_RX, and then dont use USB Function
    expander->digitalWrite(USB_SEL, LOW);`
    
    I use the CH422G to access the backlight CAN BUS on USB and my SD card functions.
    
    @Lzw655

ReacherS3, 2024-04-02:
`/*
 * Connect the SD card to the following pins:
 *
 * SD Card | ESP32
 *    D2       -
 *    D3       SS
 *    CMD      MOSI
 *    VSS      GND
 *    VDD      3.3V
 *    CLK      SCK
 *    VSS      GND
 *    D0       MISO
 *    D1       -
 */
#include "FS.h"
#include "SD.h"
#include "SPI.h"
#include <ESP_IOExpander_Library.h>

// Extend IO Pin define
#define TP_RST 1
#define LCD_BL 2
#define LCD_RST 3
#define SD_CS 4
#define USB_SEL 5

// I2C Pin define
#define I2C_MASTER_NUM 0
#define I2C_MASTER_SDA_IO 8
#define I2C_MASTER_SCL_IO 9

#define SD_MOSI 11
#define SD_CLK  12
#define SD_MISO 13
#define SD_SS -1

void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\n", dirname);

    File root = fs.open(dirname);
    if(!root){
        Serial.println("Failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println("Not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");
            Serial.println(file.name());
            if(levels){
                listDir(fs, file.path(), levels -1);
            }
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("  SIZE: ");
            Serial.println(file.size());
        }
        file = root.openNextFile();
    }
}

void createDir(fs::FS &fs, const char * path){
    Serial.printf("Creating Dir: %s\n", path);
    if(fs.mkdir(path)){
[... 173 lines cut ...]
    }

    uint64_t cardSize = SD.cardSize() / (1024 * 1024);
    Serial.printf("SD Card Size: %lluMB\n", cardSize);

    listDir(SD, "/", 0);
    createDir(SD, "/mydir");
    listDir(SD, "/", 0);
    removeDir(SD, "/mydir");
    listDir(SD, "/", 2);
    writeFile(SD, "/hello.txt", "Hello ");
    appendFile(SD, "/hello.txt", "World!\n");
    readFile(SD, "/hello.txt");
    deleteFile(SD, "/foo.txt");
    renameFile(SD, "/hello.txt", "/foo.txt");
    readFile(SD, "/foo.txt");
    testFileIO(SD, "/test.txt");
    Serial.printf("Total space: %lluMB\n", SD.totalBytes() / (1024 * 1024));
    Serial.printf("Used space: %lluMB\n", SD.usedBytes() / (1024 * 1024));
}

void loop(){

}
`

ReacherS3, 2024-04-02:
[image]
[image]

I use the CH422G for CAN bus communication, Backlight, SD Card etc.

@Lzw655

Lzw655, 2024-05-08:
Hi @ReacherS3, sorry for the late. I have been very busy these days. 

Thanks to the contribution of @lboue,  we can use CH422G in the [PR ](https://github.com/esp-arduino-libs/ESP32_IO_Expander/pull/4) now. Can you try it again?

lboue, 2024-05-12:
> Hi @ReacherS3, sorry for the late. I have been very busy these days.
> 
> Thanks to the contribution of @lboue, we can use CH422G in the [PR ](https://github.com/esp-arduino-libs/ESP32_IO_Expander/pull/4) now. Can you try it again?

Thanks @Lzw655
This CH422G test file is working with arduino-esp3 3.0.0-alpha3 release: [CH422G_Test_3.0.0-alpha3.ino](https://github.com/lboue/ESP32_IO_Expander/blob/CH422_test/examples/TestFunctions/CH422G_Test_3.0.0-alpha3.ino)

ReacherS3, 2024-05-12:
Thanks to the both of you and the effort and contributions you are awesome guys I am busy myself these days no need to apologize.

I will load up everything once I get back to that part of the project integrating temperature sensors now and GPS tracking capabilities once that's complete I will revert back to the updated library. @lboue @Lzw655 

Once again thank you so much for all the support on this library!!

lboue, 2024-05-17:
@ReacherS3
Can you try last released versions?
- library [ESP32_Display_Panel at version 0.1.1](https://github.com/esp-arduino-libs/ESP32_Display_Panel/releases/tag/v0.1.1)
- library [ESP32_IO_Expander at version 0.0.3](https://github.com/esp-arduino-libs/ESP32_IO_Expander/releases/tag/v0.0.3)

ReacherS3, 2024-05-23:
Thanks @lboue 

Just a quick question I have managed to load all the updated files and managed to get the display running without any errors at all which is pretty good considering I always had the LCD errors they are fixed now.

My concern now is once I start the device and leave it running for about an hour or more the screen starts drifting very slowly to the side or bottom completely throwing out my loaded screen and duplicate display across one screen and everything stops functioning a reset fixes the issue but this will cause issues once installed everywhere in mass production units.

@Lzw655

Lzw655, 2024-05-24:
Hi @ReacherS3, could you take a video to show me the issue? I think it's the screen drift issue of RGB LCD. Did you try the solution [here](https://github.com/esp-arduino-libs/ESP32_Display_Panel#how-to-fix-screen-drift-issue-when-driving-rgb-lcd-with-esp32-s3)?

lboue, 2024-05-24:
@ReacherS3
Perhaps a new thread should be opened for this drift issue?

davidthefuturist, 2024-10-10:
> 
> 
> I use the CH422G for CAN bus communication, Backlight, SD Card etc.
> 
> @Lzw655

Are you able to share any code regarding your setup with CH422G and CANBUS (TWAI)? I am encountering a panel_io_i2c_rx_buffer(135) and GT911: esp_lcd_touch_gt911_read_data(228) error when my CANBUS, display panel, and touch panel are activated. Did you happen to come across that when using your Waveshare S3-LCD-4.3 unit?

davidthefuturist, 2024-10-10:
> I use the CH422G for CAN bus communication, Backlight, SD Card etc.
> 
@ReacherS3 

Are you able to share any code regarding your setup with CH422G and CANBUS (TWAI)? I am encountering a panel_io_i2c_rx_buffer(135) and GT911: esp_lcd_touch_gt911_read_data(228) error when my CANBUS, display panel, and touch panel are activated. Did you happen to come across that when using your Waveshare S3-LCD-4.3 unit?

ReacherS3, 2024-10-10:
@davidthefuturist 
I 9nly started working on this project again about a week ago life has been hectic but it's good being back.

I have manage to getting the settings and files with the code on the sweet spot where it runs wifi connection, RS485 , temperature readings running smoothly no screen tearing no restarts no issues clock updates working.

I will share the files for @Lzw655 and @lboue to double check before they contribute here.

Canbus fully functional as well dependes on what you want receive or transmit.

I mainly use RS485 to read holding registers it works very well on this device.

I will share some images and add libraries later on this week.

davidthefuturist, 2024-10-10:
@ReacherS3 No problem. What's the best way to get in touch with you and share my work so far? I can also publish a repo and add you to it if that would be convenient.

ReacherS3, 2024-10-11:
@davidthefuturist  that perfect let's do that thanks @davidthefuturist

davidthefuturist, 2024-10-11:
@ReacherS3 Just made a repo! I'll be able to upload files shortly.