esp-arduino-libs/ESP32_Display_Panel #18: Add XPT2046

Franck78, 2024-03-29:
Hello,

AliExpress vendors often fail to correctly describe (datasheets) the products they sell. That is very annoying ;) Anyway,

common small TFT like this :
https://www.aliexpress.com/item/1005006042124570.html
say ILI9341 or ST7735, good, supported by the lib.

But they say nothing about the Touchscreen driver !

On some HiRes photo, I can see XPT2046 :
https://www.buydisplay.com/download/ic/XPT2046.pdf

This one is not clearly listed as a driver.
Can you add it ?

Maybe it can be an alias for one of  CST816S, FT5x06 , GT911, GT1151 or  TT21100 touch driver ?

Lzw655, 2024-03-31:
Hi @Franck78,

Sure, I can add it in next week. It's not a big work since we already have the component [esp_lcd_touch_xpt2046](https://components.espressif.com/components/atanisoft/esp_lcd_touch_xpt2046).

Franck78, 2024-04-29:
I have the touche screen. Next step is building the intended device around it.
I confirm prod is still with XPT2046.

[image]

Lzw655, 2024-04-29:
Hi @Franck78, I am working on it. Please wait a minute.

Franck78, 2024-04-29:
oh, no hurry, I don't have pcb  and probably won't do a breakboard test ;)

Lzw655, 2024-04-29:
> oh, no hurry, I don't have pcb and probably won't do a breakboard test ;)

Sure, if your hardware is ready, please help me test the functionality of the XPT2046. Thank you.

closed 2024-04-29 by Lzw655

Franck78, 2024-05-31:
ok, started the project, under arduino IDE

I had to add four variables (from Kconfig.projbuild)
to build it.

file esp_lcd_touch_xpt2046.h:
```
#define ESP_PANEL_TOUCH_XPT2046_Z_THRESHOLD (400)
#define ESP_PANEL_TOUCH_XPT2046_INTERRUPT_MODE (false)
#define ESP_PANEL_TOUCH_XPT2046_XPT2046_VREF_ON_MODE (false)
#define ESP_PANEL_TOUCH_XPT2046_CONVERT_ADC_TO_COORDS (true)

#define CONFIG_XPT2046_Z_THRESHOLD              (ESP_PANEL_TOUCH_XPT2046_Z_THRESHOLD)
#define CONFIG_XPT2046_INTERRUPT_MODE           (ESP_PANEL_TOUCH_XPT2046_INTERRUPT_MODE)
#define CONFIG_XPT2046_VREF_ON_MODE             (ESP_PANEL_TOUCH_XPT2046_XPT2046_VREF_ON_MODE)
#define CONFIG_XPT2046_CONVERT_ADC_TO_COORDS    (ESP_PANEL_TOUCH_XPT2046_CONVERT_ADC_TO_COORDS)

```

Lzw655, 2024-05-31:
Okay... I also added these configurations in the file [ESP_Panel_Conf.h](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/ESP_Panel_Conf.h) of the library.

Franck78, 2024-05-31:
oh, I just updated the library and did not check more than that. I have ESP_Panel_Conf.h in my app directory and didn't check it.

Lzw655, 2024-06-01:
So, did the XTP2046 driver work well for you?

Franck78, 2024-06-01:
> So, did the XTP2046 driver work well for you?

good question ! I'm lost. I'm questioning the purpose of this (esp32_display_panel) !!!

My unique goal is drawing or writing things on the LCD. 

Apparently, there is no such a thing here. A quick look at  "arduino GFX lib" seems to have its own 'driver' for the LCD, the touch and says nothing about binding on your (optimized) driver !!!!!!!!!