esp-arduino-libs/ESP32_Display_Panel #19: platformio可以使用这个库吗

lowino, 2024-04-04:
我用arduino IDE 开发，编译有点慢，想转到vscode。platformio里也有一个esp32_display_panel库，发现不同步的，用不了。platformio可以使用这个库吗？

Lzw655, 2024-04-07:
抱歉，这个库目前还未在 platformio 上使用过，我会抽空尝试一下。

isthaison, 2024-05-28:
```

[env:ESP-LCD]
platform = espressif32@6.7.0
board = ESP-LCD
framework = arduino
platform_packages = 
	platformio/framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git
	platformio/framework-arduinoespressif32-libs@https://github.com/espressif/esp32-arduino-libs.git#idf-release/v5.1
upload_speed = 921600
monitor_speed = 115200
build_flags = 
	-DBOARD_HAS_PSRAM
	-DLV_CONF_INCLUDE_SIMPLE
	-DDISABLE_ALL_LIBRARY_WARNINGS
	-DARDUINO_USB_CDC_ON_BOOT=1
	-DCORE_DEBUG_LEVEL=1
	-DLV_LVGL_H_INCLUDE_SIMPLE
	-I src
	-D VN=1
	-D ESP32S3=1
lib_deps = 
	https://github.com/esp-arduino-libs/ESP32_Display_Panel.git
	https://github.com/esp-arduino-libs/ESP32_IO_Expander.git
    https://github.com/lvgl/lvgl.git#release/v8.3

```
Please try following the above configuration.

hrituraj-rohariya, 2024-05-29:
Thanks a lot @Lzw655, this would mean a lot, 

Hello @isthaison Congrats And Thanks a lot for letting us know that at least efforts to make the library work on platformio are not always futile...

I am using PlatfformIo to drive RGB LCD 3 wire spi DataPins(15)

According to your Comment
https://github.com/esp-arduino-libs/ESP32_Display_Panel/issues/19#issuecomment-2134217055

have modified my platform.ini file
hope its correct

[platform.ini.txt](https://github.com/esp-arduino-libs/ESP32_Display_Panel/files/15481223/platform.ini.txt)

Project Dependancies
[dependencies.txt](https://github.com/esp-arduino-libs/ESP32_Display_Panel/files/15481335/dependencies.txt)

This is main.cpp directing to just one One Lvgl Task
[main.txt](https://github.com/esp-arduino-libs/ESP32_Display_Panel/files/15481255/main.txt)

Here is the task defined
[DEVICE_LVGL_V8.txt](https://github.com/esp-arduino-libs/ESP32_Display_Panel/files/15481293/DEVICE_LVGL_V8.txt)

Library Specific
[ESP_Panel_Board_Custom.txt](https://github.com/esp-arduino-libs/ESP32_Display_Panel/files/15481312/ESP_Panel_Board_Custom.txt)
[ESP_Panel_Conf.txt](https://github.com/esp-arduino-libs/ESP32_Display_Panel/files/15481313/ESP_Panel_Conf.txt)

 
here is the compile log  to further understand...

PlatformIO Test log

[TESTLOG_Platform_esp32@6_7_0_IDF_5_1_1.log](https://github.com/esp-arduino-libs/ESP32_Display_Panel/files/15481200/TESTLOG_Platform_esp32%406_7_0_IDF_5_1_1.log)

Lzw655, 2024-05-29:
Hi @isthaison, 

Thank you for testing it. I checked your [ESP_Panel_Board_Custom.txt](https://github.com/esp-arduino-libs/ESP32_Display_Panel/files/15481312/ESP_Panel_Board_Custom.txt) file and the macro [ESP_PANEL_USE_CUSTOM_BOARD ](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/5c44656c4d143ecf53aa5864aefe40bda30875ef/ESP_Panel_Board_Custom.h#L12) should be set to `1` which is `0` now. Please change it and try again.

isthaison, 2024-05-29:
@Lzw655  You tagged the wrong person 👯 
@hrituraj-rohariya

Lzw655, 2024-05-29:
Oh, that's embarrassing... 😂

hrituraj-rohariya, 2024-05-29:
🤣🤣🤣

hrituraj-rohariya, 2024-05-29:
> Hi @isthaison,
>
> Thank you for testing it. I checked your [ESP_Panel_Board_Custom.txt](https://github.com/esp-arduino-libs/ESP32_Display_Panel/files/15481312/ESP_Panel_Board_Custom.txt) file and the [...]

Hello @Lzw655  i have defined the ESP_PANEL_USE_CUSTOM_BOARD to 1in ESP_Config.h file...and have  included the file in ESP_Panel_Board_Custom.h
Its a Macro and its working from  there too...

appleshaman, 2024-06-08:
> ```
>
> [env:ESP-LCD]
> platform = espressif32@6.7.0
> board = ESP-LCD
> framework = arduino
> platform_packages =
> 	platformio/framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git [...]

> ```
>
> [env:ESP-LCD]
> platform = espressif32@6.7.0
> board = ESP-LCD
> framework = arduino
> platform_packages =
> 	platformio/framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git [...]

Hello @isthaison, I was just wondering where to find the installed library after adding them through ini file? Could not find them either in the lib or libdeps folder

isthaison, 2024-06-08:
there are include on folder .pio

Nguyễn Thái Sơn

appleshaman, 2024-06-10:
> […](#)

Thanks mate, it appeared after I restart the vs code, but I have one problems is once I add 
these two lines:
platform_packages = 
	platformio/framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git
	platformio/framework-arduinoespressif32-libs@https://github.com/espressif/esp32-arduino-libs.git#idf-release/v5.1
it always gives me an error that says:
Error! Failed to extract upstream toolchainconfigurations:
Bad package version `esp-12.2.0_20230208`

Lzw655, 2024-06-17:
Duplicated with #47

Lzw655, 2024-06-17:
Hi @lowino,

你好，这个库刚刚合入了一个在 PlatformIO 上使用的示例，可以看看是否符合你的需求。

isthaison, 2024-06-17:
https://github.com/esp-arduino-libs/ESP32_Display_Panel?tab=readme-ov-file#how-to-fix-screen-drift-issue-when-driving-rgb-lcd-with-esp32-s3. They should handle several devices like here, C:\Users\~\.platformio. it also launches like arduino

isthaison, 2024-06-17:
[image]
I see you are using a package from platformio that is supporting it. it does not support arduino-esp32 v3.x.x yet. Please follow the instructions I have suggested above.

appleshaman, 2024-06-17:
That finally works,  thanks!

lowino, 2024-06-24:
编译通过 。感谢 @isthaison @Lzw655 
在编译时，当使用Adafruit的板子进行编译时，会出现与toolchain相关的错误，这可能是由于esp32更新到6.X版本产生的问题。使用其他板子进行编译则未出现此问题，这个信息可能对遇到类似情况的开发者有所帮助。

closed 2024-06-24 by lowino