esp-arduino-libs/ESP32_Display_Panel #22: Support SC7283 driver IC ?

jerryliugithub, 2024-05-09:
I have successfully tested the WiFiClock example using the ST7262 driver IC with an 800x480 LCD. Now, I'm attempting to make the same RGB interface work with an SC7283 driver IC and a 480x272 LCD. However, due to the difference in resolution, the display appears in the bottom right corner, showing only a quarter of the screen. 

When I modify the 
#define ESP_PANEL_LCD_WIDTH (480) 
#define ESP_PANEL_LCD_HEIGHT (272)
the following error appears : 
13:55:55.308 -> Build:Mar 27 2021
13:55:55.308 -> rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
13:55:55.342 -> Saved PC:0x4037b8ee
13:55:55.342 -> SPIWP:0xee
13:55:55.342 -> mode:DIO, clock div:1
13:55:55.342 -> load:0x3fce3818,len:0x508
13:55:55.342 -> load:0x403c9700,len:0x4
13:55:55.342 -> load:0x403c9704,len:0xad0
13:55:55.342 -> load:0x403cc700,len:0x29d8
13:55:55.342 -> entry 0x403c9880
13:55:55.782 -> Squareline WiFi clock example start
13:55:55.857 -> Initialize panel device
13:55:55.857 -> D (79) ESP_Panel: Panel init start
13:55:55.857 -> D (79) ESP_Panel: Use LCD
13:55:55.857 -> D (79) ESP_Panel: Use RGB bus
13:55:55.857 -> D (79) ESP_Panel: Create LCD bus
13:55:55.891 -> D (81) ESP_Panel: Create LCD device
13:55:55.891 -> D (84) ESP_Panel: Initialize host
13:55:55.891 -> D (87) ESP_Panel: Panel init end
13:55:55.891 -> D (90) ESP_PanelHost: Destory
13:55:55.891 -> D (93) ESP_Panel: Panel begin start
13:55:55.891 -> D (96) ESP_Panel: Begin LCD
13:55:55.891 -> E (99) lcd_panel.rgb: esp_lcd_new_rgb_panel(251): fb size must be even multiple of bounce buffer size
13:55:55.891 -> 
13:55:55.974 -> assert failed: virtual bool ESP_PanelLcd_ST7262::init() C:\Users\jerry\projects\arduinov3\portable\libraries\ESP32_Display_Panel\src\lcd\ST7262.cpp:65 ((esp_lcd_new_rgb_panel(vendor_config.rgb_config
13:55:55.974 -> 
13:55:55.974 -> 
13:55:55.974 -> Backtrace: 0x4037742a:0x3fcebdb0 0x4037e46d:0x3fcebdd0 0x403847dd:0x3fcebdf0 0x42006bda:0x3fcebf⸮⸮⸮

In the case of using the same RGB interface, can some parameters modified for the ST7262 be applied to the SC7283? Or is there support provided for the SC7283?

Lzw655, 2024-05-09:
Hi @jerryliugithub, according to the error message output, this is caused by the RGB driver's bounce buffer size not meeting the requirements, which means that the size of the RGB frame buffer should be an even multiple of the bounce buffer size. 

For a resolution of 800x480, if the [configuration file](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/403f51a64521b1027af217c69bf1a488efb17a0f/src/board/espressif/ESP32_S3_LCD_EV_BOARD_2.h#L57) and the [WiFiClock example](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/403f51a64521b1027af217c69bf1a488efb17a0f/examples/SquareLine/v8/WiFiClock/WiFiClock.ino#L801) (if the avoid tearing feature is enabled) default to `ESP_PANEL_LCD_WIDTH * 10` (800 * 10), it may meet the requirements. However, for a resolution of 480x272, the bounce buffer size = `480 * 10` does not meet the requirements. 

Therefore, if you have enabled the avoid tearing feature, you can modify `LVGL_PORT_RGB_BOUNCE_BUFFER_SIZE` in `lvgl_port_v8.h` in the example to `272 * 20`. If not, you need to modify `ESP_PANEL_LCD_RGB_BOUNCE_BUF_SIZE` in the configuration file to `272 * 20`, or directly set it in the code according to [API ](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/403f51a64521b1027af217c69bf1a488efb17a0f/examples/SquareLine/v8/WiFiClock/WiFiClock.ino#L801) `rgb_bus->configRgbBounceBufferSize(272 * 20);`.

jerryliugithub, 2024-05-10:
Hi @Lzw655 , That's great! It resolved my problem so precisely, I'm really grateful. Now that this issue is sorted, another problem has emerged with the touch IC. Currently, the LCD is using the ST1633i, but it's not supported in the current library. I wonder if you have any suggestions. Thank you for your tremendous help once again.

Lzw655, 2024-05-10:
If you're not in a hurry, I can support it in the library in these two weeks. Do you have the datasheet of ST1633i? And, since I don't have the hardware, please help me to assist with testing.

jerryliugithub, 2024-05-10:
That's great! There are no issues with the testing part. Currently, based on the preliminary information available on the website, I have access to the following data links: https://www.crystalfontz.com/products/document/4590/Sitronix_Touch_IC_Protocol_A_V2.7.pdf and https://www.orientdisplay.com/wp-content/uploads/2021/12/ST1633i_datasheet_v1.6.pdf. 

I'm not sure if they're helpful. Additionally, if possible, we'll further reach out to the supplier to obtain detailed information.

Lzw655, 2024-05-10:
OK, I will try to make a driver for it, thank you!

Lzw655, 2024-05-23:
Hi @jerryliugithub, I added the st1633 (i2c) driver in the branch [feat/add_touch_st1633](https://github.com/esp-arduino-libs/ESP32_Display_Panel/tree/feat/add_touch_st1633). Please help me test it. Thank you very much!

jerryliugithub, 2024-05-27:
I have not yet enabled the TOUCH function. During the initial compilation, the following error occurred. Did I miss anything?

C:\Users\jerry\projects\arduinov3\portable\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp: In member function 'bool ESP_Panel::init()':
C:\Users\jerry\projects\arduinov3\portable\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:216:24: error: 'ESP_PANEL_LCD_RGB_CLK_HZ' was not declared in this scope; did you mean 'ESP_PANEL_LCD_SPI_CLK_HZ'?
  216 |             .pclk_hz = ESP_PANEL_LCD_RGB_CLK_HZ,
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~
      |                        ESP_PANEL_LCD_SPI_CLK_HZ
C:\Users\jerry\projects\arduinov3\portable\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:219:34: error: 'ESP_PANEL_LCD_RGB_HPW' was not declared in this scope; did you mean 'ESP_PANEL_LCD_BUS_HOST'?
  219 |             .hsync_pulse_width = ESP_PANEL_LCD_RGB_HPW,
      |                                  ^~~~~~~~~~~~~~~~~~~~~
      |                                  ESP_PANEL_LCD_BUS_HOST
C:\Users\jerry\projects\arduinov3\portable\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:220:33: error: 'ESP_PANEL_LCD_RGB_HBP' was not declared in this scope; did you mean 'ESP_PANEL_LCD_BUS_HOST'?
  220 |             .hsync_back_porch = ESP_PANEL_LCD_RGB_HBP,
      |                                 ^~~~~~~~~~~~~~~~~~~~~
      |                                 ESP_PANEL_LCD_BUS_HOST
C:\Users\jerry\projects\arduinov3\portable\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:221:34: error: 'ESP_PANEL_LCD_RGB_HFP' was not declared in this scope; did you mean 'ESP_PANEL_LCD_BUS_HOST'?
  221 |             .hsync_front_porch = ESP_PANEL_LCD_RGB_HFP,
      |                                  ^~~~~~~~~~~~~~~~~~~~~
      |                                  ESP_PANEL_LCD_BUS_HOST
C:\Users\jerry\projects\arduinov3\portable\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:222:34: error: 'ESP_PANEL_LCD_RGB_VPW' was not declared in this scope; did you mean 'ESP_PANEL_LCD_RST_LEVEL'?
  222 |             .vsync_pulse_width = ESP_PANEL_LCD_RGB_VPW,
      |                                  ^~~~~~~~~~~~~~~~~~~~~
      |                                  ESP_PANEL_LCD_RST_LEVEL
C:\Users\jerry\projects\arduinov3\portable\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:223:33: error: 'ESP_PANEL_LCD_RGB_VBP' was not declared in this scope; did you mean 'ESP_PANEL_LCD_RST_LEVEL'?
  223 |             .vsync_back_porch = ESP_PANEL_LCD_RGB_VBP,
      |                                 ^~~~~~~~~~~~~~~~~~~~~
      |                                 ESP_PANEL_LCD_RST_LEVEL
C:\Users\jerry\projects\arduinov3\portable\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:224:34: error: 'ESP_PANEL_LCD_RGB_VFP' was not declared in this scope; did you mean 'ESP_PANEL_LCD_RST_LEVEL'?
  224 |             .vsync_front_porch = ESP_PANEL_LCD_RGB_VFP,
      |                                  ^~~~~~~~~~~~~~~~~~~~~
      |                                  ESP_PANEL_LCD_RST_LEVEL
C:\Users\jerry\projects\arduinov3\portable\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:226:36: error: 'ESP_PANEL_LCD_RGB_PCLK_ACTIVE_NEG' was not declared in this scope
  226 |                 .pclk_active_neg = ESP_PANEL_LCD_RGB_PCLK_ACTIVE_NEG,
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\jerry\projects\arduinov3\portable\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:229:23: error: 'ESP_PANEL_LCD_RGB_DATA_WIDTH' was not declared in this scope; did you mean 'SOC_LCD_RGB_DATA_WIDTH'?
  229 |         .data_width = ESP_PANEL_LCD_RGB_DATA_WIDTH,
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                       SOC_LCD_RGB_DATA_WIDTH
C:\Users\jerry\projects\arduinov3\portable\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:230:27: error: 'ESP_PANEL_LCD_RGB_PIXEL_BITS' was not declared in this scope; did you mean 'ESP_PANEL_LCD_SPI_PARAM_BITS'?
  230 |         .bits_per_pixel = ESP_PANEL_LCD_RGB_PIXEL_BITS,
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                           ESP_PANEL_LCD_SPI_PARAM_BITS
C:\Users\jerry\projects\arduinov3\portable\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:231:20: error: 'ESP_PANEL_LCD_RGB_FRAME_BUF_NUM' was not declared in this scope
  231 |         .num_fbs = ESP_PANEL_LCD_RGB_FRAME_BUF_NUM,
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\jerry\projects\arduinov3\portable\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:232:34: error: 'ESP_PANEL_LCD_RGB_BOUNCE_BUF_SIZE' was not declared in this scope
  232 |         .bounce_buffer_size_px = ESP_PANEL_LCD_RGB_BOUNCE_BUF_SIZE,
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\jerry\projects\arduinov3\portable\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:235:27: error: 'ESP_PANEL_LCD_RGB_IO_HSYNC' was not declared in this scope; did you mean 'ESP_PANEL_LCD_SPI_IO_CS'?
  235 |         .hsync_gpio_num = ESP_PANEL_LCD_RGB_IO_HSYNC,
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |                           ESP_PANEL_LCD_SPI_IO_CS
C:\Users\jerry\projects\arduinov3\portable\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:236:27: error: 'ESP_PANEL_LCD_RGB_IO_VSYNC' was not declared in this scope; did you mean 'ESP_PANEL_LCD_SPI_IO_CS'?
  236 |         .vsync_gpio_num = ESP_PANEL_LCD_RGB_IO_VSYNC,
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |                           ESP_PANEL_LCD_SPI_IO_CS
[... cut ...]
C:\Users\jerry\projects\arduinov3\portable\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:255:13: error: 'ESP_PANEL_LCD_RGB_IO_DATA14' was not declared in this scope; did you mean 'ESP_PANEL_LCD_SPI_IO_DC'?
  255 |             ESP_PANEL_LCD_RGB_IO_DATA14,
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |             ESP_PANEL_LCD_SPI_IO_DC
C:\Users\jerry\projects\arduinov3\portable\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:256:13: error: 'ESP_PANEL_LCD_RGB_IO_DATA15' was not declared in this scope; did you mean 'ESP_PANEL_LCD_SPI_IO_DC'?
  256 |             ESP_PANEL_LCD_RGB_IO_DATA15,
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |             ESP_PANEL_LCD_SPI_IO_DC

Lzw655, 2024-05-27:
Could you please share the file named "ESP_Panel_Board_Custom.h" that is being used by the project?

jerryliugithub, 2024-05-27:
/*
 * SPDX-FileCopyrightText: 2023-2024 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#pragma once

// *INDENT-OFF*

/* Set to 1 if using a custom board */
#define ESP_PANEL_USE_CUSTOM_BOARD       (1)         // 0/1

#if ESP_PANEL_USE_CUSTOM_BOARD

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
//////////////////////////// Please update the following macros to configure the LCD panel /////////////////////////////
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
/* Set to 1 when using an LCD panel */
#define ESP_PANEL_USE_LCD           (1)     // 0/1

#if ESP_PANEL_USE_LCD
/**
 * LCD Controller Name. Choose one of the following:
 *      - GC9A01, GC9B71, GC9503
 *      - ILI9341
 *      - NV3022B
 *      - SH8601
 *      - SPD2010
 *      - ST7262, ST7701, ST7789, ST7796, ST77916, ST77922
 */
#define ESP_PANEL_LCD_NAME          ST7262

/* LCD resolution in pixels */
#define ESP_PANEL_LCD_WIDTH         (480)
#define ESP_PANEL_LCD_HEIGHT        (272)

/* LCD Bus Settings */
/**
 * If set to 1, the bus will skip to initialize the corresponding host. Users need to initialize the host in advance.
 * It is useful if other devices use the same host. Please ensure that the host is initialized only once.
 *
 * Set to 1 if only the RGB interface is used without the 3-wire SPI interface,
 */
#define ESP_PANEL_LCD_BUS_SKIP_INIT_HOST    (1)     // 0/1
/**
 * LCD Bus Type. Choose one of the following:
 *      - ESP_PANEL_BUS_TYPE_I2C (not ready)
 *      - ESP_PANEL_BUS_TYPE_SPI
 *      - ESP_PANEL_BUS_TYPE_QSPI
 *      - ESP_PANEL_BUS_TYPE_I80 (not ready)
 *      - ESP_PANEL_BUS_TYPE_RGB (only supported for ESP32-S3)
 */
#define ESP_PANEL_LCD_BUS_TYPE      (ESP_PANEL_BUS_TYPE_RGB)
/**
 * LCD Bus Parameters.
 *
 * Please refer to https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/lcd.html and
 * https://docs.espressif.com/projects/esp-iot-solution/en/latest/display/lcd/index.html for more details.
 *
 */
#if ESP_PANEL_LCD_BUS_TYPE == ESP_PANEL_BUS_TYPE_SPI

    #define ESP_PANEL_LCD_BUS_HOST_ID           (1)     // Typically set to 1
    #define ESP_PANEL_LCD_SPI_IO_CS             (5)
#if !ESP_PANEL_LCD_BUS_SKIP_INIT_HOST
    #define ESP_PANEL_LCD_SPI_IO_SCK            (7)
    #define ESP_PANEL_LCD_SPI_IO_MOSI           (6)
    #define ESP_PANEL_LCD_SPI_IO_MISO           (-1)    // -1 if not used
#endif
[... 274 lines cut ...]
// #define ESP_PANEL_BEGIN_LCD_END_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_TOUCH_START_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_TOUCH_END_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_BACKLIGHT_START_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_BACKLIGHT_END_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_END_FUNCTION( panel )

/**
 * Do not change the following versions, they are used to check if the configurations in this file are compatible with
 * the current version of `ESP_Panel_Board_Custom.h` in the library. The detailed rules are as follows:
 *
 *   1. If the major version is not consistent, then the configurations in this file are incompatible with the library
 *      and must be replaced with the file from the library.
 *   2. If the minor version is not consistent, this file might be missing some new configurations, which will be set to
 *      default values. It is recommended to replace it with the file from the library.
 *   3. Even if the patch version is not consistent, it will not affect normal functionality.
 *
 */
#define ESP_PANEL_BOARD_CUSTOM_FILE_VERSION_MAJOR 0
#define ESP_PANEL_BOARD_CUSTOM_FILE_VERSION_MINOR 1
#define ESP_PANEL_BOARD_CUSTOM_FILE_VERSION_PATCH 0

#endif /* ESP_PANEL_USE_CUSTOM_BOARD */

// *INDENT-OFF*

jerryliugithub, 2024-05-27:
`/*
 * SPDX-FileCopyrightText: 2023-2024 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#pragma once

// *INDENT-OFF*

/* Set to 1 if using a custom board */
#define ESP_PANEL_USE_CUSTOM_BOARD       (1)         // 0/1

#if ESP_PANEL_USE_CUSTOM_BOARD

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
//////////////////////////// Please update the following macros to configure the LCD panel /////////////////////////////
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
/* Set to 1 when using an LCD panel */
#define ESP_PANEL_USE_LCD           (1)     // 0/1

#if ESP_PANEL_USE_LCD
/**
 * LCD Controller Name. Choose one of the following:
 *      - GC9A01, GC9B71, GC9503
 *      - ILI9341
 *      - NV3022B
 *      - SH8601
 *      - SPD2010
 *      - ST7262, ST7701, ST7789, ST7796, ST77916, ST77922
 */
#define ESP_PANEL_LCD_NAME          ST7262

/* LCD resolution in pixels */
#define ESP_PANEL_LCD_WIDTH         (480)
#define ESP_PANEL_LCD_HEIGHT        (272)

/* LCD Bus Settings */
/**
 * If set to 1, the bus will skip to initialize the corresponding host. Users need to initialize the host in advance.
 * It is useful if other devices use the same host. Please ensure that the host is initialized only once.
 *
 * Set to 1 if only the RGB interface is used without the 3-wire SPI interface,
 */
#define ESP_PANEL_LCD_BUS_SKIP_INIT_HOST    (1)     // 0/1
/**
 * LCD Bus Type. Choose one of the following:
 *      - ESP_PANEL_BUS_TYPE_I2C (not ready)
 *      - ESP_PANEL_BUS_TYPE_SPI
 *      - ESP_PANEL_BUS_TYPE_QSPI
 *      - ESP_PANEL_BUS_TYPE_I80 (not ready)
 *      - ESP_PANEL_BUS_TYPE_RGB (only supported for ESP32-S3)
 */
#define ESP_PANEL_LCD_BUS_TYPE      (ESP_PANEL_BUS_TYPE_RGB)
/**
 * LCD Bus Parameters.
 *
 * Please refer to https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/lcd.html and
 * https://docs.espressif.com/projects/esp-iot-solution/en/latest/display/lcd/index.html for more details.
 *
 */
#if ESP_PANEL_LCD_BUS_TYPE == ESP_PANEL_BUS_TYPE_SPI

    #define ESP_PANEL_LCD_BUS_HOST_ID           (1)     // Typically set to 1
    #define ESP_PANEL_LCD_SPI_IO_CS             (5)
#if !ESP_PANEL_LCD_BUS_SKIP_INIT_HOST
    #define ESP_PANEL_LCD_SPI_IO_SCK            (7)
    #define ESP_PANEL_LCD_SPI_IO_MOSI           (6)
    #define ESP_PANEL_LCD_SPI_IO_MISO           (-1)    // -1 if not used
#endif
[... 274 lines cut ...]
// #define ESP_PANEL_BEGIN_LCD_END_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_TOUCH_START_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_TOUCH_END_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_BACKLIGHT_START_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_BACKLIGHT_END_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_END_FUNCTION( panel )

/**
 * Do not change the following versions, they are used to check if the configurations in this file are compatible with
 * the current version of `ESP_Panel_Board_Custom.h` in the library. The detailed rules are as follows:
 *
 *   1. If the major version is not consistent, then the configurations in this file are incompatible with the library
 *      and must be replaced with the file from the library.
 *   2. If the minor version is not consistent, this file might be missing some new configurations, which will be set to
 *      default values. It is recommended to replace it with the file from the library.
 *   3. Even if the patch version is not consistent, it will not affect normal functionality.
 *
 */
#define ESP_PANEL_BOARD_CUSTOM_FILE_VERSION_MAJOR 0
#define ESP_PANEL_BOARD_CUSTOM_FILE_VERSION_MINOR 1
#define ESP_PANEL_BOARD_CUSTOM_FILE_VERSION_PATCH 0

#endif /* ESP_PANEL_USE_CUSTOM_BOARD */

// *INDENT-OFF*`

Lzw655, 2024-05-27:
Thank you for your information. I noticed there is a wrong header order after updating the configuration file `ESP_Panel_Board_Internal.h`. And I fixed it in the branch [feat/add_touch_st1633](https://github.com/esp-arduino-libs/ESP32_Display_Panel/tree/feat/add_touch_st1633). Please update it and try again. Thank you very much!

Lzw655, 2024-05-27:
Ok... It's a little strange. I did set the I2C address of ST1633 to be 0x55 in the driver. Please allow me to test the output waveform with a logic analyzer tomorrow. Thank you for your assistance!

jerryliugithub, 2024-05-28:
The problem seems to be the I2C ACK signal?
Example of WireScan.ino, ACK can be received normally at 0x55, while other addresses are NACK
But in the example of TFT LCD Touch, there is no ACK for 0x55 at 100KHz/400KHz. Could it be that the SDA PIN is not released?

WireScan.ino --
[image]

Touch I2C 100KHz --
[image]

Touch I2C 400KHz --
[image]

Lzw655, 2024-05-28:
Thank you for capturing the waveform yourself. I'm sorry, but I also can't determine the key issue. Did you enable the SDA and SCL pull-up in the `WireScan.ino`?

jerryliugithub, 2024-05-29:
I directly used the example in the library and added setPins(). I have an external resistor pull high in the hardware. The program is as follows

#include "Wire.h"

void setup() {
  Serial.begin(115200);
  Wire.setPins(8, 9);
  Wire.begin();
}

void loop() {
  byte error, address;
  int nDevices = 0;

  delay(5000);

  Serial.println("Scanning for I2C devices ...");
  for(address = 0x01; address < 0x7f; address++){
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    if (error == 0){
      Serial.printf("I2C device found at address 0x%02X\n", address);
      nDevices++;
    } else if(error != 2){
      Serial.printf("Error %d at address 0x%02X\n", error, address);
    }
  }
  if (nDevices == 0){
    Serial.println("No I2C devices found");
  }
}

Lzw655, 2024-05-29:
Oh, I see. Can you try to read a byte from `0x00` register based on `WireScan.ino`?
[image]

jerryliugithub, 2024-05-29:
I can read 16bytes using the WireMaster.ino example in the library. And when I press the screen, the value also changes.

code:

#include "Wire.h"

#define I2C_DEV_ADDR 0x55

uint32_t i = 0;

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Wire.setPins(8, 9);
  Wire.begin();
}

void loop() {
  delay(5000);

  //Write message to the slave
  //Wire.beginTransmission(I2C_DEV_ADDR);
  //Wire.printf("Hello World! %lu", i++);
  //uint8_t error = Wire.endTransmission(true);
  //Serial.printf("endTransmission: %u\n", error);
  
  //Read 16 bytes from the slave
  uint8_t bytesReceived = Wire.requestFrom(I2C_DEV_ADDR, 16);
  Serial.printf("requestFrom: %u\n", bytesReceived);
  if((bool)bytesReceived){ //If received more than zero bytes
    uint8_t temp[bytesReceived];
    Wire.readBytes(temp, bytesReceived);
    log_print_buf(temp, bytesReceived);
  }
}

log:

09:48:10.650 -> GPIO Info:
09:48:10.650 -> ------------------------------------------
09:48:10.650 ->   GPIO : BUS_TYPE[bus/unit][chan]
09:48:10.650 ->   --------------------------------------  
09:48:10.650 ->      8 : I2C_MASTER_SDA[0]
09:48:10.650 ->      9 : I2C_MASTER_SCL[0]
09:48:10.650 ->     43 : UART_TX[0]
09:48:10.650 ->     44 : UART_RX[0]
09:48:10.650 -> ============ After Setup End =============
09:48:15.657 -> requestFrom: 16
09:48:15.657 -> 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // ................
09:48:20.681 -> requestFrom: 16
09:48:20.681 -> 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // ................
09:48:25.681 -> requestFrom: 16
09:48:25.681 -> 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // ................
09:48:30.723 -> requestFrom: 16
09:48:30.723 -> 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // ................
09:48:35.737 -> requestFrom: 16
09:48:35.737 -> 0x00, 0x00, 0x00, 0x86, 0xaa, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // ................
09:48:40.761 -> requestFrom: 16
09:48:40.761 -> 0x00, 0x00, 0x00, 0x86, 0xaa, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // ................
09:48:45.744 -> requestFrom: 16
09:48:45.744 -> 0x00, 0x00, 0x10, 0xce, 0x3c, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // ....<...........
09:48:50.781 -> requestFrom: 16
09:48:50.781 -> 0x00, 0x00, 0x10, 0xce, 0x3c, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // ....<...........
09:48:55.799 -> requestFrom: 16
09:48:55.799 -> 0x00, 0x00, 0x10, 0xce, 0x3c, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // ....<...........
09:49:00.787 -> requestFrom: 16
09:49:00.787 -> 0x00, 0x00, 0x10, 0xce, 0x3c, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, // ....<...........

Lzw655, 2024-05-29:
That's really odd. I reviewed the implementation of the `Wire` class, and it seems to utilize the same low-level driver as the LCD library. Also, can you please double-check if running only the `I2C.ino` example from the `examples/Touch/I2C/` directory causes any problems?

jerryliugithub, 2024-05-29:
I'll check the logic analyzer again.

log:

10:22:24.186 -> ------------------------------------------
10:22:24.186 -> Board Info:
10:22:24.186 -> ------------------------------------------
10:22:24.186 ->   Arduino Board     : ESP32S3_DEV
10:22:24.186 ->   Arduino Variant   : esp32s3
10:22:24.186 ->   Arduino FQBN      : esp32:esp32:esp32s3:JTAGAdapter=default,PSRAM=opi,FlashMode=qio,FlashSize=8M,LoopCore=1,EventsCore=1,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,PartitionScheme=default_8MB,CPUFreq=240,UploadSpeed=921600,DebugLevel=verbose,EraseFlash=none,ZigbeeMode=default
10:22:24.233 -> ============ Before Setup End ============
10:22:24.313 -> [   633][V][esp32-hal-uart.c:409] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(44) txPin(43)
10:22:24.313 -> I2C touch example start
10:22:24.313 -> Create I2C bus
10:22:24.313 -> D (404) ESP_PanelBus_I2C: Init host[0]
10:22:24.313 -> D (404) ESP_PanelBus_I2C: Panel IO @0x3fcece64 created
10:22:24.360 -> Create touch device
10:22:24.360 -> D (415) ESP_PanelTouch: Disable interruption
10:22:24.360 -> I (415) ST1633: Firmware version: 0(8.3.2.5), Max.X: 16, Max.Y: 257
10:22:24.360 -> I (416) ST1633: Touch panel create success, version: 0.1.0
10:22:24.360 -> I2C touch example end
10:22:24.360 -> =========== After Setup Start ============
10:22:24.360 -> INTERNAL Memory Info:
10:22:24.360 -> ------------------------------------------
10:22:24.360 ->   Total Size        :   397836 B ( 388.5 KB)
10:22:24.360 ->   Free Bytes        :   362576 B ( 354.1 KB)
10:22:24.360 ->   Allocated Bytes   :    29812 B (  29.1 KB)
10:22:24.360 ->   Minimum Free Bytes:   357528 B ( 349.1 KB)
10:22:24.360 ->   Largest Free Block:   327668 B ( 320.0 KB)
10:22:24.407 -> ------------------------------------------
10:22:24.407 -> SPIRAM Memory Info:
10:22:24.407 -> ------------------------------------------
10:22:24.407 ->   Total Size        :  8388608 B (8192.0 KB)
10:22:24.407 ->   Free Bytes        :  8386020 B (8189.5 KB)
10:22:24.407 ->   Allocated Bytes   :      260 B (   0.3 KB)
10:22:24.407 ->   Minimum Free Bytes:  8386020 B (8189.5 KB)
10:22:24.407 ->   Largest Free Block:  8257524 B (8064.0 KB)
10:22:24.407 -> ------------------------------------------
10:22:24.407 -> GPIO Info:
10:22:24.407 -> ------------------------------------------
10:22:24.407 ->   GPIO : BUS_TYPE[bus/unit][chan]
10:22:24.407 ->   --------------------------------------  
10:22:24.455 ->     43 : UART_TX[0]
10:22:24.455 ->     44 : UART_RX[0]
10:22:24.455 -> ============ After Setup End =============
10:22:24.502 -> E (568) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed
10:22:24.502 -> E (569) ST1633: i2c_read_bytes(207): Read param failed
10:22:24.502 -> E (569) ST1633: read_data(131): Read advanced info failed
10:22:24.502 -> 
10:22:24.549 -> assert failed: bool ESP_PanelTouch::readRawData(uint8_t, int) C:\Users\jerry\projects\arduinov3\portable\libraries\ESP32_Display_Panel\src\touch\ESP_PanelTouch.cpp:166 ((esp_lcd_touch_read_data(handl
10:22:24.549 -> 
10:22:24.549 -> 
10:22:24.549 -> Backtrace: 0x4037742e:0x3fcebf40 0x4037c769:0x3fcebf60 0x403823b5:0x3fcebf80 0x42002467:0x3fcec0⸮

jerryliugithub, 2024-05-29:
The test found that when the SCL clock drops below 100KHz, I2C can communicate successfully. It may be caused by the long wiring of my hardware.
The following is the data communicated through setup() and loop(). I guess setup() may not have enabled the touch ic action, so there is no response when the screen is touched, but it is certain that I2C continues to act.

setup()
[image]

loop()
[image]

Lzw655, 2024-05-29:
Yes, the cable being too long can indeed affect the quality of I2C communication. 

In the setup(), only a few registers representing touch information are read, while in the loop(), the coordinates are continuously read. 

Are you now able to successfully obtain the correct coordinates?

jerryliugithub, 2024-05-29:
No, I modified loop() to confirm, but there is no response to any touch on the screen, only loop() keeps printing.

code:

void loop()
{
    Serial.println("loop()"); 
    //while(1);
    ESP_PanelTouchPoint point[EXAMPLE_TOUCH_READ_POINTS_NUM];
    int read_touch_result = touch->readPoints(point, EXAMPLE_TOUCH_READ_POINTS_NUM, -1);

    if (read_touch_result > 0) {
        for (int i = 0; i < read_touch_result; i++) {
            Serial.printf("Touch point(%d): x %d, y %d, strength %d\n", i, point[i].x, point[i].y, point[i].strength);
        }
    } else if (read_touch_result < 0) {
        Serial.println("Read touch point failed");
    }
#if EXAMPLE_TOUCH_PIN_NUM_INT < 0
    delay(30);
#endif
}

log:
          :                             :
12:14:19.727 -> I2C touch example start
12:14:19.727 -> Create I2C bus
12:14:19.727 -> D (404) ESP_PanelBus_I2C: Init host[0]
12:14:19.727 -> D (404) ESP_PanelBus_I2C: Panel IO @0x3fcece64 created
12:14:19.727 -> Create touch device
12:14:19.727 -> D (415) ESP_PanelTouch: Disable interruption
12:14:19.775 -> I (432) ST1633: Firmware version: 0(8.3.2.5), Max.X: 16, Max.Y: 256
12:14:19.775 -> I (432) ST1633: Touch panel create success, version: 0.1.0
12:14:19.775 -> I2C touch example end
12:14:19.775 -> =========== After Setup Start ============
12:14:19.775 -> INTERNAL Memory Info:
12:14:19.775 -> ------------------------------------------
12:14:19.775 ->   Total Size        :   397836 B ( 388.5 KB)
12:14:19.775 ->   Free Bytes        :   362576 B ( 354.1 KB)
12:14:19.775 ->   Allocated Bytes   :    29812 B (  29.1 KB)
12:14:19.775 ->   Minimum Free Bytes:   357528 B ( 349.1 KB)
12:14:19.775 ->   Largest Free Block:   327668 B ( 320.0 KB)
12:14:19.775 -> ------------------------------------------
12:14:19.810 -> SPIRAM Memory Info:
12:14:19.810 -> ------------------------------------------
12:14:19.810 ->   Total Size        :  8388608 B (8192.0 KB)
12:14:19.810 ->   Free Bytes        :  8386020 B (8189.5 KB)
12:14:19.810 ->   Allocated Bytes   :      260 B (   0.3 KB)
12:14:19.810 ->   Minimum Free Bytes:  8386020 B (8189.5 KB)
12:14:19.810 ->   Largest Free Block:  8257524 B (8064.0 KB)
12:14:19.810 -> ------------------------------------------
12:14:19.810 -> GPIO Info:
12:14:19.810 -> ------------------------------------------
12:14:19.844 ->   GPIO : BUS_TYPE[bus/unit][chan]
12:14:19.844 ->   --------------------------------------  
12:14:19.844 ->     43 : UART_TX[0]
12:14:19.844 ->     44 : UART_RX[0]
12:14:19.844 -> ============ After Setup End =============
12:14:19.892 -> loop()
12:14:19.927 -> loop()
12:14:19.994 -> loop()
12:14:20.078 -> loop()
12:14:20.160 -> loop()
12:14:20.228 -> loop()
12:14:20.308 -> loop()
12:14:20.379 -> loop()
12:14:20.427 -> loop()
12:14:20.511 -> loop()
12:14:20.595 -> loop()
12:14:20.642 -> loop()
12:14:20.711 -> loop()
12:14:20.794 -> loop()
12:14:20.877 -> loop()
12:14:20.961 -> loop()
12:14:20.996 -> loop()
12:14:21.078 -> loop()
12:14:21.160 -> loop()
           :                  :

Lzw655, 2024-05-29:
I reviewed the I2C data you parsed and compared it according to the protocol in the manual. I found the following two issues with the library:

1. The format for reading the maximum resolution of X and Y is incorrect and has been corrected.
2. According to the protocol in the manual, the coordinates are only valid when both the **finger** and **valid** flags are set to 1. However, based on the data parsed in the diagram, the coordinates may still be valid even if these flags are 0. Therefore, I have modified the condition to determine validity based on whether the coordinates are zero.

[image]

The library has been updated to address these two points. Please update your branch. I believe it should work correctly this time.

jerryliugithub, 2024-05-29:
Great! It can already respond to the touch position, but it doesn't respond immediately every time smoothly. It sometimes responds and sometimes doesn't respond. I wonder if it has something to do with the hardware? I will continue to verify...

In addition, the problem is that the current response can only be up to 2 points of touch? The strength is always 0

log:
15:29:14.569 -> Touch point(0): x 159, y 79, strength 0
15:29:14.569 -> Touch point(0): x 159, y 79, strength 0
15:29:14.616 -> Touch point(0): x 159, y 79, strength 0
15:29:14.663 -> Touch point(0): x 86, y 166, strength 0
15:29:14.710 -> Touch point(0): x 86, y 166, strength 0
15:29:14.758 -> Touch point(0): x 85, y 165, strength 0
15:29:14.758 -> Touch point(0): x 84, y 164, strength 0
15:29:14.804 -> Touch point(0): x 84, y 164, strength 0
15:29:14.852 -> Touch point(0): x 84, y 164, strength 0
15:29:14.899 -> Touch point(0): x 84, y 164, strength 0
15:29:14.946 -> Touch point(0): x 84, y 164, strength 0
15:29:14.946 -> Touch point(0): x 84, y 164, strength 0
15:29:14.993 -> Touch point(0): x 84, y 164, strength 0
15:29:15.041 -> Touch point(0): x 84, y 164, strength 0
15:29:15.088 -> Touch point(0): x 84, y 164, strength 0
15:29:15.135 -> Touch point(0): x 84, y 164, strength 0
15:29:15.135 -> Touch point(0): x 84, y 164, strength 0
15:29:15.182 -> Touch point(0): x 84, y 164, strength 0
15:29:15.229 -> Touch point(0): x 84, y 164, strength 0
15:29:15.276 -> Touch point(0): x 81, y 161, strength 0
15:29:15.322 -> Touch point(0): x 81, y 161, strength 0
15:29:15.369 -> Touch point(0): x 81, y 163, strength 0
15:29:15.369 -> Touch point(1): x 186, y 104, strength 0
15:29:15.369 -> Touch point(0): x 81, y 164, strength 0
15:29:15.369 -> Touch point(1): x 183, y 103, strength 0
15:29:15.417 -> Touch point(0): x 82, y 164, strength 0
15:29:15.417 -> Touch point(1): x 182, y 103, strength 0
15:29:15.463 -> Touch point(0): x 82, y 164, strength 0
15:29:15.463 -> Touch point(1): x 181, y 102, strength 0
15:29:15.511 -> Touch point(0): x 81, y 164, strength 0
15:29:15.511 -> Touch point(1): x 180, y 102, strength 0
15:29:15.558 -> Touch point(0): x 80, y 164, strength 0

Lzw655, 2024-05-29:
Because the clock frequency of I2C is low, the read time will be longer. You can improve the hardware later to increase the frequency.

Currently, it should be able to read up to [EXAMPLE_TOUCH_READ_POINTS_NUM ](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/5c44656c4d143ecf53aa5864aefe40bda30875ef/examples/Touch/I2C/I2C.ino#L59)(5). Can you capture the waveform and parsed data when there are two touch points? I want to verify if there is data for the second point. 

Additionally, the protocol mentioned in the manual does not include any strength-related data, so it might not be supported.

jerryliugithub, 2024-05-30:
Please understand my unstable target board. I just fixed it just now! This is the information I got when using the INT pin.
Once the INT pin is not used, it is very difficult.But this may also be related to my hardware.

Now is it possible to apply it in the example WiFiClock.ino?

Waveform :

[image]

[image]

log : 

::
10:09:11.265 -> loop()
10:09:11.265 -> D (437025) ESP_PanelTouch: Touch panel @0x3fced0ec touched (138, 159, 0)
10:09:11.265 -> D (437026) ESP_PanelTouch: Touch panel @0x3fced0ec touched (47, 168, 0)
10:09:11.265 -> D (437033) ESP_PanelTouch: Touch panel @0x3fced0ec touched (44, 169, 0)
10:09:11.265 -> D (437039) ESP_PanelTouch: Touch panel @0x3fced0ec touched (256, 215, 0)
10:09:11.312 -> D (43704Touch interrupt callback
10:09:11.312 -> 5) ESP_PanelTouch: Touch panel @0x3fced0ec touched (80, 54, 0)
10:09:11.312 -> ToTouch interrupt callback
10:09:11.312 -> Touch interrupt callback
10:09:11.312 -> uch point(0): x 138, y 159, strength 0
10:09:11.312 -> Touch point(1): x 47, y 168, strength 0
10:09:11.312 -> Touch point(2): x 44, y 169, strength 0Touch interrupt callback
10:09:11.312 -> 
10:09:11.312 -> Touch point(3): x 256, y 215, strength 0
10:09:11.312 -> Touch point(4): x 80, y 54, strength 0
10:09:11.312 -> loop()
10:09:11.312 -> Touch interrupt callback
10:09:11.312 -> D (437087) ESP_PanelTouch: Touch panel @0x3fced0ec touched (138, 159, 0)
10:09:11.359 -> D (437087) ESP_PanelTouch: Touch panel @0x3fced0ec touched (47, 168, 0)
10:09:11.359 -> D (437094) ESP_PanelTouch: Touch panel @0x3fced0ec touched (44, 169, 0)
10:09:11.359 -> D (437100) ESP_PanelTouch: Touch panel @0x3fced0ec touched (256, 215, 0)
10:09:11.359 -> D (437107) ESP_PanelTouch: Touch panel @0x3fced0ec touched (80, 54, 0)
10:09:11.359 -> Touch point(0): x 138, y 159, strength 0
10:09:11.359 -> Touch point(1): x 47, y 168, strength 0
10:09:11.359 -> Touch point(2): x 44, y 169, strength 0
10:09:11.359 -> Touch point(3): x 256, y 215, strength 0
10:09:11.359 -> Touch point(4): x 80, y 54, strength 0
10:09:11.359 -> loop()
10:09:11.359 -> D (437139) ESP_PanelTouch: Touch panel @0x3fced0ec touched (138, 159, 0)
10:09:11.407 -> D (437139) ESP_PanelTouch: Touch panel @0x3fced0ec touched (47, 168, 0)
10:09:11.407 -> D (437144) ESP_PanelTouch: Touch panel @0x3fced0ec touched (44, 169, 0)
10:09:11.407 -> D (437150) ESP_PanelTouch: Touch panel @0x3fced0ec touched (256, 215, 0)
10:09:11.407 -> D (437156) ESP_PanelTouch: Touch panel @0x3fced0ec touched (80, 54, 0)
10:09:11.407 -> Touch point(0): x 138, y 159, strength 0
10:09:11.407 -> Touch point(1): x 47, y 168, strength 0
10:09:11.407 -> Touch point(2): x 44, y 169, strength 0
10:09:11.407 -> Touch point(3): x 256, y 215, strength 0
10:09:11.407 -> Touch point(4): x 80, y 54, strength 0
10:09:11.407 -> loop()
::

Lzw655, 2024-05-30:
As far as I know, some touch chips, such as CST816, need to wait for the INT signal to read data. Therefore, it is possible that ST1633 operates in a similar manner. 

Of course, you can directly use it by replacing the `ESP_Panel_Board_Custom.h` file in WiFiClock.

Lzw655, 2024-05-30:
So, may I merge the [PR](https://github.com/esp-arduino-libs/ESP32_Display_Panel/pull/33) if you think the ST1633 driver is ok?

jerryliugithub, 2024-05-30:
Finally touch control can be implemented in the WiFiClock example. I think the ST1633 driver is ok. Thanks for your help with the lcd driver and touch driver and the speed of processing. What a great library !

Lzw655, 2024-05-30:
Congrats! Thanks to your usage and demands, it will continue to get better. I will also keep optimizing it.

closed 2024-05-30 by jerryliugithub