esp-arduino-libs/ESP32_Display_Panel #24: SquareLine Porting Example --> error: 'ESP_Panel' does not name a type

lboue, 2024-05-12:
Hello,

I try to compile this demo file [Porting.ino](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/examples/SquareLine/v8/Porting/Porting.ino) but I get a compilation error in the Arduino IDE: `error: 'ESP_Panel' does not name a type`.

Can you help me correct this error?

### Step to reproduce
* Copy this Githup repo
* Edit `ESP_Panel_Board_Custom.h`
* Build demo file [Porting.ino](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/examples/SquareLine/v8/Porting/Porting.ino)

Lzw655, 2024-05-16:
Hi @lboue, did you set the macro `ESP_PANEL_USE_CUSTOM_BOARD` to `1` in the file `ESP_Panel_Board_Custom.h`?

lboue, 2024-05-17:
> Hi @lboue, did you set the macro `ESP_PANEL_USE_CUSTOM_BOARD` to `1` in the file `ESP_Panel_Board_Custom.h`?

Thanks. It build now.

closed 2024-05-17 by lboue