esp-arduino-libs/ESP32_Display_Panel #26: PanelTest.ino --> Compilation error: 'ESP_Panel' does not name a type; did you mean 'ESP_PanelLcd'?

lboue, 2024-05-17:
Hello,

I am trying to build [PanelTest.ino](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/examples/Panel/PanelTest/PanelTest.ino) but I am facing this error. Can you help me to fix it?

```
Generating function prototypes...
C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-12.2.0_20230208/bin/xtensa-esp32s3-elf-g++ -c @C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32s3/flags/cpp_flags -w -Os -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32S3_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32S3_DEV" -DARDUINO_VARIANT="esp32s3" -DARDUINO_PARTITION_default_8MB -DARDUINO_HOST_OS="windows" -DARDUINO_FQBN="esp32:esp32:esp32s3:UploadSpeed=921600,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,CPUFreq=240,FlashMode=qio,FlashSize=8M,PartitionScheme=default,DebugLevel=none,PSRAM=opi,LoopCore=1,EventsCore=1,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DBOARD_HAS_PSRAM -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=0 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 @C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32s3/flags/defines -IC:\Users\user\AppData\Local\Temp\.arduinoIDE-unsaved2024417-368-1odpf7z.43k\PanelTest -iprefix C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32s3/include/ @C:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32s3/flags/includes -IC:\Users\user\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083/esp32s3/qio_opi/include -IC:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-rc3\cores\esp32 -IC:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-rc3\variants\esp32s3 -Ic:\Users\user\Documents\Arduino\libraries\ESP32_Display_Panel\src -Ic:\Users\user\Documents\Arduino\libraries\ESP32_IO_Expander\src @C:\Users\user\AppData\Local\Temp\arduino\sketches\9AB4D0BAFE0CD43D82EA97D153E60905/build_opt.h @C:\Users\user\AppData\Local\Temp\arduino\sketches\9AB4D0BAFE0CD43D82EA97D153E60905/file_opts C:\Users\user\AppData\Local\Temp\arduino\sketches\9AB4D0BAFE0CD43D82EA97D153E60905\sketch\PanelTest.ino.cpp -o C:\Users\user\AppData\Local\Temp\3035903189\sketch_merged.cpp
C:\Users\user\AppData\Local\Arduino15\packages\builtin\tools\ctags\5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives C:\Users\user\AppData\Local\Temp\3035903189\sketch_merged.cpp
Compiling sketch...
"C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32s3-elf-gcc\\esp-12.2.0_20230208/bin/xtensa-esp32s3-elf-g++" -MMD -c "@C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083/esp32s3/flags/cpp_flags" -w -Os -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32S3_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32S3_DEV\"" "-DARDUINO_VARIANT=\"esp32s3\"" -DARDUINO_PARTITION_default_8MB "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32s3:UploadSpeed=921600,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,CPUFreq=240,FlashMode=qio,FlashSize=8M,PartitionScheme=default,DebugLevel=none,PSRAM=opi,LoopCore=1,EventsCore=1,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DBOARD_HAS_PSRAM -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=0 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 "@C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083/esp32s3/flags/defines" "-IC:\\Users\\user\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2024417-368-1odpf7z.43k\\PanelTest" -iprefix "C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083/esp32s3/include/" "@C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083/esp32s3/flags/includes" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083/esp32s3/qio_opi/include" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-rc3\\cores\\esp32" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-rc3\\variants\\esp32s3" "-Ic:\\Users\\user\\Documents\\Arduino\\libraries\\ESP32_Display_Panel\\src" "-Ic:\\Users\\user\\Documents\\Arduino\\libraries\\ESP32_IO_Expander\\src" "@C:\\Users\\user\\AppData\\Local\\Temp\\arduino\\sketches\\9AB4D0BAFE0CD43D82EA97D153E60905/build_opt.h" "@C:\\Users\\user\\AppData\\Local\\Temp\\arduino\\sketches\\9AB4D0BAFE0CD43D82EA97D153E60905/file_opts" "C:\\Users\\user\\AppData\\Local\\Temp\\arduino\\sketches\\9AB4D0BAFE0CD43D82EA97D153E60905\\sketch\\PanelTest.ino.cpp" -o "C:\\Users\\user\\AppData\\Local\\Temp\\arduino\\sketches\\9AB4D0BAFE0CD43D82EA97D153E60905\\sketch\\PanelTest.ino.cpp.o"
C:\Users\user\AppData\Local\Temp\.arduinoIDE-unsaved2024417-368-1odpf7z.43k\PanelTest\PanelTest.ino:46:1: error: 'ESP_Panel' does not name a type; did you mean 'ESP_PanelLcd'?
   46 | ESP_Panel *panel = nullptr;
      | ^~~~~~~~~
      | ESP_PanelLcd
C:\Users\user\AppData\Local\Temp\.arduinoIDE-unsaved2024417-368-1odpf7z.43k\PanelTest\PanelTest.ino: In function 'void setup()':
C:\Users\user\AppData\Local\Temp\.arduinoIDE-unsaved2024417-368-1odpf7z.43k\PanelTest\PanelTest.ino:74:5: error: 'panel' was not declared in this scope
   74 |     panel = new ESP_Panel();
      |     ^~~~~
C:\Users\user\AppData\Local\Temp\.arduinoIDE-unsaved2024417-368-1odpf7z.43k\PanelTest\PanelTest.ino:74:17: error: expected type-specifier before 'ESP_Panel'
   74 |     panel = new ESP_Panel();
      |                 ^~~~~~~~~

Using library ESP32_Display_Panel at version 0.1.1 in folder: C:\Users\user\Documents\Arduino\libraries\ESP32_Display_Panel 
Using library ESP32_IO_Expander at version 0.0.3 in folder: C:\Users\user\Documents\Arduino\libraries\ESP32_IO_Expander 
exit status 1

Compilation error: 'ESP_Panel' does not name a type; did you mean 'ESP_PanelLcd'?
```

**Build log**
[PanelTest_build.log](https://github.com/esp-arduino-libs/ESP32_Display_Panel/files/15344185/PanelTest_build.log)

lboue, 2024-05-17:
Setting the macro `ESP_PANEL_USE_CUSTOM_BOARD `to 1 in the file ESP_Panel_Board_Custom.h fix it.
I opened a PR #27 to fix broken links.

Lzw655, 2024-05-17:
Great. Do you still need this issue?

lboue, 2024-05-17:
No. I close it.

closed 2024-05-17 by lboue

wavejaco, 2024-06-20:
Sorry, I realise this issue is closed, but I am experiencing the exact same issue that @lboue had, although with ESP32_Display_Panel at version 0.1.4 and ESP32_IO_Expander at version 0.0.2.

Setting the macro `ESP_PANEL_USE_CUSTOM_BOARD` to 1 in the file _ESP_Panel_Board_Custom.h_ didn't fix the problem.

What is the root cause of this compilation error?