esp-arduino-libs/ESP32_Display_Panel #28: `'ST7262_PANEL_IO_3WIRE_SPI_CONFIG' was not declared in this scope` using an RGB screen

lboue, 2024-05-17:
Hello,

I have a [ST7262 ](https://www.orientdisplay.com/pdf/ST7262.pdf) LCD screen and I am trying to build an example but I am facing another error:
`'ST7262_PANEL_IO_3WIRE_SPI_CONFIG' was not declared in this scope`
I don't understand why I'm getting this error with an SPI bus when I'm using an RGB screen.

```c++
c:\Users\user\Documents\Arduino\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp: In member function 'bool ESP_Panel::init()':
C:\Users\user\Documents\Arduino\Waveshare\PanelTest_LBO\PanelTest_LBO/ESP_Panel_Board_Custom.h:32:37: error: 'ST7262_PANEL_IO_3WIRE_SPI_CONFIG' was not declared in this scope; did you mean 'ST7701_PANEL_IO_3WIRE_SPI_CONFIG'?
   32 | #define ESP_PANEL_LCD_NAME          ST7262
      |                                     ^~~~~~
c:\Users\user\Documents\Arduino\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:40:57: note: in definition of macro '_LCD_PANEL_IO_3WIRE_SPI_CONFIG'
   40 |                                                         name##_PANEL_IO_3WIRE_SPI_CONFIG(line_config, scl_active_edge)
      |                                                         ^~~~
c:\Users\user\Documents\Arduino\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:213:9: note: in expansion of macro 'LCD_PANEL_IO_3WIRE_SPI_CONFIG'
  213 |         LCD_PANEL_IO_3WIRE_SPI_CONFIG(ESP_PANEL_LCD_NAME, line_config, ESP_PANEL_LCD_3WIRE_SPI_SCL_ACTIVE_EDGE);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\user\Documents\Arduino\libraries\ESP32_Display_Panel\src\ESP_Panel.cpp:213:39: note: in expansion of macro 'ESP_PANEL_LCD_NAME'
  213 |         LCD_PANEL_IO_3WIRE_SPI_CONFIG(ESP_PANEL_LCD_NAME, line_config, ESP_PANEL_LCD_3WIRE_SPI_SCL_ACTIVE_EDGE);
      |                                       ^~~~~~~~~~~~~~~~~~
"C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32s3-elf-gcc\\esp-12.2.0_20230208/bin/xtensa-esp32s3-elf-gcc" -MMD -c "@C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083/esp32s3/flags/c_flags" -w -Os -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32S3_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32S3_DEV\"" "-DARDUINO_VARIANT=\"esp32s3\"" -DARDUINO_PARTITION_default_8MB "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32s3:UploadSpeed=921600,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,CPUFreq=240,FlashMode=qio,FlashSize=8M,PartitionScheme=default,DebugLevel=none,PSRAM=opi,LoopCore=1,EventsCore=1,EraseFlash=none,JTAGAdapter=default,ZigbeeMode=default\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DBOARD_HAS_PSRAM -DARDUINO_USB_MODE=1 -DARDUINO_USB_CDC_ON_BOOT=0 -DARDUINO_USB_MSC_ON_BOOT=0 -DARDUINO_USB_DFU_ON_BOOT=0 "@C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083/esp32s3/flags/defines" "-IC:\\Users\\user\\Documents\\Arduino\\Waveshare\\PanelTest_LBO\\PanelTest_LBO" -iprefix "C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083/esp32s3/include/" "@C:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083/esp32s3/flags/includes" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-442a798083/esp32s3/qio_opi/include" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-rc3\\cores\\esp32" "-IC:\\Users\\user\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-rc3\\variants\\esp32s3" "-Ic:\\Users\\user\\Documents\\Arduino\\libraries\\ESP32_Display_Panel\\src" "-Ic:\\Users\\user\\Documents\\Arduino\\libraries\\ESP32_IO_Expander\\src" "@C:\\Users\\user\\AppData\\Local\\Temp\\arduino\\sketches\\CBB42693422360DC1E6A7709D194CD0D/build_opt.h" "@C:\\Users\\user\\AppData\\Local\\Temp\\arduino\\sketches\\CBB42693422360DC1E6A7709D194CD0D/file_opts" "c:\\Users\\user\\Documents\\Arduino\\libraries\\ESP32_Display_Panel\\src\\lcd\\base\\esp_lcd_gc9503.c" -o "C:\\Users\\user\\AppData\\Local\\Temp\\arduino\\sketches\\CBB42693422360DC1E6A7709D194CD0D\\libraries\\ESP32_Display_Panel\\lcd\\base\\esp_lcd_gc9503.c.o"

Using library ESP32_Display_Panel at version 0.1.1 in folder: C:\Users\user\Documents\Arduino\libraries\ESP32_Display_Panel 
Using library ESP32_IO_Expander at version 0.0.3 in folder: C:\Users\user\Documents\Arduino\libraries\ESP32_IO_Expander 
exit status 1

Compilation error: 'ST7262_PANEL_IO_3WIRE_SPI_CONFIG' was not declared in this scope; did you mean 'ST7701_PANEL_IO_3WIRE_SPI_CONFIG'?
```

**ESP_Panel_Board_Custom.h config** 

1. I have a ST7262 LCD screen so I set ESP_PANEL_LCD_NAME to ST7262
2. I set ESP_PANEL_USE_LCD to 1
 
```c++

#if ESP_PANEL_USE_CUSTOM_BOARD

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
//////////////////////////// Please update the following macros to configure the LCD panel /////////////////////////////
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
/* Set to 1 when using an LCD panel */
#define ESP_PANEL_USE_LCD           (1)     // 0/1

#if ESP_PANEL_USE_LCD
/**
 * LCD Controller Name. Choose one of the following:
 *      - GC9A01, GC9B71, GC9503
 *      - ILI9341
 *      - NV3022B
 *      - SH8601
 *      - SPD2010
 *      - ST7262, ST7701, ST7789, ST7796, ST77916, ST77922
 */
#define ESP_PANEL_LCD_NAME          ST7262

/* LCD resolution in pixels */
#define ESP_PANEL_LCD_WIDTH         (800)
#define ESP_PANEL_LCD_HEIGHT        (480)

/* LCD Bus Settings */
/**
 * If set to 1, the bus will skip to initialize the corresponding host. Users need to initialize the host in advance.
 * It is useful if other devices use the same host. Please ensure that the host is initialized only once.
 */
#define ESP_PANEL_LCD_BUS_SKIP_INIT_HOST    (0)     // 0/1
/**
 * LCD Bus Type. Choose one of the following:
 *      - ESP_PANEL_BUS_TYPE_I2C (not ready)
 *      - ESP_PANEL_BUS_TYPE_SPI
[... 81 lines cut ...]
    #define ESP_PANEL_LCD_RGB_IO_DATA11         (0)
    #define ESP_PANEL_LCD_RGB_IO_DATA12         (45)
    #define ESP_PANEL_LCD_RGB_IO_DATA13         (48)
    #define ESP_PANEL_LCD_RGB_IO_DATA14         (47)
    #define ESP_PANEL_LCD_RGB_IO_DATA15         (21)
#endif
#if !ESP_PANEL_LCD_BUS_SKIP_INIT_HOST
    #define ESP_PANEL_LCD_3WIRE_SPI_IO_CS               (0)
    #define ESP_PANEL_LCD_3WIRE_SPI_IO_SCK              (1)
    #define ESP_PANEL_LCD_3WIRE_SPI_IO_SDA              (2)
    #define ESP_PANEL_LCD_3WIRE_SPI_CS_USE_EXPNADER     (0)     // 0/1
    #define ESP_PANEL_LCD_3WIRE_SPI_SCL_USE_EXPNADER    (0)     // 0/1
    #define ESP_PANEL_LCD_3WIRE_SPI_SDA_USE_EXPNADER    (0)     // 0/1
    #define ESP_PANEL_LCD_3WIRE_SPI_SCL_ACTIVE_EDGE     (0)     // 0: rising edge, 1: falling edge
    #define ESP_PANEL_LCD_FLAGS_AUTO_DEL_PANEL_IO       (0)     // Delete the panel IO instance automatically if set to 1.
                                                                // If the 3-wire SPI pins are sharing other pins of the RGB interface to save GPIOs,
                                                                // Please set it to 1 to release the panel IO and its pins (except CS signal).
    #define ESP_PANEL_LCD_FLAGS_MIRROR_BY_CMD           (!ESP_PANEL_LCD_FLAGS_AUTO_DEL_PANEL_IO)
                                                                // The `mirror()` function will be implemented by LCD command if set to 1.
#endif
```

Did I miss something?

Regards

Lzw655, 2024-05-17:
Hi @lboue, please set the macro `ESP_PANEL_LCD_BUS_SKIP_INIT_HOST` to `1` if you are not using 3-wire spi.

lboue, 2024-05-17:
> `ESP_PANEL_LCD_BUS_SKIP_INIT_HOST`

@Lzw655 Thank you for your time. Build is OK now.
Where can I find this information that you don't have to initialise it for an RGB screen? I haven't seen it.

Lzw655, 2024-05-17:
Sorry, it's a missing information. And I added it in the new [PR](https://github.com/esp-arduino-libs/ESP32_Display_Panel/pull/29). Please take a look. Thank you!

lboue, 2024-05-17:
It's perfect. Thanks for the quick response!

closed 2024-05-17 by lboue

Lzw655, 2024-05-17:
You're welcome. Thank you for your contribution!