esp-arduino-libs/ESP32_Display_Panel #30: esp_lcd_touch_xpt2046.h:25:50: error: 'ESP_PANEL_TOUCH_XPT2046_Z_THRESHOLD' undeclared (first use in this function)

lboue, 2024-05-17:
Hello,

I am trying to build the [I2C Touch Example](https://github.com/esp-arduino-libs/ESP32_Display_Panel/tree/master/examples/Touch/I2C) for GT911 and I am getting this error:
```
In file included from c:\Users\Ludovic\Documents\Arduino\libraries\ESP32_Display_Panel\src\touch\base\esp_lcd_touch_xpt2046.c:25:
c:\Users\Ludovic\Documents\Arduino\libraries\ESP32_Display_Panel\src\touch\base\esp_lcd_touch_xpt2046.c: In function 'xpt2046_read_data':
c:\Users\Ludovic\Documents\Arduino\libraries\ESP32_Display_Panel\src\touch\base\esp_lcd_touch_xpt2046.h:25:50: error: 'ESP_PANEL_TOUCH_XPT2046_Z_THRESHOLD' undeclared (first use in this function)
   25 | #define CONFIG_XPT2046_Z_THRESHOLD              (ESP_PANEL_TOUCH_XPT2046_Z_THRESHOLD)
      |                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Ludovic\Documents\Arduino\libraries\ESP32_Display_Panel\src\touch\base\esp_lcd_touch_xpt2046.h:25:50: note: in definition of macro 'CONFIG_XPT2046_Z_THRESHOLD'
   25 | #define CONFIG_XPT2046_Z_THRESHOLD              (ESP_PANEL_TOUCH_XPT2046_Z_THRESHOLD)
      |                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Ludovic\Documents\Arduino\libraries\ESP32_Display_Panel\src\touch\base\esp_lcd_touch_xpt2046.h:25:50: note: each undeclared identifier is reported only once for each function it appears in
   25 | #define CONFIG_XPT2046_Z_THRESHOLD              (ESP_PANEL_TOUCH_XPT2046_Z_THRESHOLD)
      |                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\Ludovic\Documents\Arduino\libraries\ESP32_Display_Panel\src\touch\base\esp_lcd_touch_xpt2046.h:25:50: note: in definition of macro 'CONFIG_XPT2046_Z_THRESHOLD'
   25 | #define CONFIG_XPT2046_Z_THRESHOLD              (ESP_PANEL_TOUCH_XPT2046_Z_THRESHOLD)
      |                                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Using library ESP32_Display_Panel at version 0.1.1 in folder: C:\Users\Ludovic\Documents\Arduino\libraries\ESP32_Display_Panel 
Using library ESP32_IO_Expander at version 0.0.3 in folder: C:\Users\Ludovic\Documents\Arduino\libraries\ESP32_IO_Expander 
exit status 1
```

**Config**
```c++
#define EXAMPLE_TOUCH_NAME              GT911
#define EXAMPLE_TOUCH_WIDTH             (480)
#define EXAMPLE_TOUCH_HEIGHT            (480)
#define EXAMPLE_TOUCH_I2C_FREQ_HZ       (400 * 1000)
#define EXAMPLE_TOUCH_READ_POINTS_NUM   (5)

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
//////////////////// Please update the following configuration according to your board spec ////////////////////////////
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
#define EXAMPLE_TOUCH_PIN_NUM_I2C_SCL   (9)
#define EXAMPLE_TOUCH_PIN_NUM_I2C_SDA   (8)
#define EXAMPLE_TOUCH_PIN_NUM_RST       (-1)
#define EXAMPLE_TOUCH_PIN_NUM_INT       (-1)
```

Could you help me to fix it?

Lzw655, 2024-05-18:
Hi @lboue, to solve this issue, you can use the latest `ESP_Panel_Conf.h` file from the library. The latest version includes the xpt2046 driver, and it requires some configurations from `ESP_Panel_Conf.h`. 

However, incompatibility with the old version of the file is indeed a problem. I will fix this as soon as possible. Thank you for pointing out!

lboue, 2024-05-18:
Thanks. I'll give it a try.

closed 2024-05-18 by lboue