esp-arduino-libs/ESP32_Display_Panel #31: Issue Erupted with update 2 days back

hrituraj-rohariya, 2024-05-21:
hello there,
Great lib incorporating multiple LCD and touch drivers...Great work

kindly help out in my issue.
i am trying to run LVGL squareline ui on a Waveshare share 4.3 inch RGB display,GT911 touch drivers running on  Esp32S3 SOC
(ram:8,flash:8) using platform io IDE.

i am using a custom board and have necessary includes in code as far i could figure out. 

Found Success Running the code with tearing down effect for RGB panel with previous version of library.

with latest update pio compiler fails to compile with error codes described below

kindly help...

necessary details included below..

thanks in advance. 

#----------------------------------------Here is My LVGL Task code--------------------------------------------------------------------------

#include <Arduino.h>
#include "ESP_Panel_Library.h"
#include <lvgl.h>
#include <ui.h>
#include "lvgl_port_v8.h"
#include <ESP_Panel_Conf.h>
#include <ESP_Panel_Board_Custom.h>
#define LVGL_TASK_MAX_DELAY_MS (500)
#define LVGL_TASK_MIN_DELAY_MS (1)

void LVtask(void *arg)
{
       Serial.println("lvgl porting example start");

       Serial.println("Initialize panel device");
       ESP_Panel *panel = new ESP_Panel();
       panel->init();
#if LVGL_PORT_AVOID_TEAR
       // When avoid tearing function is enabled, configure the RGB bus according to the LVGL configuration
       ESP_PanelBus_RGB *rgb_bus = static_cast<ESP_PanelBus_RGB *>(panel->getLcd()->getBus());
       rgb_bus->configRgbFrameBufferNumber(LVGL_PORT_DISP_BUFFER_NUM);
       rgb_bus->configRgbBounceBufferSize(LVGL_PORT_RGB_BOUNCE_BUFFER_SIZE);
#endif
       panel->begin();

       Serial.println("Initialize LVGL");
       lvgl_port_init(panel->getLcd(), panel->getTouch());

       Serial.println("Create UI");
       /* Lock the mutex due to the LVGL APIs are not thread-safe */
       lvgl_port_lock(-1);

       /* Initialize LVGL UI generated by Squareline */
       ui_init();

       /* Release the mutex */
       lvgl_port_unlock();

       Serial.println("Squareline porting example end");
       uint32_t task_delay_ms = LVGL_TASK_MAX_DELAY_MS;
       while (1)
       {

              // Lock the mutex due to the LVGL APIs are not thread-safe
              lvgl_port_lock(-1);
              task_delay_ms = lv_timer_handler();
              // Release the mutex
              lvgl_port_unlock();
              if (task_delay_ms > LVGL_TASK_MAX_DELAY_MS)
[... 481 lines cut ...]
     ^
.pio/libdeps/waveshare4_3inch/ESP32_Display_Panel/src/bus/QSPI.cpp:24:46: note: in expansion of macro 'ESP_PANEL_IO_QSPI_CONFIG_DEFAULT'
     io_config((esp_lcd_panel_io_spi_config_t)ESP_PANEL_IO_QSPI_CONFIG_DEFAULT(cs_io))
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\waveshare4_3inch\lib2ce\ESP32_Display_Panel\bus\QSPI.cpp.o] Error 1
In file included from .pio/libdeps/waveshare4_3inch/ESP32_Display_Panel/src/bus/SPI.cpp:12:
.pio/libdeps/waveshare4_3inch/ESP32_Display_Panel/src/bus/SPI.cpp: In constructor 'ESP_PanelBus_SPI::ESP_PanelBus_SPI(int, int, int, int, int)':
.pio/libdeps/waveshare4_3inch/ESP32_Display_Panel/src/bus/SPI.h:36:5: error: 'esp_lcd_panel_io_spi_config_t::<unnamed struct>' has no non-static data member named 'quad_mode'     
     }
     ^
.pio/libdeps/waveshare4_3inch/ESP32_Display_Panel/src/bus/SPI.cpp:19:46: note: in expansion of macro 'ESP_PANEL_IO_SPI_CONFIG_DEFAULT'
     io_config((esp_lcd_panel_io_spi_config_t)ESP_PANEL_IO_SPI_CONFIG_DEFAULT(cs_io, dc_io))
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/waveshare4_3inch/ESP32_Display_Panel/src/bus/SPI.cpp: In constructor 'ESP_PanelBus_SPI::ESP_PanelBus_SPI(int, int)':
.pio/libdeps/waveshare4_3inch/ESP32_Display_Panel/src/bus/SPI.h:36:5: error: 'esp_lcd_panel_io_spi_config_t::<unnamed struct>' has no non-static data member named 'quad_mode'     
     }
     ^
.pio/libdeps/waveshare4_3inch/ESP32_Display_Panel/src/bus/SPI.cpp:32:46: note: in expansion of macro 'ESP_PANEL_IO_SPI_CONFIG_DEFAULT'
     io_config((esp_lcd_panel_io_spi_config_t)ESP_PANEL_IO_SPI_CONFIG_DEFAULT(cs_io, dc_io))
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\waveshare4_3inch\lib2ce\ESP32_Display_Panel\bus\SPI.cpp.o] Error 1
========================================================================== [FAILED] Took 147.32 seconds ==========================================================================

 *  The terminal process "C:\Users\hritu\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

lboue, 2024-05-21:
@hrituraj-rohariya 
Could you attach a log file instead. Here it's unreadable.

hrituraj-rohariya, 2024-05-21:
[file.log](https://github.com/esp-arduino-libs/ESP32_Display_Panel/files/15389860/file.log)

thanks for promt reply here is the log file if it helps

hrituraj-rohariya, 2024-05-21:
I Have previous version of Library too.. but was not able to use Bounce Buffer to avoid Tearing down Of screen(in my case RGB screen)

have replaced the Esp32 SDK in /user/appdata/arduino15/esp32...

but yet configRgbFrameBufferNumber,  configRgbBounceBufferSize functions were not defined in the RGB.cpp file of previous library thus could not use Avoid_TearDown macro...

kindly help..

Thanks in Advance

lboue, 2024-05-21:
Can you share your ESP_Panel_Conf.h file in attachment?

hrituraj-rohariya, 2024-05-21:
Sure...
[main.txt](https://github.com/esp-arduino-libs/ESP32_Display_Panel/files/15392084/main.txt)
[platform.ini.txt](https://github.com/esp-arduino-libs/ESP32_Display_Panel/files/15392085/platform.ini.txt)

find them attached below..

I have included platform.ini as well as main as text too... hope this resolves the issue

Thanks

[DEVICE_LVGL_V8.txt](https://github.com/esp-arduino-libs/ESP32_Display_Panel/files/15392021/DEVICE_LVGL_V8.txt)
[ESP_Panel_Board_Custom.txt](https://github.com/esp-arduino-libs/ESP32_Display_Panel/files/15392022/ESP_Panel_Board_Custom.txt)
[ESP_Panel_Conf.txt](https://github.com/esp-arduino-libs/ESP32_Display_Panel/files/15392023/ESP_Panel_Conf.txt)

Lzw655, 2024-05-22:
Hi @hrituraj-rohariya, 

Sorry, I haven't tested this library on PlatformIO yet, but I checked the old version library you provided, and it works with `arduino-esp32 v2.x`. The latest version of the library only supports `arduino-esp32 v3.x`. You can see the dependencies [here](https://github.com/esp-arduino-libs/ESP32_Display_Panel?tab=readme-ov-file#dependencies-and-versions). 

Could you check the version of `arduino-esp32` you are currently using?

hrituraj-rohariya, 2024-05-22:
Hi @Lzw655 and Hello @lboue ..

Thanks A Lot for your Concern.. 

It took a little long to figure out IDF version.. 
 It was v4.4.3

hope esp32 sdk version is same as Idf version..

[image]

As far as Instruction regarding the library usage provided in readme file have be followed by me...

have replaced the necessary sdk in ./user/appData/arduino15/esp32/..
[image]

But was Unable to use Bounce Buffer in Previous version of library. RGB.cpp Has no defines for configRgbFrameBufferNumber, configRgbBounceBufferSize  Functions .

Did some more research regarding the same and found few changes in migration guide of esp32 from version2.x to version 3.x
could these be possible causes
[image]
  

If still no Solution...Kindly advice how either update my esp32 SDK to desired version...or to use Bounce Buffer values and Avoid_TearDown macro

Thanks in Advance

hrituraj-rohariya, 2024-05-22:
Secondly....

i will feel lucky, If i could be any Kind of help in Testing and formulating the Platformio Version of Library, in any possible way.

Thanks

Lzw655, 2024-05-22:
Hi @hrituraj-rohariya,

Thank you for your detailed debugging and analysis. I will also investigate as soon as possible whether PlatformIO can support esp32-arduino v3.x and try using this library.

Lzw655, 2024-05-22:
Hi @lboue, do you have experience using arduino-esp32 on PlatformIO?

lboue, 2024-05-22:
> Hi @lboue, do you have experience using arduino-esp32 on PlatformIO?

No, I've never done that.

hrituraj-rohariya, 2024-05-22:
Hi @Lzw655 and Hello @lboue 

Thank you both for your efforts...

Did some more research and found out that Platformio is yet to support esp32 version 3.x..

[image]

This Discussion has Valued information about Platformio and its future with Expressif..as far as this Discussion states...PIO would either wait till stable version of ESP32 Version 3.0x.. comes or may not support newer SDK due to financial matters between Expressif and PIO..

 
Discussion link:
https://github.com/platformio/platform-espressif32/issues/1225

With this Info.. Possibility  of making library work on Platformio finishes for Present...

I am new to Expressif Idf  and Arduino Ide is not my prefered choice Usually..

what would you suggest between both...

How to use our library in Expressif IDF...

Any valuable links you People could suggest...

Thanks a lot

Lzw655, 2024-05-23:
Thank you for your research. Compared to IDF, the Arduino IDE is simpler to set up and more convenient for development. However, it is relatively less flexible and has lower performance. For mass production, we recommend using IDF for development.

The design goal of this library is to be compatible with both IDF and Arduino IDE. As you can see, we have a project [test_apps ](https://github.com/esp-arduino-libs/ESP32_Display_Panel/tree/v0.0.2/test_apps) in the older version. It is based on IDF and used to test functions. 

However, due to the incomplete Kconfig in the latest version, it is not yet convenient to use it in IDF. I can complete it this week and provide an example for you.

hrituraj-rohariya, 2024-05-23:
Thank you very very much for everything....

Would eagerly waiting for Expressif IDF codes...

Hope to see this library replacing TFT_espi lovyanGFX and all other Graphic Driver libraries...

Regards

hrituraj-rohariya, 2024-05-23:
Thanks you very much for Everything..

Would be eagerly waiting for Supported Expressif IDF codes

Hope to very soon see this Library replaceing Tft_espi lovyanGfx and other Graphics Driver Libraries atleast

Good luck and My Best wishes...

Thanks and waiting...

Regards

Lzw655, 2024-05-23:
Thank you for your high praise and expectations. We will strive to further improve this library and hope you can provide more valuable suggestions. 😁

hrituraj-rohariya, 2024-05-26:
Hello @Lzw655  and hi@lboue 

Hope you are doing Great..

Saturday Night was busy I Guess... I saw Your New Merge to Main Branch...

Sorry But i was not able to Compile with PlatformIO

in Between...Did Some researches to find out that platform can be changed in platform.ini file present as central spine for all PlatformIo projects(Counter part of sdkconfig in idf)

In .ini file the platform tag has to be updated as per need

[image]

 More Information regarding various releases can be found here
Link: 
https://github.com/platformio/platform-espressif32/releases

As per my Understanding PlatformIO espressif32 6.4.0 supports espressif32 Idf version 5.1.1
[image]

and PlatformIO espressif32 6.6.0 supports espressif32 Idf version 5.2.1
[image]

And the latest Supported is PlatformIO espressif32 6.7.0 
[image]

Well Our library Depends on IDF 5.1.1 if i am not wrong....

here are log files for PlatformIO espressif32 6.4.0  and 6.6.0 compile logs..

[Uploading TESTLOG_Platform_esp32@6_4_0_IDF_5_1_1.log…]()
[TESTLOG_Platform_esp32@6_6_0_IDF_5_2_1.log](https://github.com/esp-arduino-libs/ESP32_Display_Panel/files/15447940/TESTLOG_Platform_esp32%406_6_0_IDF_5_2_1.log)

Hope Some solution is found through logs

But gradually PlatmormIo is loosing "Dependable" Factor....

Can you guide me how to use bounce buffer in arduinoide...

and i expect Solution only with Expressif Ide

Eagerly waiting for Expressif Idf example...

Thanks in Advance

Lzw655, 2024-05-27:
Hi @hrituraj-rohariya, 

Thank you for your detailed research. Based on the LOG file you provided, I found the following line:
 
```
Processing waveshare4_3inch (platform: espressif32 @ 6.6.0; board: esp32-s3-devkitc-1; framework: arduino)
```

This indicates that you are using `arduino` as the framework. However, as mentioned in the image you provided, the current PlatformIO only supports Arduino - 2.0.x versions, so the ESP32_Display_Panel library cannot be compiled successfully. Once I have improved this library to support IDF, you can try using `espidf` as the framework.

To use the bounce buffer in Arduino IDE, please refer to the [README](https://github.com/esp-arduino-libs/ESP32_Display_Panel?tab=readme-ov-file#how-to-fix-screen-drift-issue-when-driving-rgb-lcd-with-esp32-s3).

hrituraj-rohariya, 2024-05-27:
Thanks You Very Much for All your efforts...

By Profession i am A Surgeon, Doctor...

We "Medico's" are good at Research.... 

Hope it helps someone, Some where...

I would close this issue at present as PlatformIO or to be specific Arduino Wrapper limits the Possibility of Usage for at least RGB Display 3wire SPI and data pins(15)...

Espressif IDF remains the production level Solution as per your Guidance above

 Let me Get back to you with my success/Unsuccessful trails on IDF by the time Library is Efficient as per your Standards

Pleasure Having conversation 

Thanks

closed 2024-05-27 by hrituraj-rohariya

Lzw655, 2024-05-28:
Hi @hrituraj-rohariya, @isthaison said he has successfully compiled this library on PlatformIO in [issue](https://github.com/esp-arduino-libs/ESP32_Display_Panel/pull/34#issuecomment-2134762567). Could you refer to [his modifications](https://github.com/esp-arduino-libs/ESP32_Display_Panel/issues/19#issuecomment-2134217055) and give it a try?