esp-arduino-libs/ESP32_Display_Panel #36: Can not find ESP_IOExander.h?

jerrylong-km, 2024-05-27:
I compiled 3wireSPI_RGB example, but reported "Can not find ESP_IOExpander.h" error, I checked the library files, there is no ESP_IOExpander.h file, should I delete this line in RGB.h?
[image]

Lzw655, 2024-05-27:
Hi @jerrylong-km, this library relies on another library [ESP32_IO_Expander](https://github.com/esp-arduino-libs/ESP32_IO_Expander). You can check the information [here](https://github.com/esp-arduino-libs/ESP32_Display_Panel#dependencies-and-versions) and should install it instead of changing the source code.

karllambrechts, 2025-01-24:
Hi,
I have the same problem: 

fatal error: ESP_IOExpander.h: No such file or directory
   13 | #include "ESP_IOExpander.h"

The file ESP_IOExpander.h is nowhere in the directories or subdirectories. However the library ESP32_IO_Expander (note the extra "_") is in the list under "libraries". I checked the page indicated by Lzw655 ([here: (https://github.com/esp-arduino-libs/ESP32_Display_Panel#dependencies-and-versions)), but i did not find the relevant information. Where can I find?
(Greatly appreciated because I burned already many many hours on this issue - afraid of asking stupid questions ;-) )

Lzw655, 2025-01-24:
Hi @karllambrechts,

No worries, please feel free to ask questions. I'm sorry you encountered this issue. I think you are using the latest ESP32_IO_Expander v1.x version, which is incompatible with ESP32_Display_Panel v0.x. You need to reinstall it to the v0.x version.

karllambrechts, 2025-01-24:
Hi,

Indeed, following  versions were installed: ESP32_Display_Panel v0.2.2, and ESP32_IO_Expander v1.0.1. 
I changed to ESP32_IO_Expander v0.1.0. 
The former error :
          fatal error: ESP_IOExpander.h: No such file or directory
          13 | #include "ESP_IOExpander.h"
was no longer there, but unfortunately this gives: (3x)
          error 'ESP_Panel' does not name a type; did you mean 'ESP_PanelLcd'?
          57 | ESP_Panel *panel = nullptr;
What version combinations of ESP32_Display_Panel , and ESP32_IO_Expander or worth trying? (In combination with which example?)

Thanks

GraemeKP, 2025-01-29:
+1 from me too.
I've been trying lots of example code for this module (and similar). Nothing, "out of the box" works. 
I know my module is ok as I can load the pre-built binaries.

If anyone has the library revisions which result in a successful build, that would be fantastic!

Pa3555, 2025-01-31:
flollowing as i have the same error

[image]

Lzw655, 2025-02-05:
> +1 from me too. I've been trying lots of example code for this module (and similar). Nothing, "out of the box" works. I know my module is ok as I can load the pre-built binaries.
> 
> If anyone has the library revisions which result in a successful build, that would be fantastic!

Hi @GraemeKP,

Sorry for the late. Please refer to the [documentation](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/docs/How_To_Use.md#configuration-instructions-1) and configure the corresponding files based on your development board type before using the `ESP_Panel` drivers.

Lzw655, 2025-02-05:
> flollowing as i have the same error [@Lzw655](https://github.com/Lzw655) any solution ?
> 
> [image]

Hi @Pa3555,

Did you change to ESP32_IO_Expander v0.1.0?