esp-arduino-libs/ESP32_Display_Panel #38: How to correctly add other I2C component libraries?

jerryliugithub, 2024-05-30:
When adding additional components such as RTC and Sensor to the same I2C BUS, they have their own independent arduino libraries (using wire.h). Any suggestions when integrating into this library?  Thanks !

Lzw655, 2024-05-31:
Hi @jerryliugithub,

Sorry, this is indeed a problem that has been troubling us, and currently, there is no ideal solution. However, the underlying I2C driver used by this library is the same as the one used by the `Wire` library, so theoretically, you can try the following steps:

1. Initialize the I2C bus using `Wire` before using the `Display_Panel` in your program.
2. Set `ESP_PANEL_TOUCH_BUS_SKIP_INIT_HOST` to `1` in the `Display_Panel`, which means touch will not initialize the I2C bus.
3. Implement mutual exclusion using a semaphore in your program when reading the coordinates or interacting with other I2C devices.

jerryliugithub, 2024-06-03:
I successfully used Arduino wire.h to initialize I2C, and it works properly (with ESP_PANEL_TOUCH_BUS_SKIP_INIT_HOST set to 1).

Next, I plan to add a semaphore in the bool ESP_PanelTouch::readRawData(uint8_t max_points_num, int timeout_ms) function in ESP_PanelTouch.cpp. Is this the correct approach?

I encountered some issues with the expander IO. In my project, besides using it for SPI+RGB, I will also use it to operate some GPIO functions. However, I'm not sure where to start in the library. Any suggestions would be greatly appreciated.

Lzw655, 2024-06-04:
Hi @jerryliugithub,

I think you can just add a semaphore in the project instead of changing the code in the library. Here is a simple example:

```c
// file: test.ino

// Create a semaphore
i2c_sem = xSemaphoreCreateBinary();

...

// Take it before operating the same i2c bus
if (xSemaphoreTake(i2c_sem , timeout_ticks) == pdTRUE) {
    // Read touch point or operate expander IO
    ESP_PanelTouchPoint point;
    int read_touch_result = touch->readPoints(&point, 1);
    if (read_touch_result > 0) {
        Serial.printf("Touch point(%d): x %d, y %d, strength %d\n", i, point.x, point.y, point.strength);
    }

    ...

    // Release it after operation
    xSemaphoreGive(i2c_sem);
}

...

```

To use the expander with the library, there are two methods:

1. If you set the macro [ESP_PANEL_USE_EXPANDER](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/9cdc53e0f8ba78a3d4530bd1fc510f7b3b0f5a9f/ESP_Panel_Board_Custom.h#L306) to `1`, the library will handle the initialization internally. You can then call the function [panel->getExpander()](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/9cdc53e0f8ba78a3d4530bd1fc510f7b3b0f5a9f/src/ESP_Panel.h#L74) to access it externally.

2. If you set the macro [ESP_PANEL_USE_EXPANDER](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/9cdc53e0f8ba78a3d4530bd1fc510f7b3b0f5a9f/ESP_Panel_Board_Custom.h#L306) to `0`, you need to initialize the expander externally and then use the function [panel->configExpander()](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/9cdc53e0f8ba78a3d4530bd1fc510f7b3b0f5a9f/src/ESP_Panel.h#L40C10-L40C24) to install it in the library.

jerryliugithub, 2024-06-04:
Thank you for your quick response and detailed explanation. I am currently trying to extend the example SquareLine/v8/WiFiClock.ino. Since the underlying implementation already includes I2C Touch functionality, as shown in the following code, it would conflict with the additional I2C that I am planning to add. This is why I intend to modify the library. Thank you for the further explanation about the Expander. I will try to understand it.

#lvgl_port_v8.cpp
static lv_indev_t *indev_init(ESP_PanelTouch *tp)
-> indev_drv_tp.read_cb = touchpad_read;

static void touchpad_read(lv_indev_drv_t *indev_drv, lv_indev_data_t *data)
-> int read_touch_result = tp->readPoints(&point, 1);

#ESP_PanelTouch.cpp
int ESP_PanelTouch::readPoints(ESP_PanelTouchPoint points[], uint8_t num, int timeout_ms)
-> ESP_PANEL_CHECK_FALSE_RET(readRawData(num, timeout_ms), -1, "Read raw data failed");

bool ESP_PanelTouch::readRawData(uint8_t max_points_num, int timeout_ms)
-> ESP_PANEL_CHECK_ERR_RET(esp_lcd_touch_read_data(handle), false, "Read data failed");
   esp_lcd_touch_get_coordinates(handle, x, y, strength, &_tp_points_num, max_points_num);

closed 2024-06-12 by jerryliugithub

Lzw655, 2025-05-16:
Hi @velocitronics,

I'm sorry you've encountered this problem. Based on your description, it seems you've also used an IO expander with an I2C interface. So, in addition to enabling the `ESP_PANEL_TOUCH_BUS_SKIP_INIT_HOST` macro, please try enabling the `ESP_PANEL_BOARD_EXPANDER_SKIP_INIT_HOST` as well. 

If the problem still persists, please refer to the [FAQ](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/docs/envs/use_with_arduino.md#how-to-debug-when-the-screen-doesnt-light-up-using-the-library-in-arduino-ide) to obtain the complete debug log and then share it with me. I'll help you locate the problem.

velocitronics, 2025-05-16:
Hi Lzw655,

 I managed to FINALLY get the IO expander working. I did not skip the initialization because that doesn't work. I instead did this:

Above setup:

ESP_Panel *panel;  // Declare globally
ESP_IOExpander *expander = NULL; // Declare globally
SemaphoreHandle_t i2c_sem;  // Create global semaphore

void setup()

... other init code

i2c_sem = xSemaphoreCreateMutex();  // Create semaphore for safe access

....other init code

In setup after panel->begin();

delay(500);

expander = panel->getExpander();
expander->pinMode(4, OUTPUT);  // Ensure pin is set to OUTPUT

if (!expander) {
    Serial.println("Error: Expander is not initialized!");
  } else {
    Serial.println("Expander initialized!");
  }   

Here is the function I created:

void safeExpanderWrite(uint8_t pin, uint8_t val) {
  if (xSemaphoreTake(i2c_sem, pdMS_TO_TICKS(100)) == pdTRUE) {  // Take semaphore
    expander->digitalWrite(pin, val);  // Write to the expander safely
    xSemaphoreGive(i2c_sem);  // Release bus after transaction
  } else {
    Serial.println("I2C bus busy—try again later!");
  }
}

Here is the code I used to toggle the IO epander pin 4 in the loop as a test:

if (expander) {
    static bool toggleState = false;  // Keeps track of pin state

    safeExpanderWrite(4, toggleState);  // Write HIGH or LOW to pin 4
    toggleState = !toggleState;  // Flip the state for the next iteration
    Serial.println("Expander toggled!");
  }

As you can see, a simple example would CLEARLY help somebody figure this out. Since the ESP32-S3 does not have enough IO pins, customers like me have to resort to exotic things like this just to use the I2C bus for other purposes. Now I have to go and create a much crazier implementation by modifying the ESP Panel code just to include a custom handler for another I2C addition to communicate with the bus. Please, make it easier for others to use the ESP Panel code instead of making it difficult, and I would also like to see Espressif make an ESP32-S3 variant with MORE IO. Stop limiting the chip size, especially if it can drive an RGB LCD. Just my two cents' worth. 

Thanks,

Eric

velocitronics, 2025-05-16:
Hi Lzw655,

 I suggest creating a write-up with instructions that show customers how to use the ESP Panel code and how to gain access to the libraries. It is VERY complex, and to do this, it requires in-depth knowledge of how the whole library works and how FreeRTOS operates. I suggest that those who created the libraries make it easier to initialize and access the functions through the ino or cpp files in Arduino and through other IDEs too. No one wants to spend several days (possibly weeks) trying to understand how the library and FreeRTOS work. I surely didn't want to waste this much time trying to figure out how to toggle an IO pin on an expander I could toggle with another chip that takes five minutes to do. Making things a bit easier to understand is all I am asking.

Thanks,

Eric