esp-arduino-libs/ESP32_Display_Panel #43: Can the v9 version of the LVGL library be used?

appleshaman, 2024-06-08:
I only see the v8 version of the LVGL library in the examples, and the configuration files in the v9 version have been significantly changed and are quite different from those in v8. So, can it be used? 

I tried using the v9 version of the library, but the program didn't run correctly. So, I'm unsure if it's because of my configuration or if the current ESP32 panel library is incompatible with the v9 library.

Lzw655, 2024-06-12:
Hi @appleshaman,

Due to certain differences between versions 8 and 9 of LVGL, the examples from LVGL v8 cannot function properly with v9. We plan to include new examples specifically for v9 in the future.

appleshaman, 2024-06-12:
Thanks. I will first try the v8 version

closed 2024-06-12 by appleshaman