esp-arduino-libs/ESP32_Display_Panel #47: PlatformIO ini file problem

appleshaman, 2024-06-15:
When I am trying to use the PlatformIO example, the ini seems could be configured correctly, I just used the default in provided, but it gives this below:
Resolving ESP-LCD dependencies...
Error! Failed to extract upstream toolchainconfigurations:
Bad package version `esp-12.2.0_20230208`
You can disable this feature via the `board_build.arduino.upstream_packages = no` setting in your `platformio.ini` file.

Lzw655, 2024-06-17:
Hi @appleshaman, 

Sorry, I can't reproduce this issue. And I can't see the reason for this, because I am not very familiar with PlatformIO.

Hi @isthaison, did you get this issue before? And do you have any suggestions?

isthaison, 2024-06-17:
I think he is missing the installation using arduino-esp32 3.x.x

closed 2024-06-18 by appleshaman