esp-arduino-libs/ESP32_Display_Panel #52: Arduino RGB example error for ESP32 S3 to run ST7262 display

eagl1, 2024-06-16:
Hello,

I installed this library and I solved first library requirement which is `esp_io_expander.h`

Then I got this error, where `esp_memory_utils.h` is located in `esp-idf-master`.
```
In file included from E:\Programs_Files\Arduino\libraries\ESP32_Display_Panel-master\src/ESP_PanelLog.h:9,
                 from E:\Programs_Files\Arduino\libraries\ESP32_Display_Panel-master\src\lcd\ESP_PanelLcd.cpp:8:
E:\Programs_Files\Arduino\libraries\ESP32_Display_Panel-master\src/ESP_Panel_Conf_Internal.h:51:35: error: operator '!' has no right operand
 #if !ESP_PANEL_CONF_INCLUDE_INSIDE
                                   ^
E:\Programs_Files\Arduino\libraries\ESP32_Display_Panel-master\src\lcd\ESP_PanelLcd.cpp:13:10: fatal error: esp_memory_utils.h: No such file or directory
 #include "esp_memory_utils.h"
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board ESP32S3 Dev Module.

```

So how to solve these two errors ?

eagl1, 2024-06-16:
OK, I solved `esp_memory_utils.h` problem by including the full file path.

But still the second error:

```
In file included from E:\Programs_Files\Arduino\libraries\ESP32_Display_Panel-master\src/ESP_Panel_Library.h:9,
                 from E:\Programs_Files\Arduino\libraries\ESP32_Display_Panel-master\examples\LCD\RGB\RGB.ino:42:
E:\Programs_Files\Arduino\libraries\ESP32_Display_Panel-master\src/ESP_Panel_Conf_Internal.h:51:35: error: operator '!' has no right operand
 #if !ESP_PANEL_CONF_INCLUDE_INSIDE
                                   ^
exit status 1
Error compiling for board ESP32S3 Dev Module.

```

Lzw655, 2024-06-17:
Hi @eagl1, 

Sorry, it was a bug in the library. And I fixed it in the newest version, please update it. 

BTW, if you don't want to update, it can also be resolved by copying the [ESP_Panel_Conf.h](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/ESP_Panel_Conf.h) into the project.

eagl1, 2024-06-17:
OK, I deleted the previous library folder and downloaded the new one, and I get this error:

```
Arduino: 1.8.19 (Windows 10), Board: "ESP32S3 Dev Module, Disabled, Disabled, QIO 80MHz, 4MB (32Mb), Core 1, Core 1, Hardware CDC and JTAG, Disabled, Disabled, Disabled, UART0 / Hardware CDC, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), 921600, None, Disabled"

E:\Programs_Files\Arduino\libraries\ESP32_Display_Panel-master\src\backlight\ESP_PanelBacklight.cpp: In constructor 'ESP_PanelBacklight::ESP_PanelBacklight(int, bool, bool)':

E:\Programs_Files\Arduino\libraries\ESP32_Display_Panel-master\src\backlight\ESP_PanelBacklight.cpp:18:92: error: no matching function for call to 'ledc_timer_config_t::ledc_timer_config_t(<brace-enclosed initializer list>)'

     _channel_config(ESP_PANEL_BACKLIGHT_LEDC_CHANNEL_CONFIG_DEFAULT(io_num, light_up_level))

                                                                                            ^

In file included from E:\Programs_Files\Arduino\libraries\ESP32_Display_Panel-master\src\backlight\ESP_PanelBacklight.h:9,

                 from E:\Programs_Files\Arduino\libraries\ESP32_Display_Panel-master\src\backlight\ESP_PanelBacklight.cpp:8:

C:\Users\eagle\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.17/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note: candidate: 'ledc_timer_config_t::ledc_timer_config_t()'

 } ledc_timer_config_t;

   ^~~~~~~~~~~~~~~~~~~

C:\Users\eagle\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.17/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note:   candidate expects 0 arguments, 1 provided

C:\Users\eagle\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.17/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note: candidate: 'constexpr ledc_timer_config_t::ledc_timer_config_t(const ledc_timer_config_t&)'

C:\Users\eagle\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.17/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'const ledc_timer_config_t&'

C:\Users\eagle\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.17/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note: candidate: 'constexpr ledc_timer_config_t::ledc_timer_config_t(ledc_timer_config_t&&)'

C:\Users\eagle\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.17/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'ledc_timer_config_t&&'

exit status 1

Error compiling for board ESP32S3 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

Lzw655, 2024-06-17:
The library depends on arduino-esp v3.0.x SDK. For more details, please check the [docs](https://github.com/esp-arduino-libs/ESP32_Display_Panel/tree/master?tab=readme-ov-file#dependencies-and-versions).

eagl1, 2024-06-17:
I updated the package to the latest one 3.0.1 and got these errors:

```
Arduino: 1.8.19 (Windows 10), Board: "ESP32S3 Dev Module, Disabled, Disabled, QIO 80MHz, 4MB (32Mb), Core 1, Core 1, Hardware CDC and JTAG, Disabled, Disabled, Disabled, UART0 / Hardware CDC, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), 921600, None, Disabled, Disabled"

In file included from C:\Users\eagle\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32s3/include/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:79,

                 from C:\Users\eagle\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32s3/include/freertos/FreeRTOS-Kernel/include/freertos/portable.h:59,

                 from C:\Users\eagle\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32s3/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:71,

                 from E:\Programs_Files\Arduino\libraries\ESP32_Display_Panel-master\src\lcd\ESP_PanelLcd.cpp:15:

C:\Users\eagle\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32s3/include/esp_hw_support/include/esp_memory_utils.h:29:20: error: redefinition of 'bool esp_dram_match_iram()'

   29 | inline static bool esp_dram_match_iram(void) {

      |                    ^~~~~~~~~~~~~~~~~~~

In file included from E:\Programs_Files\Arduino\libraries\ESP32_Display_Panel-master\src\lcd\ESP_PanelLcd.cpp:13:

E:\Programs_Files\Arduino\hardware\esp-idf-master\components\esp_hw_support\include\esp_memory_utils.h:29:20: note: 'bool esp_dram_match_iram()' previously defined here

   29 | inline static bool esp_dram_match_iram(void) {

      |                    ^~~~~~~~~~~~~~~~~~~

C:\Users\eagle\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32s3/include/esp_hw_support/include/esp_memory_utils.h:41:20: error: redefinition of 'bool esp_ptr_in_iram(const void*)'

   41 | inline static bool esp_ptr_in_iram(const void *p) {

      |                    ^~~~~~~~~~~~~~~

E:\Programs_Files\Arduino\hardware\esp-idf-master\components\esp_hw_support\include\esp_memory_utils.h:41:20: note: 'bool esp_ptr_in_iram(const void*)' previously defined here

   41 | inline static bool esp_ptr_in_iram(const void *p) {

      |                    ^~~~~~~~~~~~~~~

C:\Users\eagle\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32s3/include/esp_hw_support/include/esp_memory_utils.h:57:20: error: redefinition of 'bool esp_ptr_in_dram(const void*)'

   57 | inline static bool esp_ptr_in_dram(const void *p) {

      |                    ^~~~~~~~~~~~~~~

E:\Programs_Files\Arduino\hardware\esp-idf-master\components\esp_hw_support\include\esp_memory_utils.h:57:20: note: 'bool esp_ptr_in_dram(const void*)' previously defined here

   57 | inline static bool esp_ptr_in_dram(const void *p) {

      |                    ^~~~~~~~~~~~~~~

C:\Users\eagle\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32s3/include/esp_hw_support/include/esp_memory_utils.h:69:20: error: redefinition of 'bool esp_ptr_in_diram_dram(const void*)'

   69 | inline static bool esp_ptr_in_diram_dram(const void *p) {

      |                    ^~~~~~~~~~~~~~~~~~~~~

E:\Programs_Files\Arduino\hardware\esp-idf-master\components\esp_hw_support\include\esp_memory_utils.h:69:20: note: 'bool esp_ptr_in_diram_dram(const void*)' previously defined here

   69 | inline static bool esp_ptr_in_diram_dram(const void *p) {

      |                    ^~~~~~~~~~~~~~~~~~~~~

C:\Users\eagle\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32s3/include/esp_hw_support/include/esp_memory_utils.h:81:20: error: redefinition of 'bool esp_ptr_in_diram_iram(const void*)'

   81 | inline static bool esp_ptr_in_diram_iram(const void *p) {

      |                    ^~~~~~~~~~~~~~~~~~~~~

E:\Programs_Files\Arduino\hardware\esp-idf-master\components\esp_hw_support\include\esp_memory_utils.h:81:20: note: 'bool esp_ptr_in_diram_iram(const void*)' previously defined here
[... 132 lines cut ...]

  326 | inline static bool esp_stack_ptr_in_dram(uint32_t sp)

      |                    ^~~~~~~~~~~~~~~~~~~~~

C:\Users\eagle\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32s3/include/esp_hw_support/include/esp_memory_utils.h:337:20: error: redefinition of 'bool esp_stack_ptr_is_sane(uint32_t)'

  337 | inline static bool esp_stack_ptr_is_sane(uint32_t sp)

      |                    ^~~~~~~~~~~~~~~~~~~~~

E:\Programs_Files\Arduino\hardware\esp-idf-master\components\esp_hw_support\include\esp_memory_utils.h:351:20: note: 'bool esp_stack_ptr_is_sane(uint32_t)' previously defined here

  351 | inline static bool esp_stack_ptr_is_sane(uint32_t sp)

      |                    ^~~~~~~~~~~~~~~~~~~~~

exit status 1

Error compiling for board ESP32S3 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

eagl1, 2024-06-17:
OK, the code complied with no issues.

eagl1, 2024-06-17:
This is the working `.json` configuration from the platformio project.

```
{
  "build": {
    "arduino": {
      "ldscript": "esp32s3_out.ld",
      "partitions": "default_16MB.csv",
      "memory_type": "qio_opi"
    },
    "core": "esp32",
    "extra_flags": [
      "'-D ARDUINO_ESP32S3_DEV'",
      "'-D BOARD_HAS_PSRAM'",
      "'-D ARDUINO_USB_MODE=1'",
      "'-D ARDUINO_RUNNING_CORE=1'",
      "'-D ARDUINO_EVENT_RUNNING_CORE=1'",
      "'-D ARDUINO_USB_CDC_ON_BOOT=0'",
      "'-D ESP32_8048S043C'",
      "'-D DISPLAY_WIDTH=800'",
      "'-D DISPLAY_HEIGHT=480'",
      "'-D LVGL_BUFFER_PIXELS=(DISPLAY_WIDTH*DISPLAY_HEIGHT)'",
      "'-D LVGL_BUFFER_MALLOC_FLAGS=(MALLOC_CAP_SPIRAM|MALLOC_CAP_8BIT)'",
      "'-D GPIO_BCKL=2'",
      "'-D DISPLAY_ST7262_PAR'",
      "'-D ST7262_PANEL_CONFIG_CLK_SRC=LCD_CLK_SRC_PLL160M'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_PCLK_HZ=(12.5*1000000)'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_H_RES=DISPLAY_WIDTH'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_V_RES=DISPLAY_HEIGHT'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_HSYNC_PULSE_WIDTH=4'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_HSYNC_BACK_PORCH=8'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_HSYNC_FRONT_PORCH=8'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_VSYNC_PULSE_WIDTH=4'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_VSYNC_BACK_PORCH=8'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_VSYNC_FRONT_PORCH=8'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_FLAGS_HSYNC_IDLE_LOW=false'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_FLAGS_VSYNC_IDLE_LOW=false'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_FLAGS_DE_IDLE_HIGH=false'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_FLAGS_PCLK_ACTIVE_NEG=true'",
      "'-D ST7262_PANEL_CONFIG_TIMINGS_FLAGS_PCLK_IDLE_HIGH=false'",
      "'-D ST7262_PANEL_CONFIG_DATA_WIDTH=16'",
      "'-D ST7262_PANEL_CONFIG_SRAM_TRANS_ALIGN=4'",
      "'-D ST7262_PANEL_CONFIG_PSRAM_TRANS_ALIGN=64'",
      "'-D ST7262_PANEL_CONFIG_HSYNC_GPIO_NUM=39'",
      "'-D ST7262_PANEL_CONFIG_VSYNC_GPIO_NUM=41'",
      "'-D ST7262_PANEL_CONFIG_DE_GPIO_NUM=40'",
      "'-D ST7262_PANEL_CONFIG_PCLK_GPIO_NUM=42'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R0=8'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R1=3'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R2=46'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R3=9'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_R4=1'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G0=5'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G1=6'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G2=7'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G3=15'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G4=16'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_G5=4'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B0=45'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B1=48'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B2=47'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B3=21'",
      "'-D ST7262_PANEL_CONFIG_DATA_GPIO_B4=14'",
      "'-D ST7262_PANEL_CONFIG_DISP_GPIO_NUM=GPIO_NUM_NC'",
      "'-D ST7262_PANEL_CONFIG_FLAGS_DISP_ACTIVE_LOW=false'",
      "'-D ST7262_PANEL_CONFIG_FLAGS_RELAX_ON_IDLE=false'",
      "'-D ST7262_PANEL_CONFIG_FLAGS_FB_IN_PSRAM=true'",
      "'-D DISPLAY_SWAP_XY=false'",
      "'-D DISPLAY_MIRROR_X=false'",
      "'-D DISPLAY_MIRROR_Y=false'",
      "'-D BOARD_HAS_TOUCH'",
      "'-D CONFIG_ESP_LCD_TOUCH_MAX_POINTS=1'",
      "'-D CONFIG_ESP_LCD_TOUCH_MAX_BUTTONS=0'",
      "'-D TOUCH_GT911_I2C'",
      "'-D GT911_I2C_HOST=I2C_NUM_0'",
      "'-D GT911_I2C_CONFIG_SDA_IO_NUM=19'",
      "'-D GT911_I2C_CONFIG_SCL_IO_NUM=20'",
      "'-D GT911_I2C_CONFIG_SDA_PULLUP_EN=GPIO_PULLUP_DISABLE'",
      "'-D GT911_I2C_CONFIG_SCL_PULLUP_EN=GPIO_PULLUP_DISABLE'",
      "'-D GT911_I2C_CONFIG_MASTER_CLK_SPEED=400000'",
      "'-D GT911_I2C_CONFIG_CLK_FLAGS=I2C_SCLK_SRC_FLAG_FOR_NOMAL'",
      "'-D GT911_IO_I2C_CONFIG_DEV_ADDR=GT911_IO_I2C_CONFIG_DEV_ADDRESS_5D'",
      "'-D GT911_IO_I2C_CONFIG_CONTROL_PHASE_BYTES=1'",
      "'-D GT911_IO_I2C_CONFIG_DC_BIT_OFFSET=0'",
      "'-D GT911_IO_I2C_CONFIG_LCD_CMD_BITS=16'",
      "'-D GT911_IO_I2C_CONFIG_LCD_PARAM_BITS=0'",
      "'-D GT911_IO_I2C_CONFIG_FLAGS_DC_LOW_ON_DATA=false'",
      "'-D GT911_IO_I2C_CONFIG_FLAGS_DISABLE_CONTROL_PHASE=true'",
      "'-D GT911_TOUCH_CONFIG_X_MAX=DISPLAY_WIDTH'",
      "'-D GT911_TOUCH_CONFIG_Y_MAX=DISPLAY_HEIGHT'",
      "'-D GT911_TOUCH_CONFIG_RST_GPIO_NUM=38'",
      "'-D GT911_TOUCH_CONFIG_INT_GPIO_NUM=18'",
      "'-D GT911_TOUCH_CONFIG_LEVELS_RESET=0'",
      "'-D GT911_TOUCH_CONFIG_LEVELS_INTERRUPT=0'",
      "'-D TOUCH_SWAP_XY=false'",
      "'-D TOUCH_MIRROR_X=false'",
      "'-D TOUCH_MIRROR_Y=false'",
      "'-D BOARD_HAS_TF'",
      "'-D TF_CS=10'",
      "'-D TF_SPI_MOSI=11'",
      "'-D TF_SPI_SCLK=12'",
      "'-D TF_SPI_MISO=13'"
    ],
    "f_cpu": "240000000L",
    "f_flash": "80000000L",
    "flash_mode": "qio",
    "hwids": [
      [
        "0x303A",
        "0x1001"
      ]
    ],
    "mcu": "esp32s3",
    "variant": "esp32s3"
  },
  "connectivity": [
    "wifi"
  ],
  "debug": {
    "openocd_target": "esp32s3.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "esp32-8048S043C",
  "upload": {
    "flash_size": "16MB",
    "maximum_ram_size": 327680,
    "maximum_size": 16777216,
    "use_1200bps_touch": true,
    "wait_for_upload_port": true,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://www.aliexpress.com/item/1005006110360174.html",
  "vendor": "Sunton"
}
```

So I edited the pin arrangement to the one in your arduino RGB sketch and it still not working, I'm sure there's still something missing in the code.

eagl1, 2024-06-17:
The serial monitor is continuously trying to initialize the display. This is the serial output:

[image]

eagl1, 2024-06-17:
OK, I checked the supported boards.

I think I have to do add a new file for the display I have `esp32-8048S043C`.

Lzw655, 2024-06-18:
> The serial monitor is continuously trying to initialize the display. This is the serial output: [...]

When using RGB LCD, ESP32-S3 needs to enable PSRAM. There are some [recommended configurations in Arduino IDE](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/src/board/Board_Instructions.md#recommended-configurations-in-the-arduino-ide).

[image]

eagl1, 2024-06-18:
I'm still getting this error:

```
Arduino: 1.8.19 (Windows 10), Board: "ESP32S3 Dev Module, Disabled, OPI PSRAM, QIO 80MHz, 4MB (32Mb), Core 1, Core 1, Hardware CDC and JTAG, Disabled, Disabled, Disabled, UART0 / Hardware CDC, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi), 921600, None, Disabled, Disabled"

In file included from E:\projects_electronics_programming\tft\other sketches\GFX_Library_for_Arduino\GFX_Library_for_Arduino.ino:11:

Arduino_GFX_dev_device.h:14:1: error: 'Arduino_ESP32RGBPanel' does not name a type; did you mean 'Arduino_ESP32S2PAR8Q'?

   14 | Arduino_ESP32RGBPanel *rgbpanel = new Arduino_ESP32RGBPanel(

      | ^~~~~~~~~~~~~~~~~~~~~

      | Arduino_ESP32S2PAR8Q

Arduino_GFX_dev_device.h:22:1: error: 'Arduino_RGB_Display' does not name a type

   22 | Arduino_RGB_Display *gfx = new Arduino_RGB_Display(

      | ^~~~~~~~~~~~~~~~~~~

E:\projects_electronics_programming\tft\other sketches\GFX_Library_for_Arduino\GFX_Library_for_Arduino.ino: In function 'void setup()':

GFX_Library_for_Arduino:50:8: error: 'gfx' was not declared in this scope

   50 |   if (!gfx->begin())

      |        ^~~

GFX_Library_for_Arduino:56:7: error: 'gfx' was not declared in this scope

   56 |   w = gfx->width();

      |       ^~~

E:\projects_electronics_programming\tft\other sketches\GFX_Library_for_Arduino\GFX_Library_for_Arduino.ino: In function 'void loop()':

GFX_Library_for_Arduino:148:5: error: 'gfx' was not declared in this scope

  148 |     gfx->drawFastHLine(0, i, w, c);

      |     ^~~

GFX_Library_for_Arduino:156:3: error: 'gfx' was not declared in this scope

  156 |   gfx->setCursor(0, 0);

      |   ^~~

E:\projects_electronics_programming\tft\other sketches\GFX_Library_for_Arduino\GFX_Library_for_Arduino.ino: In function 'void serialOut(const char*, int32_t, uint32_t, bool)':

GFX_Library_for_Arduino:226:5: error: 'gfx' was not declared in this scope

  226 |     gfx->fillScreen(BLACK);

      |     ^~~

E:\projects_electronics_programming\tft\other sketches\GFX_Library_for_Arduino\GFX_Library_for_Arduino.ino: In function 'void printnice(const char*, long int)':

GFX_Library_for_Arduino:236:3: error: 'gfx' was not declared in this scope

  236 |   gfx->setTextSize(tsb);

      |   ^~~

E:\projects_electronics_programming\tft\other sketches\GFX_Library_for_Arduino\GFX_Library_for_Arduino.ino: In function 'int32_t testFillScreen()':

GFX_Library_for_Arduino:272:3: error: 'gfx' was not declared in this scope

  272 |   gfx->fillScreen(WHITE);
[... 147 lines cut ...]
E:\projects_electronics_programming\tft\other sketches\GFX_Library_for_Arduino\GFX_Library_for_Arduino.ino: In function 'int32_t testFilledRoundRects()':

GFX_Library_for_Arduino:638:5: error: 'gfx' was not declared in this scope

  638 |     gfx->fillRoundRect(cx - i2, cy - i2, i, i, i / 8, gfx->color565(0, i, 0));

      |     ^~~

E:\projects_electronics_programming\tft\other sketches\GFX_Library_for_Arduino\GFX_Library_for_Arduino.ino: In function 'int32_t testRoundRects()':

GFX_Library_for_Arduino:654:5: error: 'gfx' was not declared in this scope

  654 |     gfx->drawRoundRect(cx - i2, cy - i2, i, i, i / 8, gfx->color565(i, 0, 0));

      |     ^~~

exit status 1

'Arduino_ESP32RGBPanel' does not name a type; did you mean 'Arduino_ESP32S2PAR8Q'?

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

Lzw655, 2024-06-18:
Sorry, I am not familiar with the Arduino_GFX library.

eagl1, 2024-06-18:
No, I'm sorry my mistake. I'm working on both libraries + platformio projects.

I did the PSRAM enable but still the same problem.

Lzw655, 2024-06-19:
Okay... That's a little strange. Could you set the `Core Debug Level` to `Debug` and check the log again? It will show the PSRAM status.

[image]

eagl1, 2024-06-19:
OK, I think it worked.

This is the RGB example:

[image]

I have another question, if I wanted to add ST7262 source code to another arduino C++ library, how to copy the init data ? because I want to add this display to TFT_eSPI library because it helps me using GUIslice software and support our college projects.

Thanks,

Lzw655, 2024-06-20:
Nice! However, I have no idea how to add this display to the TFT_eSPI library.