esp-arduino-libs/ESP32_Display_Panel #57: PanelTest.ino --> Compilation error: 'ESP_Panel' does not name a type

wavejaco, 2024-06-20:
I am experiencing the exact same issue that @lboue had in [Issue 26](https://github.com/esp-arduino-libs/ESP32_Display_Panel/issues/26), although with ESP32_Display_Panel at version 0.1.4 and ESP32_IO_Expander at version 0.0.2.

The suggestion in that issue of setting the macro `ESP_PANEL_USE_CUSTOM_BOARD` to 1 in the file _ESP_Panel_Board_Custom.h_ didn't fix the problem. Clearly something else is wrong. What is the root cause of this compilation error?

Any help will be greatly appreciated, thanks.

closed 2024-06-20 by wavejaco

Lzw655, 2024-06-20:
Great! And I am thinking about outputting some messages to specify the path of the files like *ESP_Panel_Board_Custom.h*.

wavejaco, 2024-06-20:
> Great! And I am thinking about outputting some messages to show the path of the files like _ESP_Panel_Board_Custom.h_.

That is a good idea and should avoid problems such as the one I encountered. In hindsight it seems obvious which is the correct file to change, but some messages showing the path of the files should help to avoid this issue in the future.

Lzw655, 2025-02-25:
Hi @ciapamusche,

Can you share the full error message?