esp-arduino-libs/ESP32_Display_Panel #58: ESP_PANEL_LCD_RGB_IO_DATAx pins mapping to display RGB pins

wavejaco, 2024-06-20:
For a custom board, the GPIO pin layout is specified in the _ESP_Panel_Board_Custom.h_ file. These pins are named ESP_PANEL_LCD_RGB_IO_DATA0 up to ESP_PANEL_LCD_RGB_IO_DATA15. For a board such as the [Waveshare ESP32-S3 4.3 inch](https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-4.3) display module, the LCD pin layout is described in terms of R, G and B pins. Below is an extract from the above site:

[image]

How do these pins (R3-R7, G2-G7 and B3-B7) map to the pins ESP_PANEL_LCD_RGB_IO_DATAx (x = 0 to 15) in _ESP_Panel_Board_Custom.h_?

Lzw655, 2024-06-24:
Hi @wavejaco,

Sorry for the delay. The ESP32-S3 only supports 16-bit RGB565 and 8-bit RGB888. Typically, we connect a 24-bit RGB888 LCD by only using the high bits, as illustrated in the table below.

| RGB888 Bit | RGB IO    |  LCD Bit  |
|------------|-------------|---------------|
| Bit 0      | -       | B0            |
| Bit 1      | -       | B1            |
| Bit 2      | -       | B2            |
| Bit 3      | DATA0       | B3            |
| Bit 4      | DATA1       | B4            |
| Bit 5      | DATA2       | B5            |
| Bit 6      | DATA3       | B6            |
| Bit 7      | DATA4       | B7            |
| Bit 8      | -       | G0            |
| Bit 9      | -       | G1            |
| Bit 10     | DATA5      | G2            |
| Bit 11     | DATA6      | G3            |
| Bit 12     | DATA7      | G4            |
| Bit 13     | DATA8     | G5            |
| Bit 14     | DATA9      | G6            |
| Bit 15     | DATA10      | G7            |
| Bit 16     | -      | R0            |
| Bit 17     | -      | R1            |
| Bit 18     | -      | R2            |
| Bit 19     | DATA11      | R3            |
| Bit 20     | DATA12      | R4            |
| Bit 21     | DATA13      | R5            |
| Bit 22     | DATA14      | R6            |
| Bit 23     | DATA15      | R7            |

lboue, 2024-07-09:
@wavejaco Did it work?

closed 2024-07-10 by Lzw655

wavejaco, 2024-07-10:
> @wavejaco Did it work?

My apologies for the late reply. Yes, this worked perfectly. Thanks.