esp-arduino-libs/ESP32_Display_Panel #60: The Arduino RGB example of the ESP32 S3 running the ST7262 display has a dotted line on the far right

H-sw123, 2024-06-21:
Hello,I use the LVGL example from ESP-IDF and it displays fine without dashed lines
My screen is 800x480.
I have tried changing the timing, clock, double buffer, BOUNCE, flush_callback coordinates, but none of them worked.

[image]
How do I fix this?
Any help is appreciated!

Lzw655, 2024-06-24:
Hi @H-sw123,

I think it's maybe an RGB screen drift issue. Did you try to enable [XIP on PSRAM](https://github.com/esp-arduino-libs/ESP32_Display_Panel?tab=readme-ov-file#how-to-fix-screen-drift-issue-when-driving-rgb-lcd-with-esp32-s3) ?

H-sw123, 2024-06-25:
Hello,
Thanks for your suggestion, I solved this problem later by setting ESP_PANEL_LCD_RGB_PCLK_ACTIVE_NEG to True.
You can close this question.
Regards!

closed 2024-06-25 by H-sw123