esp-arduino-libs/ESP32_Display_Panel #61: How to enable serial port for the Platform IO

appleshaman, 2024-06-22:
My current ESP32 can run LVGL code normally, but the serial port cannot print properly. It only automatically prints the following:

ESP-ROM:esp32s3-20210327
Build: Mar 27 2021
rst:0x1 (POWERON), boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode: DIO, clock div:1
load:0x3fce3818, len:0x508
load:0x403c9700, len:0x4
load:0x403c9704, len:0xad0
load:0x403cc700, len:0x29e4
entry 0x403c9880

Other LVGL functions work fine; I guess I missed something in the config, but I don't know what happened. I also tried the Porting function for the Arduino IDE; it can print the expected information, and I tried enabling the ESP_PANEL_ENABLE_LOG in the ESP_Panel_Conf; it can print the information correctly, 
here is my ini:
[env:ESP-LCD]
platform = espressif32@6.7.0
board = ESP-LCD
framework = arduino
platform_packages =
	platformio/framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git
	platformio/framework-arduinoespressif32-libs@https://github.com/espressif/esp32-arduino-libs.git#idf-release/v5.1
upload_speed = 921600
monitor_speed = 115200
build_flags =
	-DBOARD_HAS_PSRAM
	-DLV_CONF_INCLUDE_SIMPLE
	-DDISABLE_ALL_LIBRARY_WARNINGS
	-DARDUINO_USB_CDC_ON_BOOT=1
	-DCORE_DEBUG_LEVEL=1
	-DLV_LVGL_H_INCLUDE_SIMPLE
	-I src
	-D VN=1
	-D ESP32S3=1
	-DCONFIG_SPIRAM_TYPE_OPI
lib_deps =
	https://github.com/esp-arduino-libs/ESP32_Display_Panel.git
	https://github.com/esp-arduino-libs/ESP32_IO_Expander.git
    https://github.com/lvgl/lvgl.git#release/v8.3
   
   I guess my board file is probably correct since the LVGL function can work
   {
    "build": {
      "arduino": {
        "ldscript": "esp32s3_out.ld",
        "partitions": "default_8MB.csv",
        "memory_type": "qio_opi"
      },
      "core": "esp32",
      "extra_flags": [
        "-DBOARD_HAS_PSRAM",
        "-DARDUINO_USB_MODE=1",
        "-DARDUINO_RUNNING_CORE=1",
        "-DARDUINO_EVENT_RUNNING_CORE=1"
      ],
      "f_cpu": "240000000L",
      "f_flash": "80000000L",
      "flash_mode": "qio",
      "hwids": [["0x303A", "0x1001"]],
      "mcu": "esp32s3",
      "variant": "esp32s3"
    },
    "connectivity": ["wifi"],
    "debug": {
      "default_tool": "esp-builtin",
      "onboard_tools": ["esp-builtin"],
      "openocd_target": "esp32s3.cfg"
    },
    "frameworks": ["arduino", "espidf"],
    "name": "ESP-LCD (8M QIO Flash & OPI PSRAM)",
    "upload": {
      "flash_size": "8MB",
      "maximum_ram_size": 327680,
      "maximum_size": 8388608,
      "require_upload_port": true,
      "speed": 921600
    },
    "url": "",
    "vendor": "ESP-LCD"
  }

Lzw655, 2024-06-24:
Hi @appleshaman,

Are you using a USB port or a UART port? If it's a USB port, I think you can resolve the issue by [enabling "USB CDC on boot"](https://community.platformio.org/t/enabling-usb-cdc-on-boot-on-esp32-c3-devkit/33346).

appleshaman, 2024-06-24:
I am using the UART port, and it seems adding -DARDUINO_USB_CDC_ON_BOOT=1 into the ini does not help. Any other way to solve this problem? thanks

Lzw655, 2024-06-24:
Oh, that's the issue. You should remove `-DARDUINO_USB_CDC_ON_BOOT=1` if using the UART port.

appleshaman, 2024-06-24:
I see, it's gone, thanks for your help!

closed 2024-06-24 by appleshaman