esp-arduino-libs/ESP32_Display_Panel #63: LVGL v9  support

isthaison, 2024-06-26:


Lzw655, 2024-06-27:
Hi @isthaison,

This can be achieved by porting the [esp_lvgl_port](https://github.com/espressif/esp-bsp/tree/master/components/esp_lvgl_port) component, and I intend to do this next month.

martinroger, 2024-10-24:
@Lzw655 Did you get any chance to progress with that port ?

Lzw655, 2024-10-25:
I apologize for not completing this as scheduled. I've been extremely busy lately, but I believe I should have some time to work on it next month. 💪

lsroka76, 2024-11-06:
@Lzw655 May I ask how far is Your progess in that subject? I'm asking because I've already started porting by myself but I don't want to duplicate thing that may already be done?

Lzw655, 2024-11-07:
Hi @lsroka76, the porting work has not started yet, but we are planning for it, and it is expected to begin next week.

fbiego, 2025-01-06:
any progress on v9 support?

Lzw655, 2025-01-06:
Hi @fbiego, I'm really sorry. This work will start after the release of version 1.0. Due to the significant changes in the new version, it has taken much longer than we initially expected.

bobskigit, 2025-02-06:
lvgl v9 drivers have been developed for some of the boards mentioned already.

They work seamlessly in platformIO

https://github.com/rzeldent/esp32-smartdisplay

Erriez, 2025-07-20:
WIP: [VIEWE UEDX48270043E-ESP32 480x272 4.3 inch multi-point capacitive touch PlatformIO LVGL9](https://github.com/Erriez/UEDX48270043E-ESP32-4.3inch-Touch-Display/tree/staging/errriez/add_platformio_lvgl9/examples/platformio/lvgl_v9_port)

Erriez, 2025-07-28:
Yes, standard configuration with LVGL9 is functional, but noticed lower performance. API's and benchmarks are different between 8 and 9. Now tesing with double buffering and different screen configurations before creating a PR.

Basitadam, 2025-07-28:
> Yes, standard configuration with LVGL9 is functional, but noticed lower performance. API's and benchmarks are different between 8 and 9. Now tesing with double buffering and different screen configurations before creating a PR.

Thank you @Erriez . Good news

closed 2026-11-15 by isthaison