esp-arduino-libs/ESP32_Display_Panel #64: LCD Display and Touch Only Work If "RESET" is Pressed During Power Up and Serial Monitor is Opened

zyan003, 2024-06-26:
`Board: Waveshare ESP32-S3-Touch-LCD-7 with ESP32-S3-WROOM-1-N8R8`
`#define ESP_PANEL_LCD_NAME          ST7262`
`#define ESP_PANEL_TOUCH_NAME        GT911`

Hi, I have encountered some weird issues with ESP32_Display_Panel. After uploading the code to ESP32-S3, it does not reset automatically and LCD display shows nothing. The error messages are something like this:
```
E (30) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed
E (30) GT911: touch_gt911_read_cfg(362): GT911 read error!
E (31) GT911: esp_lcd_touch_new_i2c_gt911(128): GT911 init failed
E (37) GT911: Error (0xffffffff)! Touch controller GT911 initialization failed!
```

However, if I hold the "RESET" button when powering up and open serial monitor, then press "RESET" again, LCD starts showing content and the touch works fine as well. Here is some information which may help. The issue may be something related to UART.
```
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------  
    19 : USB_DM
    20 : USB_DP
    43 : UART_TX[0]
    44 : UART_RX[0]
```

If anyone has any advice or comments, please let me know. Many thanks!

Lzw655, 2024-06-26:
Hi @zyan003,

I checked the schematic of the Waveshare ESP32-S3-Touch-LCD-7 board. And I think there are two points that may cause the issue:

1. The `CTP_RST` pin is not pulled up. It is connected to the EXIO1 pin of CH422 which is input mode when powered. I think it should be pulled up first.

[image]

2. The `CTP_IRQ` pin is not pulled down. The I2C address of GT911 is decided by the level of the INT pin as shown in the figure. And the library uses `0x5D` as the default address of GT911. So I think you should refer to the [code](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/c695ee2a42bb33ed5d72c485e29c602057485765/src/board/espressif/ESP32_S3_BOX_3.h#L198-L201) to set the pin to low level before resetting it.

[image]

zyan003, 2024-06-26:
Thanks for your time and answer, I will give it a shot.

zyan003, 2024-06-26:
I notice that I have to press the physical "RESET" button for at least 400ms when powering up to have program working properly. So I update the code by

```
#define ESP_PANEL_TOUCH_IO_RST          (1)
#define ESP_PANEL_TOUCH_RST_LEVEL       (1)  // 0: low level, 1: high level

#define ESP_PANEL_TOUCH_IO_INT          (4)
#define ESP_PANEL_TOUCH_INT_LEVEL       (0)         // 0: low level, 1: high level

#define ESP_PANEL_BEGIN_TOUCH_START_FUNCTION(panel)  \
{  \
    gpio_set_direction((gpio_num_t)ESP_PANEL_TOUCH_IO_INT, GPIO_MODE_OUTPUT); /* Set CTP_IRQ pin to output mode */ \
    gpio_set_level((gpio_num_t)ESP_PANEL_TOUCH_IO_INT, 0); /* Set CTP_IRQ pin to low */ \
    vTaskDelay(pdMS_TO_TICKS(10)); /* Wait for 10ms */ \
    panel->_expander_ptr->pinMode(ESP_PANEL_TOUCH_IO_RST, OUTPUT); /* Set CTP_RST pin to output mode */ \
    panel->_expander_ptr->digitalWrite(ESP_PANEL_TOUCH_IO_RST, HIGH); /* Pull up CTP_RST pin */ \
    vTaskDelay(pdMS_TO_TICKS(100)); /* Wait for 100ms */ \
    panel->_expander_ptr->digitalWrite(ESP_PANEL_TOUCH_IO_RST, LOW); /* Reset GT911 by setting CTP_RST pin to low */ \
    vTaskDelay(pdMS_TO_TICKS(400)); /* Wait for 400ms */ \
    panel->_expander_ptr->digitalWrite(ESP_PANEL_TOUCH_IO_RST, HIGH); /* Pull up CTP_RST pin again to complete reset */ \
}
```

I also enable IO expander
```
#define ESP_PANEL_USE_EXPANDER          (1)         // 0/1
#if ESP_PANEL_USE_EXPANDER
#define ESP_PANEL_EXPANDER_NAME         CH422G

#define ESP_PANEL_EXPANDER_SKIP_INIT_HOST       (0)     // 0/1
#define ESP_PANEL_EXPANDER_I2C_ADDRESS          (0x20)
#if !ESP_PANEL_EXPANDER_SKIP_INIT_HOST
    #define ESP_PANEL_EXPANDER_HOST_ID          (0)     // Typically set to 0
    #define ESP_PANEL_EXPANDER_I2C_CLK_HZ       (400 * 1000)
                                                        // Typically set to 400K
    #define ESP_PANEL_EXPANDER_I2C_SCL_PULLUP   (1)     // 0/1
    #define ESP_PANEL_EXPANDER_I2C_SDA_PULLUP   (1)     // 0/1
    #define ESP_PANEL_EXPANDER_I2C_IO_SCL       (9)
    #define ESP_PANEL_EXPANDER_I2C_IO_SDA       (8)
#endif
#endif
```

I am not sure why macro is correct and do not fully understand the delay and timing in the diagram in previous comment. Am I on the right track? Now, the program is still now working proper if I am not pressing "RESET" during power-up. Thanks!

rolloo, 2024-06-26:
Waveshare ESP32-S3-Touch-LCD-7  ver1.1
_ESP_Panel_Board_Custom.h_

```
#pragma once

// *INDENT-OFF*

/* Set to 1 if using a custom board */
#define ESP_PANEL_USE_CUSTOM_BOARD       (1)         // 0/1

#if ESP_PANEL_USE_CUSTOM_BOARD

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
//////////////////////////// Please update the following macros to configure the LCD panel /////////////////////////////
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
/* Set to 1 when using an LCD panel */
#define ESP_PANEL_USE_LCD           (1)     // 0/1

#if ESP_PANEL_USE_LCD

#define ESP_PANEL_LCD_NAME          ST7262

/* LCD resolution in pixels */
#define ESP_PANEL_LCD_WIDTH         (800)
#define ESP_PANEL_LCD_HEIGHT        (480)

/* LCD Bus Settings */

#define ESP_PANEL_LCD_BUS_SKIP_INIT_HOST    (1)     // 0/1

#define ESP_PANEL_LCD_BUS_TYPE      (ESP_PANEL_BUS_TYPE_RGB)

#if ESP_PANEL_LCD_BUS_TYPE == ESP_PANEL_BUS_TYPE_RGB

    #define ESP_PANEL_LCD_RGB_CLK_HZ            (16 * 1000 * 1000)
    #define ESP_PANEL_LCD_RGB_HPW               (4)
    #define ESP_PANEL_LCD_RGB_HBP               (8)
    #define ESP_PANEL_LCD_RGB_HFP               (8)
    #define ESP_PANEL_LCD_RGB_VPW               (4)
    #define ESP_PANEL_LCD_RGB_VBP               (8)
    #define ESP_PANEL_LCD_RGB_VFP               (8)
    #define ESP_PANEL_LCD_RGB_PCLK_ACTIVE_NEG   (1)     // 0: rising edge, 1: falling edge
    #define ESP_PANEL_LCD_RGB_DATA_WIDTH        (16)    //  8 | 16
    #define ESP_PANEL_LCD_RGB_PIXEL_BITS        (16)    // 24 | 16
    #define ESP_PANEL_LCD_RGB_FRAME_BUF_NUM     (1)     // 1/2/3
    #define ESP_PANEL_LCD_RGB_BOUNCE_BUF_SIZE   (ESP_PANEL_LCD_WIDTH * 10)     // Bounce buffer size in bytes. This function is used to avoid screen drift.
                                                        // To enable the bounce buffer, set it to a non-zero value. Typically set to `ESP_PANEL_LCD_WIDTH * 10`
                                                        // The size of the Bounce Buffer must satisfy `width_of_lcd * height_of_lcd = size_of_buffer * N`,
                                                        // where N is an even number.
    #define ESP_PANEL_LCD_RGB_IO_HSYNC          (46)
    #define ESP_PANEL_LCD_RGB_IO_VSYNC          (3)
    #define ESP_PANEL_LCD_RGB_IO_DE             (5)   // -1 if not used
    #define ESP_PANEL_LCD_RGB_IO_PCLK           (7)
    #define ESP_PANEL_LCD_RGB_IO_DISP           (-1)    // -1 if not used
    #define ESP_PANEL_LCD_RGB_IO_DATA0          (14)
    #define ESP_PANEL_LCD_RGB_IO_DATA1          (38)
    #define ESP_PANEL_LCD_RGB_IO_DATA2          (18)
    #define ESP_PANEL_LCD_RGB_IO_DATA3          (17)
    #define ESP_PANEL_LCD_RGB_IO_DATA4          (10)
    #define ESP_PANEL_LCD_RGB_IO_DATA5          (39)
    #define ESP_PANEL_LCD_RGB_IO_DATA6          (0)
    #define ESP_PANEL_LCD_RGB_IO_DATA7          (45)
#if ESP_PANEL_LCD_RGB_DATA_WIDTH > 8
    #define ESP_PANEL_LCD_RGB_IO_DATA8          (48)
    #define ESP_PANEL_LCD_RGB_IO_DATA9          (47)
    #define ESP_PANEL_LCD_RGB_IO_DATA10         (21)
    #define ESP_PANEL_LCD_RGB_IO_DATA11         (1)
    #define ESP_PANEL_LCD_RGB_IO_DATA12         (2)
    #define ESP_PANEL_LCD_RGB_IO_DATA13         (42)
[... 88 lines cut ...]

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
///////////////////////////// Please update the following macros to configure the IO expander //////////////////////////
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
/* Set to 0 if not using IO Expander */
#define ESP_PANEL_USE_EXPANDER          (0)         // 0/1

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
///////////////////////////// Please utilize the following macros to execute any additional code if required. //////////
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
// #define ESP_PANEL_BEGIN_START_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_EXPANDER_START_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_EXPANDER_END_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_LCD_START_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_LCD_END_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_TOUCH_START_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_TOUCH_END_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_BACKLIGHT_START_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_BACKLIGHT_END_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_END_FUNCTION( panel )

#endif /* ESP_PANEL_USE_CUSTOM_BOARD */

// *INDENT-OFF*
```

WebDust21, 2024-06-26:
Funny...I'm running into the exact same issue with the exact same board.  

So far, I've come across the following:
 - with the default Waveshare code (well, after I fixed it up so it would run), the LCD sometimes initializes, and sometimes it doesn't.  (Always fails with the I2C bus errors when it doesn't work.)  Pressing the Reset button does not fix the issue--but power cycling it _does_ sometimes fix the issue.  When it works upon startup, it works.  And when it doesn't work on startup, no amount of resetting fixes it.
 - After enabling the CH422G and adding the reset code (independently from this issue--I *just* found it trying to solve my problems!), it is fairly consistent: every _other_ time I press the Reset button, the display will properly initialize.  Meaning that every alternate time, it I2C bus errors out and fails to init.  Tempts me to just use an "ESP.reset()" call if the display fails init!

I enabled debug logs, and get the following when it errors:

```
entry 0x403c9880
[   184][I][esp32-hal-psram.c:90] psramInit(): PSRAM enabled
[   198][I][display.cpp:28] rtos_hid_tft(): Initialize ESP_Panel Library
D (21) ESP_PanelHost: Add host I2C[0]
I (21) ESP_PanelHost: ESP_PanelHost::begin() called
D (22) ESP_PanelHost: Initialize host I2C[0]
D (22) ESP_PanelHost: Destroyed
[   209][I][display.cpp:37] rtos_hid_tft(): Start ESP_Panel Library
D (208) ST7262_CPP: LCD panel @0x3fca1cfc created
D (208) ESP_PanelLcd: Begin start
D (209) ESP_PanelLcd: Begin end
D (209) ESP_PanelBus_I2C: Panel IO @0x3fcece50 created
D (212) ESP_PanelTouch: Disable interruption
E (217) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed
E (223) GT911: touch_gt911_read_cfg(362): GT911 read error!
E (229) GT911: esp_lcd_touch_new_i2c_gt911(128): GT911 init failed
E (234) GT911: Error (0xffffffff)! Touch controller GT911 initialization failed!
E (242) GT911_CPP: [ESP_FAIL] begin(43): New driver failed
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400556d5  PS      : 0x00060b30  A0      : 0x8206a08a  A1      : 0x3fca1660  
A2      : 0x0000021c  A3      : 0x00000218  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3fca1620  
A10     : 0x00000000  A11     : 0x3c0901ac  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000021c  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  
```
It seems to crash (instead of just erroring out) after I enabled logs in "ESP_Panel_Conf.h":

```
/* Set to 1 if print log message for debug */
#define ESP_PANEL_ENABLE_LOG                (1)         // 0/1
```

@rolloo ...haha, you got the PCLK phase corrected!  Waveshare libs have the PCLK phase = 0, which results in a very blurry LCD appearance!

WebDust21, 2024-06-26:
EDIT: crash is pointing to my added "ESP_PANEL_BEGIN_LCD_START_FUNCTION", which I based on an example in the ESP_Panel default board defs.  For who knows what reason, the example wrote "low" twice to the reset pin ( https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/c695ee2a42bb33ed5d72c485e29c602057485765/src/board/espressif/ESP32_S3_KORVO_2.h#L211-L228 )
My experience with HD44780 LCDs is that the init code that resets them a half-dozen times before writing to them is for some reason reliable--but a straightforward single-reset init method is quite unreliable.  (Not sure why.)

GT911 reset timings are pretty involved apparently.  From the datasheet:

[image]

So I added your "INT" holddown code to the routine--and now it always initializes correctly.  For good measure, I added a delay to follow the above GT911 spec before releasing the pin to an input again.
FWIW I'm not using ESP_PANEL_TOUCH_IO_INT, as that just spills out a lot of debug messages, and is not necessary.  Haven't dug into ESP_Panel to see where it controls the IO_INT pin state (as to whether that would conflict with the patch here).

Here's what I have:
```
// #define ESP_PANEL_BEGIN_LCD_START_FUNCTION( panel )
#define ESP_PANEL_BEGIN_LCD_START_FUNCTION( panel ) \
    { \
        ESP_LOGD(TAG, "Run Waveshare 7\" LCD start function"); \
        ESP_PANEL_CHECK_NULL_RET(_expander_ptr, false, "Invalid IO expander pointer"); \
        \
        gpio_set_direction((gpio_num_t)4, GPIO_MODE_OUTPUT); /* Set CTP_IRQ pin to output mode */ \
        gpio_set_level((gpio_num_t)4, 0); /* Set CTP_IRQ pin to low */ \
        vTaskDelay(pdMS_TO_TICKS(10)); /* Wait for 10ms */ \
        \
        panel->_expander_ptr->multiPinMode(IO_EXPANDER_PIN_NUM_1 | IO_EXPANDER_PIN_NUM_2 | IO_EXPANDER_PIN_NUM_3 | IO_EXPANDER_PIN_NUM_6, OUTPUT); \
        /* Reset LCD */ \
        panel->_expander_ptr->digitalWrite(2, LOW); /*DISP-BACKLIGHT off for test*/ \
        panel->_expander_ptr->digitalWrite(1, LOW); /*CTP_RST*/ \
        panel->_expander_ptr->digitalWrite(3, LOW); /*LCD_RST*/ \
        vTaskDelay(pdMS_TO_TICKS(20)); \
        panel->_expander_ptr->digitalWrite(1, LOW); /*CTP_RST*/ \
        panel->_expander_ptr->digitalWrite(3, LOW); /*LCD_RST*/ \
        vTaskDelay(pdMS_TO_TICKS(100)); \
        panel->_expander_ptr->digitalWrite(1, HIGH); /*CTP_RST*/ \
        panel->_expander_ptr->digitalWrite(3, HIGH); /*LCD_RST*/ \
        \
        vTaskDelay(pdMS_TO_TICKS(60)); /* GT911 datasheet has 50mS spec before INPUT */ \
        gpio_set_direction((gpio_num_t)4, GPIO_MODE_INPUT); /* Set CTP_IRQ pin to input mode */ \
        \
        /* Turn on backlight_ptr */ \
        panel->_expander_ptr->digitalWrite(2, HIGH); /*DISP-BACKLIGHT*/ \
        panel->_expander_ptr->digitalWrite(6, HIGH); /*LCD_VDD_EN.  Adds a squeal to the board, but no visible difference.*/ \
    }
```

zyan003, 2024-06-27:
@WebDust21, thanks for the answer, I was trying your macro and the screen still cannot initialize properly without pressing "RESET" button while powering up. Is it possible to share your entire "ESP_Panel_Board_Custom.h"? Really appreciate that.

zyan003, 2024-06-27:
OK, so I add the GT911 reset timings to the main Arduino .ino file (not "ESP_Panel_Board_Custom.h") like this and the problem was solved.
```
void main() {
    ...
    expander->multiPinMode(TP_RST | LCD_BL | LCD_RST, OUTPUT);
    expander->multiDigitalWrite(TP_RST | LCD_BL | LCD_RST, HIGH);
    delay(100);
    expander->multiDigitalWrite(TP_RST | LCD_RST, LOW);
    delay(100);
    digitalWrite(GPIO_INPUT_IO_4, LOW);
    delay(100);
    expander->multiDigitalWrite(TP_RST | LCD_RST, HIGH);
    delay(200);
    ...
}
```

closed 2024-06-27 by zyan003

WebDust21, 2024-06-27:
> @WebDust21, thanks for the answer, I was trying your macro and the screen still cannot initialize properly without pressing "RESET" button while powering up. Is it possible to share your entire "ESP_Panel_Board_Custom.h"? Really appreciate that.

Sure.  Added a TXT extension so GitHub would accept it:

[ESP_Panel_Board_Custom.h.txt](https://github.com/user-attachments/files/16014509/ESP_Panel_Board_Custom.h.txt)

Note that I have it turn the backlight off during the reset cycle--that was for debug purposes to verify that the port expander code was working.  You can remove those lines if you want.

Lzw655, 2024-06-28:
Hi @WebDust21,

Could you assist me in supporting the Waveshare ESP32-S3-Touch-LCD-7 board by adding your config file? Here is the detailed [contribution guide](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/src/board/Board_Contribution_Guide.md). I can do this if you're busy. Since I lack the hardware, could you please help me test it if you find it convenient?

WebDust21, 2024-06-29:
> Hi @WebDust21,
> 
> Could you assist me in supporting the `Waveshare ESP32-S3-Touch-LCD-7` board by adding your config file?
Lemme try to finish some things up on it.  Of notable interest is the backlight functionality; the backlight is on the CH422 port expander, so PWM control is out of the question.

I noticed the following extensible calls for the backlight functionality:

```
// #define ESP_PANEL_BEGIN_BACKLIGHT_START_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_BACKLIGHT_END_FUNCTION( panel )
```

however there is no extensible call for the control functionality.  (Reading through the code, it appears that the ESP32_Display_Panel backlight component does not have any ability to support backlight on a GPIO extender.)

One major suggestion I could make to the ESP32_Display_Panel project, would be if pins could be specified with an "EXPANDER" flag, i.e. define "ESP_EXPANDER_PIN" like this:
```
#define ESP_EXPANDER_PIN 128 +
``` 
and then use it in the board configuration file like this:
```
#define ESP_PANEL_TOUCH_IO_RST          (ESP_EXPANDER_PIN 4)
```
That way it would be very easy to specify pins on an external expander (assuming just one port expander) without requiring special extensible functions for basic functionality.  (Yes, it would slightly complicate the actual pin handling functionality, as it would have to check for GPIO pin number >= 128 -> check for expander, etc.)  But I think this would be a better long-term supported modus, especially if the EXPANDER is included as part of the ESP32_Display_Panel subsystem.

Obviously, high-speed lines (RGB bus, etc.) can't be run over an expander--but low speed control lines can.

> Here is the detailed [contribution guide](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/src/board/Board_Contribution_Guide.md). I can do this if you're busy. Since I lack the hardware, could you please help me test it if you find it convenient?

I would appreciate it if you could do the necessary contribution process--and yes, I would be happy to help test the hardware.

Lzw655, 2024-07-01:
@WebDust21 Since Espressif's development board only uses the IO expander in the RGB 3-wire SPI pins, we hadn't considered supporting the IO expander in other drivers before. But you are right, we can support the IO expander in the touch and backlight drivers as well. I will complete this part of the work as soon as possible. After that, it would be great if you could help adapt it to that development board.

WebDust21, 2024-07-02:
If the ESP32_Display_Panel could support use of the IO expander in all low-speed functionality (i.e. RESET and strapping signals for the TFT LCD, as well as the touch controller), it would completely eliminate the need for custom user functions to properly initialize the screen.
I believe the GT911 initialization code has the necessary functionality to strap the INT pin when driving RESET--but in this case, the RESET pin is on the CH422 port expander--which means we can't use the default GT911 init code, as there's no way to have it address the expander for the RESET control!  (Also worth noting that the default Waveshare code does not use the GT911 INT pin functionality; the touchscreen is just polled.)

I understand that Waveshare having put the LCD backlight on the CH422 expander limits us to only "on" or "off" for the backlight (i.e. PWM not possible)--but the ESP_PanelBacklight class already has support for that.  Just needs to be addressable to the port expander!

Thanks again for your work on this library!

For reference purposes, here's the pin assignments of the Waveshare board:
[image]
("DISP" goes to the backlight control chip.  LCD_VDD_EN goes to the TFT backplane voltage boost converter.  When set high, the squeal of the boost converter can be heard, but the interestingly the TFT display appears the same with it on or off.)

@Lzw655 Yes, of course, I will be happy to help test and verify the code functionality with the Waveshare 7" LCD ESP32 board.