esp-arduino-libs/ESP32_Display_Panel #65: No Touchscreen + expander: Compilation error

rolloo, 2024-06-28:
Hello,
 I am trying to build an example for [Adafruit Qualia ESP32-S3](https://www.adafruit.com/product/5800) + [ Round RGB TTL TFT Display - 4" 720x720 - No Touchscreen](https://www.adafruit.com/product/5793)
 
- Compilation error: _'ESP_PANEL_TOUCH_BUS_HOST' was not declared in this scope_

_ESP_Panel_Board_Custom.h_
```
#define ESP_PANEL_USE_TOUCH             (0) 
#define ESP_PANEL_USE_EXPANDER          (1)
```

Lzw655, 2024-07-01:
Hi @rolloo, 

Sorry, there's a bug in the library. When setting up the expander's host, it should use the macro `ESP_PANEL_EXPANDER_HOST_ID` instead of `ESP_PANEL_TOUCH_BUS_HOST`. 

The PR fixes this problem. Can you assist me in testing it?

Lzw655, 2024-07-02:
😂 Sorry, the macro should be `ESP_PANEL_EXPANDER_HOST`. Fixed in the new commit, please try again.

closed 2024-07-07 by Lzw655