esp-arduino-libs/ESP32_Display_Panel #68: LCD RGB Element Order different between examples

lboue, 2024-07-09:
## LCD RGB Element Order different between examples
LCD RGB Element Order is different between examples. For example with LCD/RGB/RGB.ino](https://github.com/lboue/ESP32_Display_Panel/blob/CrowPanel_testing/examples/LCD/RGB/RGB.ino) and [LVGL/v8/Porting/Porting.ino](https://github.com/lboue/ESP32_Display_Panel/blob/CrowPanel_testing/examples/LVGL/v8/Porting/Porting.ino)

### LCD RGB example
* Defaut "LCD RGB Element Order" for [RGB.ino](https://github.com/lboue/ESP32_Display_Panel/blob/CrowPanel_testing/examples/LCD/RGB/RGB.ino) is BGR.

### LVGL v8 Porting example

* Defaut "LCD RGB Element Order" for [Porting.ino](https://github.com/lboue/ESP32_Display_Panel/blob/CrowPanel_testing/examples/LVGL/v8/Porting/Porting.ino)

https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/293faff58c65b1a6e03f34b13b4b513e71b8e408/examples/Panel/PanelTest/ESP_Panel_Board_Custom.h#L191

Is there a way to simplify by homogenizing?

Regards

Lzw655, 2024-07-09:
Hi @lboue,

1. The color order (B - G - R) mentioned in the comments of the LCD RGB example indicates the order of the color bars drawn from top to bottom in the `colorBarTest()`

[image]

2. The "LCD RGB Element Order" in *ESP_Panel_Board_Custom.h* refers to setting the properties of the LCD driver chip through commands. The image below shows the "RGB Element Order" for the ST7701.

[image]

lboue, 2024-07-09:
Thanks for the explanation.
So in what order should the GPIOs be indicated here, since this is not specified. I couldn't find any comments to guide me.
https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/a20b003af748d97e78cb23b7f4150f65ea38c678/examples/LCD/RGB/RGB.ino#L72-L88

lboue, 2024-07-09:
For top-down display Blue, Green, Red. I had to change the GPIO order in the example configuration. My board is board the Elecrow CrowPanel 7.0":

```
#define EXAMPLE_LCD_PIN_NUM_RGB_DISP            (-1)
#define EXAMPLE_LCD_PIN_NUM_RGB_VSYNC           (40)
#define EXAMPLE_LCD_PIN_NUM_RGB_HSYNC           (39)
#define EXAMPLE_LCD_PIN_NUM_RGB_DE              (41)
#define EXAMPLE_LCD_PIN_NUM_RGB_PCLK            (0)

// Blue
#define EXAMPLE_LCD_PIN_NUM_RGB_DATA0           (15) // B0
#define EXAMPLE_LCD_PIN_NUM_RGB_DATA1           (7)  // B1
#define EXAMPLE_LCD_PIN_NUM_RGB_DATA2           (6)  // B2
#define EXAMPLE_LCD_PIN_NUM_RGB_DATA3           (5)  // B3
#define EXAMPLE_LCD_PIN_NUM_RGB_DATA4           (4)  // B4

// Green
#define EXAMPLE_LCD_PIN_NUM_RGB_DATA5           (9)  // G0
#define EXAMPLE_LCD_PIN_NUM_RGB_DATA6           (46) // G1
#define EXAMPLE_LCD_PIN_NUM_RGB_DATA7           (3)  // G2
#if EXAMPLE_LCD_RGB_DATA_WIDTH > 8
#define EXAMPLE_LCD_PIN_NUM_RGB_DATA8           (8)  // G3
#define EXAMPLE_LCD_PIN_NUM_RGB_DATA9           (16) // G4
#define EXAMPLE_LCD_PIN_NUM_RGB_DATA10          (1)  // G5

// Red
#define EXAMPLE_LCD_PIN_NUM_RGB_DATA11          (14) // R0
#define EXAMPLE_LCD_PIN_NUM_RGB_DATA12          (21) // R1
#define EXAMPLE_LCD_PIN_NUM_RGB_DATA13          (47) // R2
#define EXAMPLE_LCD_PIN_NUM_RGB_DATA14          (48) // R3
#define EXAMPLE_LCD_PIN_NUM_RGB_DATA15          (45) // R4
```

Lzw655, 2024-07-10:
> For top-down display Blue, Green, Red. I had to change the GPIO order in the example configuration. Otherwise, the display is inverted.
> 
> My board is board the Elecrow CrowPanel 7.0":
> 
> ```
> #define EXAMPLE_LCD_PIN_NUM_RGB_DISP            (-1)
> #define EXAMPLE_LCD_PIN_NUM_RGB_VSYNC           (40)
> #define EXAMPLE_LCD_PIN_NUM_RGB_HSYNC           (39)
> #define EXAMPLE_LCD_PIN_NUM_RGB_DE              (41)
> #define EXAMPLE_LCD_PIN_NUM_RGB_PCLK            (0)
> 
> // Blue
> #define EXAMPLE_LCD_PIN_NUM_RGB_DATA0           (15) // B0
> #define EXAMPLE_LCD_PIN_NUM_RGB_DATA1           (7)  // B1
> #define EXAMPLE_LCD_PIN_NUM_RGB_DATA2           (6)  // B2
> #define EXAMPLE_LCD_PIN_NUM_RGB_DATA3           (5)  // B3
> #define EXAMPLE_LCD_PIN_NUM_RGB_DATA4           (4)  // B4
> 
> // Green
> #define EXAMPLE_LCD_PIN_NUM_RGB_DATA5           (9)  // G0
> #define EXAMPLE_LCD_PIN_NUM_RGB_DATA6           (46) // G1
> #define EXAMPLE_LCD_PIN_NUM_RGB_DATA7           (3)  // G2
> #if EXAMPLE_LCD_RGB_DATA_WIDTH > 8
> #define EXAMPLE_LCD_PIN_NUM_RGB_DATA8           (8)  // G3
> #define EXAMPLE_LCD_PIN_NUM_RGB_DATA9           (16) // G4
> #define EXAMPLE_LCD_PIN_NUM_RGB_DATA10          (1)  // G5
> 
> // Red
> #define EXAMPLE_LCD_PIN_NUM_RGB_DATA11          (14) // R0
> #define EXAMPLE_LCD_PIN_NUM_RGB_DATA12          (21) // R1
> #define EXAMPLE_LCD_PIN_NUM_RGB_DATA13          (47) // R2
> #define EXAMPLE_LCD_PIN_NUM_RGB_DATA14          (48) // R3
> #define EXAMPLE_LCD_PIN_NUM_RGB_DATA15          (45) // R4
> ```

This configuration is correct. I suggest adding connection comments for the RGB pins in `ESP_Panel_Board_Custom.h`. This will help users understand how to connect the pins between the LCD and the ESP more clearly.

closed 2024-07-10 by lboue