esp-arduino-libs/ESP32_Display_Panel #74: 'LVGL_PORT_ROTATION_DEGREE' was not declared in this scope

lboue, 2024-07-25:
```
D:\ESP32_Display_Panel_Crow_7\ESP32_Display_Panel-master\examples\SquareLine\v8\WiFiClock\lvgl_port_v8.cpp:491:2: error: #else without #if
  491 | #else
      |  ^~~~
D:\ESP32_Display_Panel_Crow_7\ESP32_Display_Panel-master\examples\SquareLine\v8\WiFiClock\lvgl_port_v8.cpp:494:2: error: #endif without #if
  494 | #endif
      |  ^~~~~
In file included from d:\ESP32_Display_Panel_Crow_7\libraries\ESP32_Display_Panel\src/board/ESP_PanelBoard.h:51,
                 from d:\ESP32_Display_Panel_Crow_7\libraries\ESP32_Display_Panel\src/ESP_Panel_Board_Internal.h:85,
                 from d:\ESP32_Display_Panel_Crow_7\libraries\ESP32_Display_Panel\src/ESP_Panel_Library.h:10,
                 from D:\ESP32_Display_Panel_Crow_7\ESP32_Display_Panel-master\examples\SquareLine\v8\WiFiClock\lvgl_port_v8.cpp:7:
d:\ESP32_Display_Panel_Crow_7\libraries\ESP32_Display_Panel\src/board/elecrow/CROWPANEL_7_0.h:221:5: error: expected unqualified-id before '{' token
  221 |     {  \
      |     ^
In file included from d:\esp32_display_panel_crow_7\libraries\lvgl\src/misc/lv_log.h:16,
                 from d:\esp32_display_panel_crow_7\libraries\lvgl\lvgl.h:25,
                 from d:\ESP32_Display_Panel_Crow_7\libraries\lvgl\src/lvgl.h:17,
                 from D:\ESP32_Display_Panel_Crow_7\ESP32_Display_Panel-master\examples\SquareLine\v8\WiFiClock\lvgl_port_v8.cpp:8:
d:\esp32_display_panel_crow_7\libraries\lvgl\src\lv_conf_internal.h:46:120: note: '#pragma message: Possible failure to include lv_conf.h, please read the comment in this file if you get errors'
   46 |         #pragma message("Possible failure to include lv_conf.h, please read the comment in this file if you get errors")
      |                                                                                                                        ^
D:\ESP32_Display_Panel_Crow_7\ESP32_Display_Panel-master\examples\SquareLine\v8\WiFiClock\lvgl_port_v8.cpp: In function 'lv_disp_t* display_init(ESP_PanelLcd*)':
D:\ESP32_Display_Panel_Crow_7\ESP32_Display_Panel-master\examples\SquareLine\v8\WiFiClock\lvgl_port_v8.cpp:488:9: error: 'LVGL_PORT_ROTATION_DEGREE' was not declared in this scope
  488 |     if (LVGL_PORT_ROTATION_DEGREE == 90) || (LVGL_PORT_ROTATION_DEGREE == 270)
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
D:\ESP32_Display_Panel_Crow_7\ESP32_Display_Panel-master\examples\SquareLine\v8\WiFiClock\lvgl_port_v8.cpp:488:42: error: expected primary-expression before '||' token
  488 |     if (LVGL_PORT_ROTATION_DEGREE == 90) || (LVGL_PORT_ROTATION_DEGREE == 270)
      |                                          ^~
In file included from d:\ESP32_Display_Panel_Crow_7\libraries\ESP32_Display_Panel\src/board/ESP_PanelBoard.h:51,
                 from d:\ESP32_Display_Panel_Crow_7\libraries\ESP32_Display_Panel\src/ESP_Panel_Board_Internal.h:85,
                 from d:\ESP32_Display_Panel_Crow_7\libraries\ESP32_Display_Panel\src/ESP_Panel_Library.h:10,
                 from D:\ESP32_Display_Panel_Crow_7\ESP32_Display_Panel-master\examples\SquareLine\v8\WiFiClock\WiFiClock.ino:81:
d:\ESP32_Display_Panel_Crow_7\libraries\ESP32_Display_Panel\src/board/elecrow/CROWPANEL_7_0.h:221:5: error: expected unqualified-id before '{' token
  221 |     {  \
      |     ^
In file included from d:\esp32_display_panel_crow_7\libraries\lvgl\src/misc/lv_log.h:16,
                 from d:\esp32_display_panel_crow_7\libraries\lvgl\lvgl.h:25,
                 from d:\ESP32_Display_Panel_Crow_7\libraries\lvgl\src/lvgl.h:17,
                 from D:\ESP32_Display_Panel_Crow_7\ESP32_Display_Panel-master\examples\SquareLine\v8\WiFiClock\WiFiClock.ino:82:
d:\esp32_display_panel_crow_7\libraries\lvgl\src\lv_conf_internal.h:46:120: note: '#pragma message: Possible failure to include lv_conf.h, please read the comment in this file if you get errors'
   46 |         #pragma message("Possible failure to include lv_conf.h, please read the comment in this file if you get errors")
      |                                                                                                                        ^

Using library ESP32_Display_Panel at version 0.1.5 in folder: D:\ESP32_Display_Panel_Crow_7\libraries\ESP32_Display_Panel 
Using library ESP32_IO_Expander at version 0.0.3 in folder: D:\ESP32_Display_Panel_Crow_7\libraries\ESP32_IO_Expander 
Using library lvgl at version 8.3.11 in folder: D:\ESP32_Display_Panel_Crow_7\libraries\lvgl 
Using library ui at version 1.0 in folder: D:\ESP32_Display_Panel_Crow_7\libraries\ui 
Using library NTPClient at version 3.2.1 in folder: D:\ESP32_Display_Panel_Crow_7\libraries\NTPClient 
Using library WiFi at version 3.0.3 in folder: C:\Users\Ludovic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.3\libraries\WiFi 
Using library Networking at version 3.0.3 in folder: C:\Users\Ludovic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.3\libraries\Network 
Using library ArduinoJson at version 6.21.5 in folder: D:\ESP32_Display_Panel_Crow_7\libraries\ArduinoJson 
Using library HTTPClient at version 3.0.3 in folder: C:\Users\Ludovic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.3\libraries\HTTPClient 
Using library NetworkClientSecure at version 3.0.3 in folder: C:\Users\Ludovic\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.3\libraries\NetworkClientSecure 
exit status 1

Compilation error: #else without #if
```

lboue, 2024-07-25:
Partially fixed by #75 for the `CROWPANEL_7_0.h` error.

lboue, 2024-07-25:
It seems related to this:

https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/289d3575fee397d5ef80e10576b92497299db4a2/examples/SquareLine/v8/WiFiClock/lvgl_port_v8.h#L96

lboue, 2024-07-25:
Partially fixed by #76 PR.

Lzw655, 2024-07-28:
Sorry, this error was introduced while I was fixing a previous issue. Thank you very much for pointing it out.

closed 2024-07-30 by Lzw655