esp-arduino-libs/ESP32_Display_Panel #87: WiFiClock example: Poor graphics quality

lboue, 2024-07-25:
WiFiClock example: Poor graphics quality

[M5STACK_M5CORES3](https://docs.m5stack.com/en/core/CoreS3)

Lzw655, 2024-07-28:
Hi @lboue, did you try to set `LV_COLOR_16_SWAP` to `1` ?

lboue, 2024-07-28:
> Hi @lboue, did you try to set `LV_COLOR_16_SWAP` to `1` ?

@Lzw655 
I've just tested it by setting this variable as indicated in the `lv_conf.h` file.
I haven't noticed any significant difference.

Lzw655, 2024-07-29:
I tried modifying `LV_COLOR_16_SWAP`, and the UI is displaying correctly. Could you please confirm if the correct file was modified?

[image]

lboue, 2024-07-29:
> I tried modifying `LV_COLOR_16_SWAP`, and the UI is displaying correctly. Could you please confirm if the correct file was modified?

Yes, It doesn't work in my case. 
Can I have your binary file for testing? Here's mine:
[WiFiClock.ino.merged.bin.txt](https://github.com/user-attachments/files/16408006/WiFiClock.ino.merged.bin.txt)

**lv_conf.h config file**

https://github.com/lboue/ESP32_Display_Panel/blob/CoreS3_testing/examples/SquareLine/v8/WiFiClock/libraries/lv_conf.h

Lzw655, 2024-07-29:
Here are mine:

[lv_conf.h.txt](https://github.com/user-attachments/files/16409821/lv_conf.h.txt)
[WiFiClock.ino.merged.bin.txt](https://github.com/user-attachments/files/16409822/WiFiClock.ino.merged.bin.txt)

lboue, 2024-07-29:
Setting this line to 1 fixed my issue.

`#if 0 /*Set it to "1" to enable content*/
`
https://github.com/lboue/ESP32_Display_Panel/blob/17d60252b86d6ed134239054f1699165c1b9a941/examples/SquareLine/v8/WiFiClock/libraries/lv_conf.h#L15

Do you know why?

closed 2024-07-29 by lboue