esp-arduino-libs/ESP32_Display_Panel #90: Dependency missing in `library.properties` file

lboue, 2024-07-28:
Would it be possible to add the libraries used in the examples as a dependency of this library?
This would avoid having to manually install the libraries to test the examples.

https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/289d3575fee397d5ef80e10576b92497299db4a2/library.properties#L11

Lzw655, 2024-07-29:
Hi @lboue,

The example documentation and .ino file specify the dependent libraries and their versions. And I believe that only the components that the source code under src depends on should be considered fixed dependencies. This follows the principle of minimal dependencies. Otherwise, if future examples rely on more components, adding all of them as fixed dependencies wouldn't be elegant.

lboue, 2024-07-29:
OK I understand the reasoning even if I don't find it practical.

closed 2024-07-29 by lboue

Lzw655, 2024-07-29:
Thanks for your understanding! And please feel free to provide your suggestions.