esp-arduino-libs/ESP32_Display_Panel #94: Support st77903 controller.

DenisMartsenyuk, 2024-07-28:
Good day dear support.
Could you add support for displays with st77903 controller please? I need this for my project, and I think in the future it will be useful for other users of your library. 
I hope for your understanding, thanks in advance :)

Lzw655, 2024-07-29:
Hi @DenisMartsenyuk, Currently, the ST77903 driver relies on new features from the ESP-IDF release/v5.3 branch. However, the esp32-arduino v3.x is built on the ESP-IDF release/v5.1. To enable these features, we have to implement a patch in the esp32-arduino SDK.

DenisMartsenyuk, 2024-07-29:
Thanks for the reply :)
Do you have plans to migrate to ESP-IDF v5.3? If yes, when to expect it?

Lzw655, 2024-07-30:
Do you mean when esp32-arduino will migrate to v5.2? From the [discussion](https://github.com/espressif/arduino-esp32/discussions/9420), it seems they are preparing to develop directly based on v5.3. Based on previous experience, I estimate this will take quite a while.

DenisMartsenyuk, 2024-07-30:
Okay, thank you :)

Lzw655, 2024-07-30:
You are welcome 😉