esp-arduino-libs/ESP32_Display_Panel #95: How to extend display to entire screen on an 800x480 screen

lboue, 2024-07-29:
**Question**

Is there a way to set up the display to take up all the space on an 800x480 screen with the Squareline Simple Wi-Fi Clock Example?

It is stated that the example is suitable for 800x480 resolution, but this does not appear to be the case. In fact, the graphic interface occupies only the central part of the screen.

https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/9d1cc1e0319d03a2b215d5b1588f11054c30d3f0/examples/SquareLine/v8/WiFiClock/README.md?plain=1#L3

Lzw655, 2024-07-30:
Hi @lboue, I'm really sorry. This example is mainly used to demonstrate how to develop using the UI exported by Squareline. Initially, to be compatible with the 320x240 resolution, the UI elements are relatively small. Even if we extend their positions to fit the entire 800x480 screen, it would look uncoordinated.

lboue, 2024-07-30:
> Hi @lboue, I'm really sorry. This example is mainly used to demonstrate how to develop using the UI exported by Squareline. Initially, to be compatible with the 320x240 resolution, the UI elements [...]

If this is the case, shouldn't the explanations in the README file be modified?

Lzw655, 2024-07-30:
You are right. We can add some explanations to remind users about this issue.