esp-arduino-libs/ESP32_Display_Panel #103: LVGL (version 8) and SQLite3

jerryliugithub, 2024-08-21:
During the use of LVGL (version 8) and SQLite3, I encountered an issue: when inserting records into the SQLite3 database, the screen experiences scrolling, and the x and y coordinates of LVGL UI elements (such as buttons and text boxes) get misaligned. After the insertion operation is completed, the position of UI elements on the screen is incorrect. Commenting out the insertion of records into the SQLite3 database resolves the issue. It seems that the insertion action is affecting LVGL's screen update.

Lzw655, 2024-08-22:
Hi @jerryliugithub,

This issue is most likely due to an RGB LCD screen drifting problem. You can follow the [steps ](https://github.com/esp-arduino-libs/ESP32_Display_Panel?tab=readme-ov-file#how-to-fix-screen-drift-issue-when-driving-rgb-lcd-with-esp32-s3)to resolve it, and refer to the [comment](https://github.com/esp-arduino-libs/arduino-esp32-sdk/issues/1#issuecomment-2300032531) for more details on the cause.

jerryliugithub, 2024-08-22:
Thank you for the quick reply. Unfortunately, I can't find the current version of arduino-esp32 V3.0.3. Could you please provide me with the esp32-3.0.3-h version? I would be very grateful!

Lzw655, 2024-08-23:
Sure, I am working on it.

jerryliugithub, 2024-08-23:
I switched the Arduino ESP32 to V3.0.2, and it displayed correctly, which solved the problem I've been stuck on for three days. Thank you very much! 
However, I've noticed something strange: the original FREE HEAP and PSRAM capacity have decreased significantly. Also, the PSRAM SIZE should be 8MB, but it has become 5.5MB?

------------------------------------------
setup()
   :
   :
15:47:59.785 -> FREE HEAP:58752   <--- Originally about 90KB.
15:47:59.785 -> PSRAM SIZE:5505024 
15:47:59.785 -> FREE PSRAM:4379564
-------------------------------------------
loop()

Lzw655, 2024-08-23:
To prevent RGB screen drift issue, the high-performance SDK version has enabled the 'XIP on PSRAM' feature by default, as shown below. For details on its functionality, please refer to the [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/external-ram.html#execute-in-place-xip-from-psram).

[image]

Lzw655, 2024-08-27:
Hi @jerryliugithub, esp32-3.0.3-h is ready now, please check the [link](https://github.com/esp-arduino-libs/arduino-esp32-sdk#for-sdks-suffixed-with--h).

jerryliugithub, 2024-08-28:
I am using 8 screens with over 150 widgets, and I have also enabled PSRAM. Is this level of memory consumption normal?
15:15:35.236 -> [HEAP] 215564
15:15:35.236 -> Initialize panel device
15:15:35.236 -> E (4389) gpio: gpio_install_isr_service(500): GPIO isr service already installed
15:15:35.236 -> Initialize LVGL
15:15:35.283 -> Create UI
15:15:35.567 -> [HEAP] 98256