esp-arduino-libs/ESP32_Display_Panel #105: Examples of using simple drawing libraries

bajacondor, 2024-08-24:
Hello,

Thank you for this great code.  I've used it to get started with a touchscreen display.
Where are some examples of using your library with simple drawing libraries like Arduino_gfx or TFT_espi.  Isn't it possible to use libs such as those to directly render to the bus/panel created by ESP32_Display_Panel?

Thank you,
David

Lzw655, 2024-08-26:
Hi David,

Thank you for your feedback. This library is currently primarily used in conjunction with LVGL. Libraries like Arduino_gfx provide great graphics rendering capabilities, and perhaps I can consider porting them over in the future.

bajacondor, 2024-08-26:
@Lzw655, Thank you.  I actually got this to work with TFT_eSPI by drawing in memory to a sprite, and then using the panelLcd drawBitmap() function.

Lzw655, 2024-08-27:
Great!

bajacondor, 2024-08-27:
It is working, but it is extremely slow.  Are there ways to optimize the
parallel RGB performance of drawBitmap()?

Lzw655, 2024-08-28:
You can try to use the [high_perf](https://github.com/esp-arduino-libs/arduino-esp32-sdk#for-sdks-suffixed-with--h) version of the arduino-esp32 SDK.