esp-arduino-libs/ESP32_Display_Panel #107: lib/ESP32_Display_Panel/src/backlight/ESP_PanelBacklight.cpp:18:92: error: no matching function for call to 'ledc_timer_config_t::ledc_timer_config_t(<brace-enclosed initializer list>)'

hgwc, 2024-09-05:
Hello,

**1. My development environment is as follows.**
Processing esp32-s3-devkitc-1-n16r8v (platform: espressif32; board: esp32-s3-devkitc-1-n16r8v; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1-n16r8v.html
PLATFORM: Espressif 32 (6.3.2) > Espressif ESP32-S3-DevKitC-1-N16R8V (16 MB QD, 8MB PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP32_IO_Expander @ 1.1.0
|-- lvgl @ 8.3.8
|-- ESP32_Display_Panel @ 0.1.7
Building in release mode

**2. Build Error**
Archiving .pio\build\esp32-s3-devkitc-1-n16r8v\liba6c\liblvgl.a
Indexing .pio\build\esp32-s3-devkitc-1-n16r8v\liba6c\liblvgl.a
Compiling .pio\build\esp32-s3-devkitc-1-n16r8v\lib30a\ESP32_Display_Panel\bus\I2C.cpp.o
lib/ESP32_Display_Panel/src/backlight/ESP_PanelBacklight.cpp: In constructor 'ESP_PanelBacklight::ESP_PanelBacklight(int, bool, bool)':
lib/ESP32_Display_Panel/src/backlight/ESP_PanelBacklight.cpp:18:92: error: no matching function for call to 'ledc_timer_config_t::ledc_timer_config_t(<brace-enclosed initializer list>)'
     _channel_config(ESP_PANEL_BACKLIGHT_LEDC_CHANNEL_CONFIG_DEFAULT(io_num, light_up_level))
                                                                                            ^
In file included from lib/ESP32_Display_Panel/src/backlight/ESP_PanelBacklight.h:9,
                 from lib/ESP32_Display_Panel/src/backlight/ESP_PanelBacklight.cpp:8:
C:/Users/user/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note: candidate: 'ledc_timer_config_t::ledc_timer_config_t()'
 } ledc_timer_config_t;
   ^~~~~~~~~~~~~~~~~~~
C:/Users/user/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note:   candidate expects 0 arguments, 1 provided
C:/Users/user/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note: candidate: 'constexpr ledc_timer_config_t::ledc_timer_config_t(const ledc_timer_config_t&)'
C:/Users/user/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note:   no known conversion for argument 1 from
'<brace-enclosed initializer list>' to 'const ledc_timer_config_t&'
C:/Users/user/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note: candidate: 'constexpr ledc_timer_config_t::ledc_timer_config_t(ledc_timer_config_t&&)'
C:/Users/user/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note:   no known conversion for argument 1 from
'<brace-enclosed initializer list>' to 'ledc_timer_config_t&&'
*** [.pio\build\esp32-s3-devkitc-1-n16r8v\lib30a\ESP32_Display_Panel\backlight\ESP_PanelBacklight.cpp.o] Error 1

**Please give me advice on resolving this issue.**

Lzw655, 2024-09-06:
Hi @hgwc,

The issue is likely caused by the version of the SDK used by `framework-arduinoespressif32` not meeting this library's requirements. You can replace it with the [framework provided in the file](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/687cf90dd0e9ae70b015f612e1f44cef3c05601c/examples/PlatformIO/platformio.ini#L5-L7).

hgwc, 2024-09-06:
Thank you for your kind advice.
I applied the framework you mentioned.
However, an error occurred in platformio saying that ESP-LCD was not included in the board name.
So, I applied esp32-s3-devkitc-1-n16r8v to the board name.

When I ran it, the following message appeared:
*** [.pio\build\esp32-s3-devkitc-1-n16r8v\bootloader.bin] Source `C:\Users\user\.platformio\packages\framework-arduinoespressif32-libs\esp32s3\bin\bootloader_qio_.elf' not found, needed
by target `.pio\build\esp32-s3-devkitc-1-n16r8v\bootloader.bin'.

Detailed logs are below.
Processing esp32-s3-devkitc-1-n16r8v (platform: espressif32; board: esp32-s3-devkitc-1-n16r8v; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Library Manager: Installing >=0.0.1 && <0.1.0
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1-n16r8v.html
PLATFORM: Espressif 32 (6.3.2) > Espressif ESP32-S3-DevKitC-1-N16R8V (16 MB QD, 8MB PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.0.4+sha.e70f4d3
 - framework-arduinoespressif32-libs @ 5.1.0+sha.d38afc77db.sha.1a029d5
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 12.2.0+20230208
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP32_Display_Panel @ 0.1.7+sha.687cf90
|-- ESP32_IO_Expander @ 0.0.3+sha.e5378da
|-- lvgl @ 8.3.11+sha.4d96c27
Building in release mode
*** [.pio\build\esp32-s3-devkitc-1-n16r8v\bootloader.bin] Source `C:\Users\user\.platformio\packages\framework-arduinoespressif32-libs\esp32s3\bin\bootloader_qio_.elf' not found, needed
by target `.pio\build\esp32-s3-devkitc-1-n16r8v\bootloader.bin'.

Lzw655, 2024-09-07:
I sincerely apologize, I am not familiar with PlatformIO. 

@isthaison Do you have any suggestions for this issue?

isthaison, 2024-09-08:
examples/PlatformIO/boards/ESP-LCD.json
please 
You must adjust the parameters to suit the devkit you are using.

isthaison, 2024-09-08:
https://github.com/platformio/platform-espressif32/blob/develop/boards/esp32-s3-devkitc-1.json
Looks like devkit-C1 doesn't have dram 😂

Ifousoft, 2024-09-29:
> Hi @hgwc,
>
> The issue is likely caused by the version of the SDK used by `framework-arduinoespressif32` not meeting this library's requirements. You can replace it with the [framework provided in the [...]

Sorry, I have the same problem...
I already used the .platformio file referenced in your post but I always get the error on ledc function...

How to solve this issue ?

Thanks !

Ifousoft, 2024-09-30:
Tried with Espressif 5.1.1, got the same error...
Here the log:
Processing ESP-LCD (platform: espressif32@5.1.1; board: ESP-LCD_espressif_5.1; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Library Manager: Installing >=0.0.1 && <0.1.0
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ESP-LCD_espressif_5.1.html
PLATFORM: Espressif 32 (5.1.1) > ESP-LCD (16M QIO Flash & OPI PSRAM) - ESP5.1 Ifou
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 2.0.6+sha.23d715a
 - framework-arduinoespressif32-libs @ 5.1.0+sha.d38afc77db.sha.1a029d5
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 77 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP32_Display_Panel @ 0.1.7+sha.43f0bc1
|-- ESP32_IO_Expander @ 0.0.3+sha.e5378da
|-- lvgl @ 8.3.11+sha.4d96c27

Next the error:
Compiling .pio\build\ESP-LCD\libd2a\ESP32_Display_Panel\bus\RGB.cpp.o
.pio/libdeps/ESP-LCD/ESP32_Display_Panel/src/backlight/ESP_PanelBacklight.cpp: In constructor 'ESP_PanelBacklight::ESP_PanelBacklight(int, bool, bool)':
.pio/libdeps/ESP-LCD/ESP32_Display_Panel/src/backlight/ESP_PanelBacklight.cpp:18:92: error: no matching function for call to 'ledc_timer_config_t::ledc_timer_config_t(<brace-enclosed initializer list>)'
     _channel_config(ESP_PANEL_BACKLIGHT_LEDC_CHANNEL_CONFIG_DEFAULT(io_num, light_up_level))
                                                                                            ^
In file included from .pio/libdeps/ESP-LCD/ESP32_Display_Panel/src/backlight/ESP_PanelBacklight.h:9,
                 from .pio/libdeps/ESP-LCD/ESP32_Display_Panel/src/backlight/ESP_PanelBacklight.cpp:8:
C:/Users/Ifou/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note: candidate: 'ledc_timer_config_t::ledc_timer_config_t()'
 } ledc_timer_config_t;
   ^~~~~~~~~~~~~~~~~~~
C:/Users/Ifou/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note:   candidate expects 0 arguments, 1 provided
C:/Users/Ifou/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note: candidate: 'constexpr ledc_timer_config_t::ledc_timer_config_t(const ledc_timer_config_t&)'
C:/Users/Ifou/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'const ledc_timer_config_t&'
C:/Users/Ifou/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note: candidate: 'constexpr ledc_timer_config_t::ledc_timer_config_t(ledc_timer_config_t&&)'
C:/Users/Ifou/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note:   no known conversion for argument 1 from '<brace-enclosed initializer list>' to 'ledc_timer_config_t&&'

Help... Thanks!

Lzw655, 2024-10-08:
Hi @Ifousoft, sorry for the late reply. Although I don't understand why this issue occurs even with Espressif 5.1.1, I believe that "framework-arduinoespressif32 @ 2.0.6" is not compatible with "framework-arduinoespressif32-libs @ 5.1.0" because the official arduino-esp32 2.0.6 is used with Espressif 4.4.x.

Ifousoft, 2024-10-08:
Thanks for your message..
But... How could I solve this issue.
I would like to use this 7 inches display...

Thanx!

Lzw655, 2024-10-08:
Could you please share your .platformio file?

Ifousoft, 2024-10-09:
Yes. Here it is:
[env:ESP-LCD]
platform = espressif32@5.1.1
board = ESP-LCD_espressif_5.1
framework = arduino
platform_packages =
	platformio/framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git
	platformio/framework-arduinoespressif32-libs@https://github.com/espressif/esp32-arduino-libs.git#idf-release/v5.1
upload_speed = 921600
monitor_speed = 115200
build_flags =
	-DBOARD_HAS_PSRAM
	-DLV_CONF_INCLUDE_SIMPLE
	-DDISABLE_ALL_LIBRARY_WARNINGS
	-DARDUINO_USB_CDC_ON_BOOT=1
	-DCORE_DEBUG_LEVEL=1
	-DLV_LVGL_H_INCLUDE_SIMPLE
	-I src
lib_deps =
	https://github.com/esp-arduino-libs/ESP32_Display_Panel.git
	https://github.com/esp-arduino-libs/ESP32_IO_Expander.git
    https://github.com/lvgl/lvgl.git#release/v8.3

isthaison, 2024-10-09:
Do you creatw new simple repo. I'll check for you

Ifousoft, 2024-10-09:
Create a github repo with all my files?

I can try to do that this evening.

isthaison, 2024-10-09:
yeb. this is simple.

isthaison, 2024-10-09:
https://github.com/esp-arduino-libs/ESP32_Display_Panel/tree/master?tab=readme-ov-file#how-to-fix-screen-drift-issue-when-driving-rgb-lcd-with-esp32-s3 .  
platformio also needs this installation, have you replaced the lib in the platformio folder?

isthaison, 2024-10-09:
[image]
 replace esp32s3

isthaison, 2024-10-10:
[image]

isthaison, 2024-10-10:
https://github.com/isthaison/ESP32_7inches_Fail  , I fork repo and fixed for you

Ifousoft, 2024-10-12:
Dear Isthaison,

Sorry but it always failed. I do it again this evening:
- Download the git as Zip
- Copy as a new project in platformio
- replace C:\Users\Ifou\.platformio\packages\framework-arduinoespressif32-libs\esp32s3 bye the one provided on the git
- pio run.

I got the same error messages... Could you share the log of your succeeded compilation to see the differences with mine ?

Thanks!

Here my log:
ments_Famille_Local\David\ESP32\Projets\ESP32_7inches_Failed_V2> pio run
Processing ESP-LCD (platform: espressif32; board: ESP-LCD; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Library Manager: Installing git+https://github.com/esp-arduino-libs/ESP32_Display_Panel.git
git version 2.39.0.windows.2
Cloning into 'C:\Users\Ifou\.platformio\.cache\tmp\pkg-installing-gvkkxck9'...
remote: Enumerating objects: 291, done.
remote: Counting objects: 100% (291/291), done.
remote: Compressing objects: 100% (220/220), done.
remote: Total 291 (delta 136), reused 151 (delta 59), pack-reused 0 (from 0)
Receiving objects: 100% (291/291), 522.62 KiB | 2.65 MiB/s, done.
Resolving deltas: 100% (136/136), done.
Library Manager: ESP32_Display_Panel@0.1.7+sha.43f0bc1 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing >=0.0.1 && <0.1.0
Library Manager: Installing git+https://github.com/esp-arduino-libs/ESP32_IO_Expander.git
git version 2.39.0.windows.2
Cloning into 'C:\Users\Ifou\.platformio\.cache\tmp\pkg-installing-3ojzjc9e'...
remote: Enumerating objects: 48, done.
remote: Counting objects: 100% (48/48), done.
remote: Compressing objects: 100% (41/41), done.
Receiving objects:  66% (32/48)sed 39 (delta 4), pack-reused 0 (from 0)
Receiving objects: 100% (48/48), 25.30 KiB | 5.06 MiB/s, done.
Resolving deltas: 100% (7/7), done.
Library Manager: ESP32_IO_Expander@0.0.3+sha.e5378da has been installed!
Library Manager: Installing git+https://github.com/lvgl/lvgl.git#release/v8.3
git version 2.39.0.windows.2
Cloning into 'C:\Users\Ifou\.platformio\.cache\tmp\pkg-installing-cg5ytndg'...
remote: Enumerating objects: 1416, done.
remote: Counting objects: 100% (1416/1416), done.
remote: Compressing objects: 100% (1285/1285), done.
remote: Total 1416 (delta 176), reused 559 (delta 109), pack-reused 0 (from 0)
Receiving objects: 100% (1416/1416), 25.43 MiB | 2.08 MiB/s, done.
Resolving deltas: 100% (176/176), done.
Updating files: 100% (1238/1238), done.
Library Manager: lvgl@8.3.11+sha.4d96c27 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ESP-LCD.html
PLATFORM: Espressif 32 (6.5.0) > ESP-LCD (16M QIO Flash & OPI PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 2.0.6+sha.23d715a
 - framework-arduinoespressif32-libs @ 5.1.0+sha.d38afc77db.sha.1a029d5
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 77 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP32_Display_Panel @ 0.1.7+sha.43f0bc1
|-- ESP32_IO_Expander @ 0.0.3+sha.e5378da
|-- lvgl @ 8.3.11+sha.4d96c27
Building in release mode
Compiling .pio\build\ESP-LCD\src\app.cpp.o
Compiling .pio\build\ESP-LCD\src\lvgl_port_v8.cpp.o
[... 4 lines cut ...]
Merged 1 ELF section
Successfully created esp32s3 image.
Compiling .pio\build\ESP-LCD\lib119\ESP32_IO_Expander\ESP_IOExpander.cpp.o
Compiling .pio\build\ESP-LCD\lib119\ESP32_IO_Expander\base\esp_io_expander.c.o
Compiling .pio\build\ESP-LCD\lib119\ESP32_IO_Expander\chip\CH422G.cpp.o
Compiling .pio\build\ESP-LCD\lib119\ESP32_IO_Expander\chip\HT8574.cpp.o
Compiling .pio\build\ESP-LCD\lib119\ESP32_IO_Expander\chip\TCA95xx_16bit.cpp.o
Compiling .pio\build\ESP-LCD\lib119\ESP32_IO_Expander\chip\TCA95xx_8bit.cpp.o
Compiling .pio\build\ESP-LCD\lib119\ESP32_IO_Expander\private\CheckResult.cpp.o
Compiling .pio\build\ESP-LCD\lib80b\ESP32_Display_Panel\ESP_Panel.cpp.o
Archiving .pio\build\ESP-LCD\lib119\libESP32_IO_Expander.a
Compiling .pio\build\ESP-LCD\lib80b\ESP32_Display_Panel\backlight\ESP_PanelBacklight.cpp.o
Compiling .pio\build\ESP-LCD\lib80b\ESP32_Display_Panel\bus\ESP_PanelBus.cpp.o
Compiling .pio\build\ESP-LCD\lib80b\ESP32_Display_Panel\bus\I2C.cpp.o
Compiling .pio\build\ESP-LCD\lib80b\ESP32_Display_Panel\bus\QSPI.cpp.o
Compiling .pio\build\ESP-LCD\lib80b\ESP32_Display_Panel\bus\RGB.cpp.o
.pio/libdeps/ESP-LCD/ESP32_Display_Panel/src/backlight/ESP_PanelBacklight.cpp: In constructor 'ESP_PanelBacklight::ESP_PanelBacklight(int, bool, bool)':
.pio/libdeps/ESP-LCD/ESP32_Display_Panel/src/backlight/ESP_PanelBacklight.cpp:18:92: error: no matching function for call to 'ledc_timer_config_t::ledc_timer_config_t(<brace-enclosed initializer list>)'
     _channel_config(ESP_PANEL_BACKLIGHT_LEDC_CHANNEL_CONFIG_DEFAULT(io_num, light_up_level))
                                                                                            ^
In file included from .pio/libdeps/ESP-LCD/ESP32_Display_Panel/src/backlight/ESP_PanelBacklight.h:9,
                 from .pio/libdeps/ESP-LCD/ESP32_Display_Panel/src/backlight/ESP_PanelBacklight.cpp:8:
C:/Users/Ifou/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32s3/include/driver/include/driver/ledc.h:56:3: note: candidate: 'ledc_timer_config_t::ledc_timer_config_t()'
 } ledc_timer_config_t;
   ^~~~~~~~~~~~~~~~~~~

isthaison, 2024-10-13:
C:/Users/Ifou/.platformio/packages/framework-arduinoespressif32@src-537c58760daầe7fcc8a1d9bbcf00b6f6 
replace in here