esp-arduino-libs/ESP32_Display_Panel #108: Can I use this library with axs15231b directly?

HerbertHe, 2024-09-09:
Can I use this library with axs15231b directly? And is there any demos can be provided?

Lzw655, 2024-09-10:
Hi @HerbertHe, 

This library currently does not support the axs15231b. The [driver](https://components.espressif.com/components/espressif/esp_lcd_axs15231b) and [example](https://github.com/espressif/esp-iot-solution/tree/master/examples/display/lcd/lcd_with_te) for this LCD are only supported in esp-idf.

HerbertHe, 2024-09-10:
Ok... I have read the source code of this project, and I found that only the listed chips can be supported 😢 

The esp-idf is really hard to use due to the network and repo size, I got failed too many times with my MacOS 😭 

Maybe I can just activate the LCD with esp-idf, or to solve to errors of demos from the seller.

Lzw655, 2024-09-14:
Sorry for the issue you encountered. If the library adds support for this LCD model, I will let you know.

HerbertHe, 2024-09-19:
@Lzw655 Thank you!

closed 2025-02-18 by Lzw655