esp-arduino-libs/ESP32_Display_Panel #111: Error: Invalid Height

YOTOTOY, 2024-09-21:
Dear:
I'm going to use this library on ESP32C6, everying goes will but it shows "Invalid Height" and reboot forever.
I don't know the reason, could you please help me to fix this?
Thank you!

Log:
Initialize lvgl
Create and begin phone
[WARN] [esp_ui_phone.cpp:66](begin): No phone stylesheet is added, adding default dark stylesheet(EUI-Phone Default Dark)
[WARN] [esp_ui_template.hpp:107](addStylesheet): Display is not set, use default display
[ERROR][esp_ui_core_home.cpp:128](calibrateCoreObjectSize): Invalid value: 24, should be in range [1, 0]
[ERROR][esp_ui_core_home.cpp:128](calibrateCoreObjectSize): Invalid Height
[ERROR][esp_ui_status_bar.cpp:205](calibrateData): Calibrate data main size min failed
[ERROR][esp_ui_phone_home.cpp:364](calibrateData): Calibrate status bar data failed
[ERROR][esp_ui_phone.cpp:46](calibrateStylesheet): Invalid home data
[ERROR][esp_ui_template.hpp:115](addStylesheet): Invalid stylesheet
[ERROR][esp_ui_phone.cpp:125](addStylesheet): Failed to add phone stylesheet
[ERROR][esp_ui_phone.cpp:68](begin): Failed to add default stylesheet
[ERROR][ESP_UI_Phone.ino:152](setup): Begin phone failed
  Biggest /     Free /    Total
SRAM : [  311284 /   330580 /   414268]
PSRAM : [       0 /        0 /        0]
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Lzw655, 2024-09-23:
Hi @YOTOTOY, thank you for using the esp-ui library. The error is caused by the wrong stylesheet parameters. Did you use a custom stylesheet and try to add it? If not, what screen resolution are you using?

YOTOTOY, 2024-09-23:
没有使用额外的样式表，驱动IC是ST７７８９V，分辨率２４０ｘ２８０，我选了ST７７８９。并且设置了分辨率。

Lzw655, 2024-09-24:
了解，这个问题是软件近期引入的一个 bug，如果没有匹配的样式表就会出现。非常抱歉，esp-ui 现在更名为了 [esp-brookesia](https://github.com/espressif/esp-brookesia)（当前还没来及上传至 Arduino，可以手动 download zip 进行使用），麻烦你使用最新的版本，现在已经修复了这个问题。

YOTOTOY, 2024-09-24:
好的，收到。感谢回复。我按照issue用platformio重新编译了一个工程，能跑起来了，但是同样的存在一个问题，设置280 的屏幕高度显示不完全，似乎是整体上移了20个像素，设置成300刚好填充满整个屏幕，但是屏幕最上方似乎缺失了一部分，可能是我的配置问题吧。

closed 2024-09-26 by YOTOTOY