esp-arduino-libs/ESP32_Display_Panel #112: DrawBitmap crashing if logging not enabled

NecroMancer05, 2024-09-29:
Hello,

I am using LVGL with this library with PlatformIO. I have enabled all logs with adding CORE_DEBUG_LEVEL=5 to build flags in platformio.ini file.
But after removing CORE_DEBUG_LEVEL line im getting crash on `panel->getLcd()->drawBitmap` call. Also its crashing below log leve < 3 like CORE_DEBUG_LEVEL=2, CORE_DEBUG_LEVEL=1.

[image]

Is it related to me or someting wrong with the library?

Thanks.

Lzw655, 2024-10-08:
Hi @NecroMancer05, could you please send the serial output log from when the crash occurred?

NecroMancer05, 2024-10-08:
Hi @Lzw655,

Now i found the issue. 
i have a log call like this `log_i("Lcd init %d", panel->init());` and after disabling logs panel::init function not calling so thats why im getting crash 😂

Thanks.

closed 2024-10-08 by NecroMancer05