esp-arduino-libs/ESP32_Display_Panel #113: Waveshare round 1.85 ESP32S3 LCD : failure to initialise touch

martinroger, 2024-10-22:
Hi, I am trying to get the library to work on a [Waveshare round lcd](https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-1.85).
I can get the QSPI LCD to start just fine, however when I try to enable the CST816S it fails to read from it.

You can find the platformio project I set up for this (including special board JSON, lib_deps etc) and the commit where the QSPI works but the I2C fails to initialise : [here](https://github.com/martinroger/WS-ESP32S3-RoundLCDTouch1_85/commit/8e29647a4ef2e8a881797a1af38d8b8d931e3049)

I have tried with and without pullups enabled (the schematic seems to indicate there are none on the circuit, and as far as I can tell this is a dedicated I2C line.

To note, the reset for the TP is on an IO Expander that I am currently not activating.

Obviously ultimately I can circumvent and initialise it all separately, but I like the all-in-one approach to the library. Am I missing something or ultimately am I forced to do a manual init / separate management of touch input?

The logs I get through serial, showing the initialisation fail : 

```
[   316][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 44 successfully set to type UART_RX (2) with bus 0x3fc945c4
  #0  0x3fc945c4 in ?? at C:/Users/Martin/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:82

[   327][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 43 successfully set to type UART_TX (3) with bus 0x3fc945c4
  #0  0x3fc945c4 in ?? at C:/Users/Martin/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:82

[   337][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 19 successfully set to type USB_DM (45) with bus 0x3fc97648
  #0  0x3fc97648 in ?? at C:/Users/Martin/.platformio/packages/framework-arduinoespressif32/cores/esp32/HWCDC.cpp:610

[   348][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 20 successfully set to type USB_DP (46) with bus 0x3fc97648
  #0  0x3fc97648 in ?? at C:/Users/Martin/.platformio/packages/framework-arduinoespressif32/cores/esp32/HWCDC.cpp:610

=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-S3
  Package           : 0
  Revision          : 0.02
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Features Bitfield : 0x00000012
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   386676 B ( 377.6 KB)
  Free Bytes        :   354432 B ( 346.1 KB)
  Allocated Bytes   :    27244 B (  26.6 KB)
  Minimum Free Bytes:   349384 B ( 341.2 KB)
  Largest Free Block:   319476 B ( 312.0 KB)
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
  Total Size        :  8388608 B (8192.0 KB)
  Free Bytes        :  8385672 B (8189.1 KB)
  Allocated Bytes   :      576 B (   0.6 KB)
  Minimum Free Bytes:  8385672 B (8189.1 KB)
  Largest Free Block:  8257524 B (8064.0 KB)
  Bus Mode          : OPI
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         : 16777216 B (16 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
Partitions Info:
[... 464 lines cut ...]
// #define ESP_PANEL_BEGIN_BACKLIGHT_END_FUNCTION( panel )
// #define ESP_PANEL_BEGIN_END_FUNCTION( panel )

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
/////////////////////////////////////////////// File Version ///////////////////////////////////////////////////////////
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
/**
 * Do not change the following versions, they are used to check if the configurations in this file are compatible with
 * the current version of `ESP_Panel_Board_Custom.h` in the library. The detailed rules are as follows:
 *
 *   1. If the major version is not consistent, then the configurations in this file are incompatible with the library
 *      and must be replaced with the file from the library.
 *   2. If the minor version is not consistent, this file might be missing some new configurations, which will be set to
 *      default values. It is recommended to replace it with the file from the library.
 *   3. Even if the patch version is not consistent, it will not affect normal functionality.
 *
 */
#define ESP_PANEL_BOARD_CUSTOM_FILE_VERSION_MAJOR 0
#define ESP_PANEL_BOARD_CUSTOM_FILE_VERSION_MINOR 2
#define ESP_PANEL_BOARD_CUSTOM_FILE_VERSION_PATCH 2

#endif /* ESP_PANEL_USE_CUSTOM_BOARD */

// *INDENT-OFF*
```

martinroger, 2024-10-23:
Hey @Lzw655 , I realised like an idiot that I left my repo private, I will change that asap. And yes in the meantime I have started suspecting that the expander constantly resets the touch.
Is there a way to init and setup the expander before anything else in the panel class? Or I guess I need to init all three components separately and assign them to a panel after?

Lzw655, 2024-10-23:
I think you can set `#define ESP_PANEL_USE_EXPANDER  (1)` and `#define ESP_PANEL_EXPANDER_SKIP_INIT_HOST (0)` in *ESP_Panel_Board_Custom.h*, then initialize the IO Expander before initializing the panel, and use `panel->configExpander()` to add it to the panel. For example:

```c
...
// Create IO Expander first (don't initialize it)
ESP_IOExpander *expander = new ESP_IOExpander_TCA95xx_8bit(...);
...
// Then create and initialize the panel
ESP_Panel *panel = new ESP_Panel();
panel->configExpander(expander);  // Should be called before `init()`
panel->init();
panel->begin();
...
```

martinroger, 2024-10-23:
Will try and report!

martinroger, 2024-10-23:
By the way the commit and repo should be visible now. Will try the trick you suggested later.

martinroger, 2024-10-23:
Update (and solution, I'ld say) in commit [4a2cc8a](https://github.com/martinroger/WS-ESP32S3-RoundLCDTouch1_85/commit/4a2cc8aff93b2467ecc7b2199ca589522c135582) : 

- The first thing to fix was that I didn't understand I2C host numbering in the config file. The TP and the Expander are on two physically separate I2C channels, but both were initialised to host 0. Obviously that can't work.
- After setting TP and expander on two separate hosts, I was still getting initialisation issues for the TP. I started digging in the `ESP_Panel.cpp` code in its begin() function, and I noticed the **ESP_PANEL_BEGIN_EXPANDER_END_FUNCTION** macro for inserting code. Lucky for me the expander is initialised before the TP, so all I had to do was inject a quick pinMode and flipping of the expander IO assigned to the TP reset, and the CST816S initialised correctly
- In the end, the trick @Lzw655 recommended worked, but I prefer this even-cleaner solution !

My **ESP_Panel_Board_Custom.h** : 

```
/*
 * SPDX-FileCopyrightText: 2023-2024 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#pragma once

// *INDENT-OFF*

/* Set to 1 if using a custom board */
#define ESP_PANEL_USE_CUSTOM_BOARD       (1)         // 0/1

#if ESP_PANEL_USE_CUSTOM_BOARD

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
//////////////////////////// Please update the following macros to configure the LCD panel /////////////////////////////
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
/* Set to 1 when using an LCD panel */
#define ESP_PANEL_USE_LCD           (1)     // 0/1

#if ESP_PANEL_USE_LCD
/**
 * LCD Controller Name. Choose one of the following:
 *      - EK9716B
 *      - GC9A01, GC9B71, GC9503
 *      - ILI9341
 *      - NV3022B
 *      - SH8601
 *      - SPD2010
 *      - ST7262, ST7701, ST7789, ST7796, ST77916, ST77922
 */
#define ESP_PANEL_LCD_NAME          ST77916

/* LCD resolution in pixels */
#define ESP_PANEL_LCD_WIDTH         (360)
#define ESP_PANEL_LCD_HEIGHT        (360)

/* LCD Bus Settings */
/**
 * If set to 1, the bus will skip to initialize the corresponding host. Users need to initialize the host in advance.
 * It is useful if other devices use the same host. Please ensure that the host is initialized only once.
 *
 * Set to 1 if only the RGB interface is used without the 3-wire SPI interface,
 */
#define ESP_PANEL_LCD_BUS_SKIP_INIT_HOST    (0)     // 0/1
/**
 * LCD Bus Type. Choose one of the following:
 *      - ESP_PANEL_BUS_TYPE_I2C (not ready)
 *      - ESP_PANEL_BUS_TYPE_SPI
 *      - ESP_PANEL_BUS_TYPE_QSPI
 *      - ESP_PANEL_BUS_TYPE_I80 (not ready)
 *      - ESP_PANEL_BUS_TYPE_RGB (only supported for ESP32-S3)
 */
#define ESP_PANEL_LCD_BUS_TYPE      (ESP_PANEL_BUS_TYPE_QSPI)
/**
 * LCD Bus Parameters.
 *
 * Please refer to https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/lcd.html and
 * https://docs.espressif.com/projects/esp-iot-solution/en/latest/display/lcd/index.html for more details.
 *
[... 488 lines cut ...]
  Total Size        :   386092 B ( 377.0 KB)
  Free Bytes        :   343044 B ( 335.0 KB)
  Allocated Bytes   :    37200 B (  36.3 KB)
  Minimum Free Bytes:   327960 B ( 320.3 KB)
  Largest Free Block:   311284 B ( 304.0 KB)
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
  Total Size        :  8388608 B (8192.0 KB)
  Free Bytes        :  8385012 B (8188.5 KB)
  Allocated Bytes   :     1156 B (   1.1 KB)
  Minimum Free Bytes:  8369156 B (8173.0 KB)
  Largest Free Block:  8257524 B (8064.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
  GPIO : BUS_TYPE[bus/unit][chan]
  --------------------------------------
     5 : LEDC[0][0]
    19 : USB_DM
    20 : USB_DP
    43 : GPIO
    44 : GPIO
============ After Setup End =============
```

Lzw655, 2024-10-24:
@martinroger Thank you for your testing. Your solution is indeed better. Macros like `ESP_PANEL_BEGIN_EXPANDER_END_FUNCTION` are specifically designed for such needs.

If it's convenient, could you please contribute the relevant information for this development board to the library? You can refer to the [documentation](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/docs/Board_Contribution_Guide.md) for guidance.

martinroger, 2024-10-24:
@Lzw655 it is ultimately the intent, yes. This and a couple other boards!

Lzw655, 2024-10-25:
Thank you very much!

martinroger, 2024-10-25:
PR #115 is out there !

PR #115 (2024-10-26): Adding support for Waveshare ESP32-S3-Touch-LCD-1.85
This should contain all the necessary changes to add support for the touch version of the [1.85in touch LCD ESP32S3 module from Waveshare](https://www.waveshare.com/esp32-s3-touch-lcd-1.85.htm).

- Closes #113
closed 2024-10-26 by Lzw655