esp-arduino-libs/ESP32_Display_Panel #114: What does the Panel test look like on the screen?

CoTECH1987, 2024-10-24:
Hi. I have compiled a test sketch of Test Panel.ino
from examples of the latest version of the ESP32_Display_Panel library.
But it seems to me that the example is not displayed correctly.
Maybe someone has a picture of what the sample should actually look like after compilation?
[image]
I'm a little unsure about the black rectangle and the thin white stripe on top

Lzw655, 2024-10-25:
Hi @CoTECH1987, Could you please share your test sketch of Test Panel.ino?

CoTECH1987, 2024-10-25:
[TEST_PANEL_TYPE_RGB.zip](https://github.com/user-attachments/files/17520591/TEST_PANEL_TYPE_RGB.zip)

CoTECH1987, 2024-10-25:
I have set up my sketch for https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-5
this is a parallel RGB interface.

CoTECH1987, 2024-10-25:
[image]

I think I've figured out the black square.
I changed the bus settings for the 1024x600 LCD as follows:

`#elif ESP_PANEL_LCD_BUS_TYPE == ESP_PANEL_BUS_TYPE_RGB

    #define ESP_PANEL_LCD_RGB_CLK_HZ            (20 * 1000 * 1000)
    #define ESP_PANEL_LCD_RGB_HPW               (88)
    #define ESP_PANEL_LCD_RGB_HBP               (188)
    #define ESP_PANEL_LCD_RGB_HFP               (44)
    #define ESP_PANEL_LCD_RGB_VPW               (6)
    #define ESP_PANEL_LCD_RGB_VBP               (16)
    #define ESP_PANEL_LCD_RGB_VFP               (3)
    #define ESP_PANEL_LCD_RGB_PCLK_ACTIVE_NEG   (1) `

But I'm still worried about the white stripe on the top and bottom now.
And I noticed that every time I click on the touchpad, my graphics on the display slides by several pixels, this can be seen if you zoom in on the picture.
Any help?

CoTECH1987, 2024-10-25:
[image]

If you keep the touchpad pressed, this session will happen all the time!

CoTECH1987, 2024-10-25:
It's been decided. The offset was due to the internal features of the esp-sdk. Installing the patched version solved the problem! thank you!

CoTECH1987, 2024-10-25:
!!!

closed 2024-10-25 by CoTECH1987