esp-arduino-libs/ESP32_Display_Panel #116: I couldn't create a simple button on the screen

CoTECH1987, 2024-10-26:
I couldn't create a simple button on the screen.

Maybe someone has a basic sketch, an example, to create graphics using this library, because I couldn't figure out the complex examples of "Porting".
 I need a primitive code to display any of the components on the screen.
Please?

Lzw655, 2024-10-27:
Hi @CoTECH1987, This library primarily provides LCD and related low-level drivers. If you want to use the `Porting` example to render a GUI on the screen, it needs to be used in conjunction with the `LVGL` library. Detailed configuration instructions can be found in `Porting.ino` and its `README`. For LVGL API usage, please refer to the [official examples](https://docs.lvgl.io/8.3/examples.html#a-button-with-a-label-and-react-on-click-event).

martinroger, 2024-10-28:
> I couldn't create a simple button on the screen.
> 
> Maybe someone has a basic sketch, an example, to create graphics using this library, because I couldn't figure out the complex examples of "Porting". I need a primitive code to display any of the components on the screen. Please?

What screen hardware are we talking about ? Can you get the BGR test example to run on it ?