esp-arduino-libs/ESP32_Display_Panel #122: request add waveshare ESP32-S3-LCD-1.47

springeye, 2024-11-10:
First of all, thank you very much for providing this library, which allows me to successfully run waveshare's ESP32-S3-Touch-LCD-4.3

I am a beginner. I don't know much about custom development boards. So can you please add support https://www.waveshare.com/wiki/ESP32-S3-LCD-1.47

Lzw655, 2024-11-11:
Hi @springeye, 

thank you very much for using this library. However, since I don't have the hardware for this development board, I'm unable to perform functional testing. Perhaps @martinroger and @Y1hsiaochunnn could help us with this.

martinroger, 2024-11-11:
Hi @springeye , I don't have the hardware but I can try to give it a look later tonight (CET). Have you ever used this library or is it your first time?
What is your preferred IDE?

springeye, 2024-11-11:
@martinroger 
Wǒ dì yī cì shǐyòng zhège kù, wǒ shǐyòng de shì platformio(framework = arduino)
46 / 5,000
I'm using this library for the first time, I'm using platformio(framework = arduino)

martinroger, 2024-11-11:
@springeye perfect. I prepared a branch of one of my "sandbox" repos for you to try and see if it works on your hardware : https://github.com/martinroger/WS-ESP32S3-RoundLCDTouch1_85/tree/springeye1_47

Please note that this contains everything that should be needed : 

- Board JSON definition file
- Platformio pointing to a version of the ArduinoESP >3.0.6
- Simple `main.cpp` file that does a B-G-R line test to check LCD functionality and shows one way to initialise the LCD
- In the include folder, `ESP_Panel_Board_Custom.h` configured correctly (I hope).

[image]

Some caveats : 

- In the configuration file, I set some custom LCD init commands as per what Waveshare is doing in their original demo software : 
``` #define ESP_PANEL_LCD_VENDOR_INIT_CMD()                                        \
    {                                                                           \
        ESP_PANEL_LCD_CMD_WITH_NONE_PARAM(120, 0x11),                           \
        {0x36, (uint8_t []){0x00}, 1, 0},                                       \
        {0x3A, (uint8_t []){0x05}, 1, 0},                                       \
        {0xB0, (uint8_t []){0x00, 0xE8}, 2, 0},                                 \
        {0xB2, (uint8_t []){0x0C, 0x0C, 0x00, 0x33, 0x33}, 5, 0},               \
        {0xB7, (uint8_t []){0x35}, 1, 0},                                       \
        {0xBB, (uint8_t []){0x35}, 1, 0},                                       \
        {0xC0, (uint8_t []){0x2C}, 1, 0},                                       \
        {0xC2, (uint8_t []){0x01}, 1, 0},                                       \
        {0xC3, (uint8_t []){0x13}, 1, 0},                                       \
        {0xC4, (uint8_t []){0x20}, 1, 0},                                       \
        {0xC6, (uint8_t []){0x0F}, 1, 0},                                       \
        {0xD0, (uint8_t []){0xA4, 0xA1}, 2, 0},                                 \
        {0xE0, (uint8_t []){0xF0, 0x00, 0x04, 0x04, 0x04, 0x05, 0x29, 0x33, 0x3E, 0x38, 0x12, 0x12, 0x28, 0x30}, 14, 0}, \
        {0xE1, (uint8_t []){0xF0, 0x07, 0x0A, 0x0D, 0x0B, 0x07, 0x28, 0x33, 0x3E, 0x36, 0x14, 0x14, 0x29, 0x32}, 14, 0}, \
        ESP_PANEL_LCD_CMD_WITH_NONE_PARAM(0, 0x21),                             \
        ESP_PANEL_LCD_CMD_WITH_NONE_PARAM(120, 0x11),                           \
        ESP_PANEL_LCD_CMD_WITH_NONE_PARAM(0, 0x29),                             \
    }
``` 
... it might work without it.
- I am unsure if/how the touch is handled on this unit. Please report, but I think this current library is not designed to handle this type of touch.
- I didn't implement anything for the RGB LED (you can use the builtin `neopixelWrite()`) or the SD card (there are many libraries for this)

**How to :**
Please clone the repo on your computer, switch to the _springeye1_47_ branch and open it with VSCode. It should take 5-10 minutes (and requires internet) to install the framework packages. Once Platformio is done and ready, compile and flash and open a _Serial Monitor_ to your board and report what comes out in the console. If it doesn't work or constantly reboots, please let us know here.

Y1hsiaochunnn, 2024-11-12:
@martinroger Thank you for your work. In fact, the register initialization content should be retained. This is the recommended configuration of the screen, and it may affect the display if it is lost or modified

springeye, 2024-11-12:
My development board was damaged by my daughter last night, so I bought a new one for verification (the delivery takes about 2-3 days).

springeye, 2024-11-26:
There is a problem, the screen display is not complete
[image]

martinroger, 2024-11-26:
Ok, this sounds like a problem of orientation, the horizontal and vertical resolutions look like they are swapped. The weird stuff at the bottom is just random data.
I can try tonight to provide a fix.

martinroger, 2024-11-26:
@springeye if you are feeling really adventurous, I had a go at modifying the init sequence of the LCD in commit [e2a4367b54db0d07ae17281d1272dbd9ab67181c](https://github.com/martinroger/WS-ESP32S3-RoundLCDTouch1_85/commit/e2a4367b54db0d07ae17281d1272dbd9ab67181c) . This might fix things but will lock things in one specific orientation. 
Full disclosure : I have never tried that before and I don't have the same hardware as yours to check it does the intended effect. Please report the results. 

@Lzw655 if you are more experienced than I am with ST7789 set-up commands, have a look and see if I did something right or wrong, please !

Lzw655, 2024-11-27:
> @springeye if you are feeling really adventurous, I had a go at modifying the init sequence of the LCD in commit [...]

I’m not very familiar with the ST7789 commands either, but after reviewing the demo project provided by Waveshare, I agree with using these new initialization parameters.