esp-arduino-libs/ESP32_Display_Panel #129: 立创实战派esp32s3 在arduino ide上使用此库自带的lvgl历程时,程序运行但屏幕无法显示,请大佬帮帮忙

chunyu-just, 2024-11-26:
程序编译下载完后的串口信息为
```c
22:38:59.682 -> D (88) ESP_Panel: Begin IO expander
22:38:59.682 -> D (92) ESP_Panel: Begin LCD
22:38:59.682 -> D (94) ESP_PanelBus_SPI: Create panel io @0x3fcecc70
22:38:59.682 -> D (98) st7789: new st7789 panel @0x3fced248
22:38:59.682 -> I (102) st7789: LCD panel create success, version: 0.0.1
22:38:59.727 -> D (128) ESP_PanelLcd: Begin start
22:38:59.803 -> D (229) st7789: send init commands success
22:38:59.803 -> D (229) ESP_PanelLcd: Begin end
22:38:59.803 -> D (230) ESP_Panel: Begin backlight
22:38:59.803 -> D (230) ESP_PanelBacklight: begin start
22:38:59.803 -> D (231) ESP_PanelBacklight: Use PWM(LEDC) to control
22:38:59.803 -> D (236) ESP_PanelBacklight: begin end
22:38:59.835 -> D (240) ESP_PanelBacklight: Set brightness to 0%
22:38:59.835 -> D (244) ESP_Panel: Panel begin end
22:38:59.835 -> Initialize LVGL
22:38:59.881 -> Create UI
22:38:59.881 -> LVGL porting example end
22:38:59.881 -> IDLE loop
22:39:00.921 -> IDLE loop
22:39:01.882 -> IDLE loop
22:39:02.898 -> IDLE loop
```
应该是程序能够运行了,但是我的开发板的屏幕不显示,黑屏,背光也没有
立创实战派s3版本的tft屏幕(使用st7789驱动)的原理图为
[image]
他的屏幕的片选引脚使用拓展io来控制(这里我不会在配置文件中设置了,所以我片选引脚设置的 -1)
[image]
arduino上的下载方式能是设置为
[image]
esp32核心版本为
[image]
相关库版本为
[image]
[image]
我将代码上传到此处
[Porting_test.zip](https://github.com/user-attachments/files/17921114/Porting_test.zip)

我想使用arduino来点亮这个开发板的屏幕已经好多天了,试了很多方法都没成功,希望有朋友能够看看我的代码配置,提供些建议或者解决方法,非常感谢

Lzw655, 2024-11-27:
@chunyu-just 你好！

我看了下附件里的工程（[Porting_test.zip](https://github.com/user-attachments/files/17921114/Porting_test.zip)），关于背光没有亮的问题，是因为设置了 `ESP_PANEL_BACKLIGHT_IDLE_OFF` 为 `1`，它会使得开发板驱动初始化时默认关闭背光，在如下 LOG 中可以看到相关信息：`22:38:59.835 -> D (240) ESP_PanelBacklight: Set brightness to 0%`。

请将 `ESP_PANEL_BACKLIGHT_IDLE_OFF` 设置为 `0`，然后再看下显示是否正常。

chunyu-just, 2024-11-27:
> @chunyu-just 你好！
> 
> 我看了下附件里的工程（[Porting_test.zip](https://github.com/user-attachments/files/17921114/Porting_test.zip)），关于背光没有亮的问题，是因为在文件 _ESP_Panel_Board_Custom.h_ 中设置了 `ESP_PANEL_BACKLIGHT_IDLE_OFF` 为 `1`，它会使得开发板驱动初始化时默认关闭背光，在如下 LOG 中可以看到相关信息：`22:38:59.835 -> D (240) ESP_PanelBacklight: Set brightness to 0%`。
> 
> 请将 `ESP_PANEL_BACKLIGHT_IDLE_OFF` 设置为 `0`，然后再看下显示是否正常。

谢谢你的建议,我修改了对应的参数后,背光可以亮了,但是屏幕还是没有显示
```c
17:50:24.659 -> Initialize panel device
17:50:24.659 -> D (6) ESP_Panel: Panel init start
17:50:24.693 -> D (7) ESP_Panel: Use LCD
17:50:24.747 -> D (57) ESP_Panel: Use SPI bus
17:50:24.747 -> D (57) ESP_Panel: Create LCD bus
17:50:24.747 -> D (57) ESP_PanelHost: Add host SPI[1]
17:50:24.747 -> D (57) ESP_Panel: Create LCD device
17:50:24.747 -> D (58) ESP_Panel: Use backlight
17:50:24.747 -> D (61) ESP_Panel: Create backlight device
17:50:24.747 -> D (64) ESP_Panel: Use IO expander
17:50:24.747 -> D (67) ESP_PanelHost: Add host I2C[0]
17:50:24.747 -> D (71) ESP_Panel: Initialize host
17:50:24.765 -> D (74) ESP_PanelHost: Initialize host I2C[0]
17:50:24.765 -> D (78) ESP_PanelHost: Initialize host SPI[1]
17:50:24.765 -> D (82) ESP_Panel: Panel init end
17:50:24.765 -> D (85) ESP_Panel: Panel begin start
17:50:24.765 -> D (88) ESP_Panel: Begin IO expander
17:50:24.810 -> D (92) ESP_Panel: Begin LCD
17:50:24.810 -> D (94) ESP_PanelBus_SPI: Create panel io @0x3fcecc70
17:50:24.810 -> D (94) ESP_PanelBus_SPI: Create panel io @0x3fcecc70
17:50:24.810 -> D (98) st7789: new st7789 panel @0x3fced248
17:50:24.810 -> I (102) st7789: LCD panel create success, version: 0.0.1
17:50:24.810 -> D (128) ESP_PanelLcd: Begin start
17:50:25.028 -> D (229) st7789: send init commands success
17:50:25.028 -> D (229) ESP_PanelLcd: Begin end
17:50:25.028 -> D (230) ESP_Panel: Begin backlight
17:50:25.028 -> D (230) ESP_PanelBacklight: begin start
17:50:25.028 -> D (231) ESP_PanelBacklight: Use PWM(LEDC) to control
17:50:25.028 -> D (236) ESP_PanelBacklight: begin end
17:50:25.028 -> D (240) ESP_PanelBacklight: Set brightness to 100%
17:50:25.028 -> D (244) ESP_Panel: Panel begin end
17:50:25.028 -> Initialize LVGL
17:50:25.028 -> Create UI
17:50:25.028 -> LVGL porting example end
17:50:25.028 -> IDLE loop
17:50:25.949 -> IDLE loop
17:50:26.985 -> IDLE loop
```

Lzw655, 2024-11-27:
LCD 的 CS 由 IO 扩展芯片的引脚控制且默认拉高，但是主程序并未控制 IO 扩展芯片将 CS 引脚拉低（LCD 的 SPI 通信要求），请在 `panel->init();` 和 `panel->begin();` 之间通过 `panel->getExpander()` 进行控制，参考示例（https://github.com/esp-arduino-libs/ESP32_IO_Expander/blob/master/README.md#examples）