esp-arduino-libs/ESP32_Display_Panel #130: A couple of code errors in ESP_PanellLcd.cpp

mhavill, 2024-11-28:
I accidentally found the following two errors in the coding of /esp-arduino-libs/ESP32_Display_Panel/blob/master/src/lcd/ESP_PanelLcd.cpp  lines 447 onward when I missed setting CONFIG_SPIRAM_RODATA true.
```
    /* Check the callback function and user data placement */
    // For RGB LCD, if the "XIP on PSRAM" function is not enabled, the callback function and user data should be
    // placed in SRAM
#if (SOC_MIPI_DSI_SUPPORTED && CONFIG_LCD_DSI_ISR_IRAM_SAFE) || \
    (SOC_LCD_RGB_SUPPORTED && CONFIG_LCD_RGB_ISR_IRAM_SAFE && !(CONFIG_SPIRAM_RODATA && CONFIG_SPIRAM_FETCH_INSTRUCTIONS))
    if (bus->getType() == ESP_PANEL_BUS_TYPE_RGB || bus->getType() == ESP_PANEL_BUS_TYPE_MIPI_DSI) {
        ESP_PANEL_CHECK_FALSE_RET(
            esp_ptr_in_iram(callback), false,
            "Callback function should be placed in IRAM, add `IRAM_ATTR` before the function"
        );
        ESP_PANEL_CHECK_FALSE_RET((esp_ptr_internal(user_data), false, "User data should be placed in SRAM"));
    }
#endif
```
1) on the line 
```
        ESP_PANEL_CHECK_FALSE_RET( esp_ptr_in_iram(callback), false, "Callback function should be placed in IRAM, add `IRAM_ATTR` before the function"   );
```
Gives the error **cannot convert 'std::function<bool(void*)>' to 'const void*'**

2)  the line 
```
        ESP_PANEL_CHECK_FALSE_RET((esp_ptr_internal(user_data), false, "User data should be placed in SRAM"));
```
Has an extra pair of brackets so it is reported as a single argument when 4 are required. Should be 
```
        ESP_PANEL_CHECK_FALSE_RET(esp_ptr_internal(user_data), false, "User data should be placed in SRAM");
```

Lzw655, 2024-11-28:
Hi @mhavill, thank you very much for pointing out this issue! I will fix it in the next version.

BTW, are you using it based on ESP-IDF? Because the issue will only be triggered when enabling `CONFIG_LCD_DSI_ISR_IRAM_SAFE` or `CONFIG_LCD_RGB_ISR_IRAM_SAFE `.

mhavill, 2024-11-28:
Hi. 
I am using it on ESP-IDF for a Waveshare ESP32-S3-LCD-Touch-4.3 with LVGL v9
Thank you for your work!

closed 2025-02-18 by Lzw655