esp-arduino-libs/ESP32_Display_Panel #131: Backlight control requirements

Y1hsiaochunnn, 2024-11-29:
According to contact with a variety of screen backlight control, can be classified as the following:

- GPIO Control
    - High and Low Levels:
         The backlight is controlled by a GPIO pin directly, where setting it to high turns on the backlight, and setting it to low turns it off.
    - PWM Signal Control:
         The backlight is controlled by a GPIO pin generating a PWM signal, with the brightness adjusted according to the duty cycle.
- Control via I2C Chip
    - The brightness (0x00~0xFF) is adjusted by writing to a specific register of the backlight control chip via the I2C bus.
        - Control Process: The MCU communicates with the backlight control chip over I2C and writes the brightness value to a specific register address.
- Control via QSPI or MIPI-DSI or others
    - The brightness (0x00~0xFF) is adjusted by sending the value to a specific register via the QSPI or MIPI-DSI protocol.
        - QSPI: Uses high-speed serial communication to write to the register.
        - MIPI-DSI: Typically used with integrated display control, enabling direct backlight brightness control through the MIPI protocol.

***
so, i know you have the plan to implement backlight control next. Maybe you can consider the above several types of backlight control scheme? Please let me know if you need anything

Lzw655, 2024-11-29:
Hi @Y1hsiaochunnn, 

Thank you very much for your suggestions. They are extremely comprehensive. Although I haven’t encountered the last method using QSPI or MIPI-DSI protocols for control.

I also realize that the current library’s approach to backlight control, using `Switch (GPIO)` and `PWM (LEDC)` methods, doesn’t meet the requirements of many development boards. Therefore, I am restactoring this library and plan to introduce a `Custom` method on top of the Switch and PWM methods to accommodate different development board requirements. The expected updated configuration file is as follows:

```c
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
///////////////////////////// Please update the following macros to configure the backlight ////////////////////////////
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
#define ESP_PANEL_USE_BACKLIGHT         (1)         // 0/1
#if ESP_PANEL_USE_BACKLIGHT
/**
 * Backlight control mode. Choose one of the following:
 *  - ESP_PANEL_BACKLIGHT_TYPE_SWITCH_GPIO: Use GPIO switch to control the backlight, only support on/off
 *  - ESP_PANEL_BACKLIGHT_TYPE_PWM_LEDC:    Use LEDC PWM to control the backlight, support brightness adjustment
 *  - ESP_PANEL_BACKLIGHT_TYPE_CUSTOM:      Use custom function to control the backlight
 *
 */
#define ESP_PANEL_BACKLIGHT_TYPE        (ESP_PANEL_BACKLIGHT_TYPE_PWM_LEDC)

#if ESP_PANEL_BACKLIGHT_TYPE != ESP_PANEL_BACKLIGHT_TYPE_CUSTOM
/* IO num of backlight pin */
#define ESP_PANEL_BACKLIGHT_IO          (47)
#define ESP_PANEL_BACKLIGHT_ON_LEVEL    (1)         // 0: low level, 1: high level
#else
/**
 * @brief Custom backlight control function
 *
 * @param[in] percent   Brightness percentage, 0-100
 * @param[in] user_data User data, default is a pointer of `Board`
 *
 */
#define ESP_PANEL_BACKLIGHT_CUSTOM_FUNCTION( percent, user_data )  \
    {  \
        Board *board = static_cast<Board *>(user_data);  \
    }
#endif /* ESP_PANEL_BACKLIGHT_TYPE */

/* Set to 1 if you want to turn off the backlight after initializing the panel; otherwise, set it to turn on */
#define ESP_PANEL_BACKLIGHT_IDLE_OFF    (0)         // 0: on, 1: off

#endif /* ESP_PANEL_USE_BACKLIGHT */
```

Lzw655, 2024-11-29:
Great! I'm looking forward to it.

Y1hsiaochunnn, 2024-12-21:
> Hi I have changed the settings in esp32_s3_touch_lcd_7.h as followed
> 
> ```
> #define ESP_PANEL_USE_BACKLIGHT         (1)         // 0/1
> #if ESP_PANEL_USE_BACKLIGHT
> /* IO num of backlight pin */
> #define ESP_PANEL_BACKLIGHT_IO          (2)
> #define ESP_PANEL_BACKLIGHT_ON_LEVEL    (1)         // 0: low level, 1: high level
> 
> /* Set to 1 if turn off the backlight after initializing the panel; otherwise, set it to turn on */
> #define ESP_PANEL_BACKLIGHT_IDLE_OFF    (0)         // 0: on, 1: off
> 
> /* Set to 1 if use PWM for brightness control */
> #define ESP_PANEL_LCD_BL_USE_PWM        (0)         // 0/1
> #endif /* ESP_PANEL_USE_BACKLIGHT */
> ```
> 
> the I am trying to disable the backlight, but it does nothing
> 
> ```
>    ESP_PanelBacklight* backlight = panel->getBacklight();
>    backlight->off();
> ```
> 
> Is that correct or do I missing something?

According to Waveshare's product schematic, the LCD power supply is controlled by the EXIO6 of the IO expansion chip. You can try calling the CH422G call, which is very common.

According to @Lzw655 , the library should now be able to control the backlight with an io extension chip in a future v1.0.0 release, which is partly why I opened the issue. It's the same thing we wanted

[image]

closed 2025-02-18 by Lzw655