esp-arduino-libs/ESP32_Display_Panel #133: 使用ESP-IDF配置ST7701后编译报错

Cathgao, 2024-12-04:
请指教是我哪里配置错了？刚入手ESP32还有很多地方不会
屏使用的是3线SPI+RGB666+背光的接口
屏的规格书：
[046AQV8855ANT.pdf](https://github.com/user-attachments/files/18005812/046AQV8855ANT.pdf)

开发环境：VSCode1.95.3+ESP-IDF5.3.1
报错信息：
```
In file included from C:/Documents/ESP32-S3/rgb_panel/managed_components/espressif__esp32_display_panel/src/lcd/ST7701.h:13,
                 from C:/Documents/ESP32-S3/rgb_panel/managed_components/espressif__esp32_display_panel/src/ESP_Panel_Library.h:40,
                 from C:/Documents/ESP32-S3/rgb_panel/managed_components/espressif__esp32_display_panel/src/panel/ESP_Panel.cpp:9:
C:/Documents/ESP32-S3/rgb_panel/managed_components/espressif__esp32_display_panel/src/panel/ESP_Panel.cpp: In member function 'bool ESP_Panel::init()':
C:/Documents/ESP32-S3/rgb_panel/managed_components/espressif__esp32_display_panel/src/panel/ESP_Panel.cpp:213:72: error: 'ESP_PANEL_LCD_3WIRE_SPI_SCL_ACTIVE_EDGE' was not declared in this scope; did you mean 'ESP_PANEL_LCD_3WIRE_SPI_SCL_USE_EXPNADER'?
  213 |         LCD_PANEL_IO_3WIRE_SPI_CONFIG(ESP_PANEL_LCD_NAME, line_config, ESP_PANEL_LCD_3WIRE_SPI_SCL_ACTIVE_EDGE);
      |                                                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Documents/ESP32-S3/rgb_panel/managed_components/espressif__esp32_display_panel/src/lcd/base/esp_lcd_st7701.h:54:21: note: in definition of macro 'ST7701_PANEL_IO_3WIRE_SPI_CONFIG'
   54 |         .spi_mode = scl_active_edge ? 1 : 0,                        \
      |                     ^~~~~~~~~~~~~~~
C:/Documents/ESP32-S3/rgb_panel/managed_components/espressif__esp32_display_panel/src/panel/ESP_Panel.cpp:41:53: note: in expansion of macro '_LCD_PANEL_IO_3WIRE_SPI_CONFIG'
   41 |                                                     _LCD_PANEL_IO_3WIRE_SPI_CONFIG(name, line_config, scl_active_edge)
      |                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Documents/ESP32-S3/rgb_panel/managed_components/espressif__esp32_display_panel/src/panel/ESP_Panel.cpp:213:9: note: in expansion of macro 'LCD_PANEL_IO_3WIRE_SPI_CONFIG'
  213 |         LCD_PANEL_IO_3WIRE_SPI_CONFIG(ESP_PANEL_LCD_NAME, line_config, ESP_PANEL_LCD_3WIRE_SPI_SCL_ACTIVE_EDGE);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Documents/ESP32-S3/rgb_panel/managed_components/espressif__esp32_display_panel/src/panel/ESP_Panel.cpp:235:34: error: 'ESP_PANEL_LCD_RGB_BOUNCE_BUF_SIZE' was not declared in this scope
  235 |         .bounce_buffer_size_px = ESP_PANEL_LCD_RGB_BOUNCE_BUF_SIZE,
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[1113/1435] Building C object esp-idf/lvgl__lvgl/CMakeFiles/__idf_lvgl__lvgl.dir/src/draw/sdl/lv_draw_sdl_mask.c.obj
ninja: build stopped: subcommand failed.

 *  终端进程“d:\Espressif\tools\ninja\1.11.1\ninja.exe”已终止，退出代码: 1。 
```
配置文件节选：
```
#
# ESP Display Panel Configurations
#
CONFIG_ESP_PANEL_CONF_FILE_SKIP=y
CONFIG_ESP_PANEL_BOARD_FILE_SKIP=y
# CONFIG_ESP_PANEL_CHECK_RESULT_ASSERT is not set
# CONFIG_ESP_PANEL_ENABLE_LOG is not set

#
# LCD touch driver
#
CONFIG_ESP_PANEL_TOUCH_MAX_POINTS=5
CONFIG_ESP_PANEL_TOUCH_MAX_BUTTONS=1

#
# XPT2046
#
CONFIG_ESP_PANEL_TOUCH_XPT2046_Z_THRESHOLD=400
# CONFIG_ESP_PANEL_TOUCH_XPT2046_INTERRUPT_MODE is not set
# CONFIG_ESP_PANEL_TOUCH_XPT2046_VREF_ON_MODE is not set
CONFIG_ESP_PANEL_TOUCH_XPT2046_CONVERT_ADC_TO_COORDS=y
# CONFIG_ESP_PANEL_TOUCH_XPT2046_ENABLE_LOCKING is not set
# end of XPT2046
# end of LCD touch driver

#
# Board
#
# CONFIG_ESP_PANEL_IGNORE_BOARD is not set
# CONFIG_ESP_PANEL_USE_SUPPORTED_BOARD is not set
CONFIG_ESP_PANEL_USE_CUSTOM_BOARD=y
CONFIG_ESP_PANEL_USE_LCD=y

#
# LCD settings
#
# CONFIG_ESP_PANEL_LCD_CONTROLLER_EK9716B is not set
# CONFIG_ESP_PANEL_LCD_CONTROLLER_GC9A01 is not set
# CONFIG_ESP_PANEL_LCD_CONTROLLER_GC9B71 is not set
# CONFIG_ESP_PANEL_LCD_CONTROLLER_GC9503 is not set
# CONFIG_ESP_PANEL_LCD_CONTROLLER_ILI9341 is not set
# CONFIG_ESP_PANEL_LCD_CONTROLLER_NV3022B is not set
# CONFIG_ESP_PANEL_LCD_CONTROLLER_SH8601 is not set
# CONFIG_ESP_PANEL_LCD_CONTROLLER_SPD2010 is not set
# CONFIG_ESP_PANEL_LCD_CONTROLLER_ST7262 is not set
CONFIG_ESP_PANEL_LCD_CONTROLLER_ST7701=y
# CONFIG_ESP_PANEL_LCD_CONTROLLER_ST7789 is not set
# CONFIG_ESP_PANEL_LCD_CONTROLLER_ST7796 is not set
# CONFIG_ESP_PANEL_LCD_CONTROLLER_ST77916 is not set
# CONFIG_ESP_PANEL_LCD_CONTROLLER_ST77922 is not set
CONFIG_ESP_PANEL_LCD_WIDTH=960
CONFIG_ESP_PANEL_LCD_HEIGHT=320

#
# Bus settings
#
# CONFIG_ESP_PANEL_LCD_BUS_SKIP_INIT_HOST is not set
# CONFIG_ESP_PANEL_LCD_BUS_TYPE_SPI is not set
# CONFIG_ESP_PANEL_LCD_BUS_TYPE_QSPI is not set
CONFIG_ESP_PANEL_LCD_BUS_TYPE_RGB=y
CONFIG_ESP_PANEL_LCD_BUS_TYPE=3

#
# RGB bus settings
#

#
# 3-wire SPI interface
#
# CONFIG_ESP_PANEL_LCD_3WIRE_SPI_CS_USE_EXPNADER is not set
# CONFIG_ESP_PANEL_LCD_3WIRE_SPI_SCL_USE_EXPNADER is not set
# CONFIG_ESP_PANEL_LCD_3WIRE_SPI_SDA_USE_EXPNADER is not set
# CONFIG_ESP_PANEL_LCD_FLAGS_AUTO_DEL_PANEL_IO is not set
CONFIG_ESP_PANEL_LCD_FLAGS_MIRROR_BY_CMD=y

#
# Pins
#
CONFIG_ESP_PANEL_LCD_3WIRE_SPI_IO_CS=0
CONFIG_ESP_PANEL_LCD_3WIRE_SPI_IO_SCK=1
CONFIG_ESP_PANEL_LCD_3WIRE_SPI_IO_SDA=2
# end of Pins
# end of 3-wire SPI interface

#
# RGB interface
#
CONFIG_ESP_PANEL_LCD_RGB_CLK_HZ=16000000
CONFIG_ESP_PANEL_LCD_RGB_HPW=10
CONFIG_ESP_PANEL_LCD_RGB_HBP=10
CONFIG_ESP_PANEL_LCD_RGB_HFP=20
CONFIG_ESP_PANEL_LCD_RGB_VPW=10
CONFIG_ESP_PANEL_LCD_RGB_VBP=10
CONFIG_ESP_PANEL_LCD_RGB_VFP=10
# CONFIG_ESP_PANEL_LCD_RGB_PCLK_ACTIVE_NEG is not set
# CONFIG_ESP_PANEL_LCD_RGB_DATA_WIDTH_8 is not set
CONFIG_ESP_PANEL_LCD_RGB_DATA_WIDTH_16=y
CONFIG_ESP_PANEL_LCD_RGB_DATA_WIDTH=16
CONFIG_ESP_PANEL_LCD_RGB_PIXEL_BITS=16

#
# Pins
#
CONFIG_ESP_PANEL_LCD_RGB_IO_HSYNC=46
CONFIG_ESP_PANEL_LCD_RGB_IO_VSYNC=3
CONFIG_ESP_PANEL_LCD_RGB_IO_DE=17
CONFIG_ESP_PANEL_LCD_RGB_IO_PCLK=9
CONFIG_ESP_PANEL_LCD_RGB_IO_DISP=-1
CONFIG_ESP_PANEL_LCD_RGB_IO_DATA0=10
CONFIG_ESP_PANEL_LCD_RGB_IO_DATA1=11
CONFIG_ESP_PANEL_LCD_RGB_IO_DATA2=12
CONFIG_ESP_PANEL_LCD_RGB_IO_DATA3=13
CONFIG_ESP_PANEL_LCD_RGB_IO_DATA4=14
CONFIG_ESP_PANEL_LCD_RGB_IO_DATA5=21
CONFIG_ESP_PANEL_LCD_RGB_IO_DATA6=47
CONFIG_ESP_PANEL_LCD_RGB_IO_DATA7=48
CONFIG_ESP_PANEL_LCD_RGB_IO_DATA8=45
CONFIG_ESP_PANEL_LCD_RGB_IO_DATA9=38
CONFIG_ESP_PANEL_LCD_RGB_IO_DATA10=39
CONFIG_ESP_PANEL_LCD_RGB_IO_DATA11=40
CONFIG_ESP_PANEL_LCD_RGB_IO_DATA12=41
CONFIG_ESP_PANEL_LCD_RGB_IO_DATA13=42
CONFIG_ESP_PANEL_LCD_RGB_IO_DATA14=2
CONFIG_ESP_PANEL_LCD_RGB_IO_DATA15=1
# end of Pins
# end of RGB interface
# end of RGB bus settings
# end of Bus settings

#
# Color settings
#
# CONFIG_ESP_PANEL_LCD_COLOR_BITS_8 is not set
CONFIG_ESP_PANEL_LCD_COLOR_BITS_16=y
CONFIG_ESP_PANEL_LCD_COLOR_BITS=16
CONFIG_ESP_PANEL_LCD_COLOR_ORDER_RGB=y
# CONFIG_ESP_PANEL_LCD_COLOR_ORDER_BGR is not set
# CONFIG_ESP_PANEL_LCD_INEVRT_COLOR is not set
# end of Color settings

#
# Transformation settings
#
# CONFIG_ESP_PANEL_LCD_SWAP_XY is not set
# CONFIG_ESP_PANEL_LCD_MIRROR_X is not set
# CONFIG_ESP_PANEL_LCD_MIRROR_Y is not set
# end of Transformation settings

CONFIG_ESP_PANEL_LCD_IO_RST=-1
# end of LCD settings

# CONFIG_ESP_PANEL_USE_TOUCH is not set
# CONFIG_ESP_PANEL_USE_BACKLIGHT is not set
# CONFIG_ESP_PANEL_USE_EXPANDER is not set
# end of Board
# end of ESP Display Panel Configurations
```

Cathgao, 2024-12-04:
是需要把

Lzw655, 2024-12-04:
Hi @Cathgao, 非常抱歉，这是库的一个 bug，我刚刚上传了一个新的分支 (https://github.com/esp-arduino-libs/ESP32_Display_Panel/tree/bugfix/fix_rgb_build_error_on_idf) 请用它尝试一下。

另外，请注意，由于 RGB LCD 的初始化参数通常不尽相同，请参考如下代码使用与实际 LCD 配套的初始化参数：

```c
static const esp_lcd_panel_vendor_init_cmd_t vendor_specific_init_default[] = {
//  {cmd, { data }, data_size, delay_ms}
    {0xFF, (uint8_t []){0x77, 0x01, 0x00, 0x00, 0x13}, 5, 0},
    {0xEF, (uint8_t []){0x08}, 1, 0},
    {0xFF, (uint8_t []){0x77, 0x01, 0x00, 0x00, 0x10}, 5, 0},
    {0xC0, (uint8_t []){0x3B, 0x00}, 2, 0},
};

    panel->init();
    ...
    ESP_PanelLcd *lcd = panel->getLcd();
    ...
    lcd->configVendorCommands(vendor_specific_init_default, sizeof(vendor_specific_init_default) / sizeof(vendor_specific_init_default[0]));
    ...
    panel->begin();
```

Cathgao, 2024-12-04:
您好，感谢解答。我尚未使用这个新的分支，但我尝试将如下图选项取消勾选后，编译通过了：
[image]
我的理解是：
如果不忽略`ESP_Panel_Conf.h`和`ESP_Panel_Board_*.h`文件的话，IDF里的配置不会生效，而是依然使用.h文件里的配置？我仅修改了IDF的配置，没有修改这些.h文件。
如有错误，欢迎指出

Lzw655, 2024-12-04:
再次抱歉，这两个配置的注解弄反了，默认使能表示的是忽略相关的配置文件从而使用 IDF 的配置。

Cathgao, 2024-12-04:
> 再次抱歉，这两个配置的注解弄反了，默认使能表示的是忽略相关的配置文件从而使用 IDF 的配置。

那就是注释是对的，我一开始没有去掉这两个勾，只修改了IDF的配置，所以头文件没有被启用，编译器报 `not declared`应该是预期的行为，这属于是我还没理解配置注释的问题
此外，我刚刚使用新分支代码覆盖后，勾上这两个选项也不会报错了，这是预期的行为吗？

Lzw655, 2024-12-04:
> > 再次抱歉，这两个配置的注解弄反了，默认使能表示的是忽略相关的配置文件从而使用 IDF 的配置。
> 
> 那就是注释是对的，我一开始没有去掉这两个勾，只修改了IDF的配置，所以头文件没有被启用，编译器报 `not declared`应该是预期的行为，这属于是我还没理解配置注释的问题 此外，我刚刚使用新分支代码覆盖后，勾上这两个选项也不会报错了，这是预期的行为吗？

1. 注释确实是错的，如果不勾选这两个选项，驱动会默认使用库中相应的配置文件而忽略 IDF 配置，由于库中配置文件默认设置 `#define ESP_PANEL_USE_CUSTOM_BOARD       (0)`，这种情况下是无法使用 `ESP_Panel` 类的，同时库中也会忽略宏 `ESP_PANEL_LCD_3WIRE_SPI_SCL_ACTIVE_EDGE`，因此不会导致编译报错，但是芯片运行起来会报错。
2. 这样的话，请优先使用新分支，并且不要取消勾选上面的选项

Cathgao, 2024-12-04:
可能我说的比较绕，我这样总结一下过程吧

1.一开始：
`Unckeck this to ignore ESP_Panel_Conf.h`和`Unckeck this to ignore ESP_Panel_Board_*.h`选中
`ESP_PANEL_USE_CUSTOM_BOARD`和`ESP_PANEL_USE_SUPPORTED_BOARD`均是`0`
编译报错

2.取消了两个选项
`Unckeck this to ignore ESP_Panel_Conf.h`和`Unckeck this to ignore ESP_Panel_Board_*.h`不勾选
`ESP_PANEL_USE_CUSTOM_BOARD`和`ESP_PANEL_USE_SUPPORTED_BOARD`均是`0`
编译不报错

3.重新勾选两个选项，并使用新分支
`Unckeck this to ignore ESP_Panel_Conf.h`和`Unckeck this to ignore ESP_Panel_Board_*.h`选中
`ESP_PANEL_USE_CUSTOM_BOARD`和`ESP_PANEL_USE_SUPPORTED_BOARD`均是`0`
编译不报错

感谢您的解答和工作。我刚开始学习ESP32的开发，正在熟悉环境，之后我会尝试您提供的初始化参数，再次感谢

closed 2024-12-23 by Lzw655