esp-arduino-libs/ESP32_Display_Panel #134: Library compiles with errors if ST7796 is selected

joelccrouch, 2024-12-04:
In ESP_Panel_Board_Custom.h, I set the following:

#define ESP_PANEL_LCD_NAME          ST7796

I get errors when I try to compile that it can't find ST7796.  The fix for version 0.1.8 is as follows:

Add #include "lcd/ST7796.h" to ESP_Panel_Library.h.

The same fix didn't work for version 0.2.1. I get a new error about SPI and QSPI.

joelccrouch, 2024-12-04:
The new error I get saids:

"E (3052) ESP_PanelLcd: attachRefreshFinishCallback(438): SPI and QSPI interfaces don't support this callback"

Lzw655, 2024-12-05:
Hi @joelccrouch, thank you for your feedback! In the new version, the SPI and QSPI LCD should use `attachDrawBitmapFinishCallback()` instead.

joelccrouch, 2024-12-05:
Ok, I solved problem by replacing all of the files in my old squareline studio example with the latest files from version 0.2.1.

joelccrouch, 2024-12-05:
Is there any chance that you could add touchscreen calibration functions to the library?  I find that the XPT2046 resistive touchscreen controller needs to be calibrated. It can't just be hard-coded.

Lzw655, 2024-12-06:
Sorry, I am not familiar with the calibration process for the XPT2046. Do you have any programs or examples that I could refer to?

joelccrouch, 2024-12-06:
This example uses the TFT_eSPI library, but since it doesn't support RGB displays I can't use it:

https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Generic/Touch_calibrate/Touch_calibrate.ino

Lzw655, 2024-12-09:
Thank you for sharing. I will use it as a reference to implement a similar screen calibration example.

joelccrouch, 2024-12-09:
Thank you!  The XPT2046 is a resistive touchscreen controller with a SPI interface.  There is a TSC2007 that is also a resistive touchscreen controller but with an I2C interface.  Can you please include that as an alternate resistive controller?

Lzw655, 2024-12-10:
Sure. But I don't have TSC2007 hardware for testing. Could you also share a TSC2007 driver that I could refer to?

joelccrouch, 2024-12-10:
Here's a driver example for tsc2007:
https://github.com/adafruit/Adafruit_TSC2007

Lzw655, 2024-12-10:
Great! I will try to add it as an alternate resistive controller.