esp-arduino-libs/ESP32_Display_Panel #138: E (104) GT911: Error (0xffffffff)! Touch controller GT911 initialization failed!

mustafaakh10, 2024-12-13:
Hi , 

I am using the waveshare 4.3 esp display : https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-4.3 . i am using the porting code for lvgl that is given on the website. My UI ran well for the first time without any issues and then i started to get this error.

D (35) ESP_PanelHost: Destroyed
D (38) ESP_Panel: Panel begin start
D (41) ESP_Panel: Begin LCD
D (75) ST7262_CPP: LCD panel @0x3fcaf574 created
D (76) ESP_PanelLcd: Begin start
D (76) ESP_PanelLcd: Begin end
D (76) ESP_Panel: Begin touch
D (77) ESP_PanelBus_I2C: Panel IO @0x3fcece48 created
D (82) ESP_PanelTouch: Disable interruption
E (86) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed
E (93) GT911: touch_gt911_read_cfg(362): GT911 read error!
E (98) GT911: esp_lcd_touch_new_i2c_gt911(128): GT911 init failed
E (104) GT911: Error (0xffffffff)! Touch controller GT911 initialization failed!
E (111) GT911_CPP: [ESP_FAIL] begin(43): New driver failed
E (116) ESP_Panel: begin(540): Begin touch failed
Initialize LVGL
D (132) ESP_PanelTouch: Get invalid handle
Create UI

this is the log information more in depth. what is the problem that i am facing

Lzw655, 2024-12-14:
Hi @mustafaakh10, 

I'm very sorry that you encountered this issue. It seems the `ESP32_S3_Touch_LCD_4_3` configuration file in this library did not correctly initialize the GT911. I have made some modifications in branch [bugfix/fix_board_esp32_s3_touch_lcd_4_3](https://github.com/esp-arduino-libs/ESP32_Display_Panel/tree/bugfix/fix_board_esp32_s3_touch_lcd_4_3), but I do not have the hardware to test it. Could you please help test it for me?

mustafaakh10, 2024-12-14:
Hi , thank you for your support . could you please elaborate a bit more. currently what i did now was copy the code from the ESP32_S3_Touch_LCD_4_3.h to the ESP_Panel_custom_board file that came with the porting lvgl code.

Compilation error: 'ESP_Panel' was not declared in this scope; did you mean 'ESP_PanelLcd'?

this is the error that i got with it. if i am not doing it correctly please let me know. thank you

Lzw655, 2024-12-16:
The waveshare 4.3 esp display is an internally supported board, you can modify the [ESP_Panel_Board_Supported.h](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/ESP_Panel_Board_Supported.h) to select it instead of the `ESP_Panel_Custom_Board` file. For more details, please check the docs [Using Supported Development Boards](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/master/docs/How_To_Use.md#using-supported-development-boards).

The error you encountered is most likely due to the missing [ESP_PANEL_USE_CUSTOM_BOARD ](https://github.com/esp-arduino-libs/ESP32_Display_Panel/blob/f9ed1848155a07e7dd5e1296af06546c1ed1ffac/ESP_Panel_Board_Custom.h#L12)macro definition.

Y1hsiaochunnn, 2024-12-23:
Hi @Lzw655 ,
One thing I have to mention is that we made extensive use of IOExpander when designing the LCD development board for the ESP32-S3, such as the ESP32-S3-Touch-LCD-4.3B, which uses the CH422G and the Touch's reset pin connects to the EXIO1.

The above is not the point, the point is the GT911 chip problem, we use a lot of GT911 touch chips found that some GT911 chips if not reset before touch initialization, then it will report an error. This is our experience with a lot of touch screens. So we usually reset GT911 in the header of the code.

As mentioned above, the best solution is to try to initialize the LCD screen and touch before the program initialization. Heavily tested

mustafaakh10, 2024-12-23:
Hi @Lzw655  sorry for the late reply. I was facing some challenges through the libraries, so I contacted the Wavehare development team, and they gave me the following demo file for the screen. In their code they used the IO expander. after using this setup i found no difficulty in driving the display. I am sharing the link for the demo with you. Thank you for your assistance.

https://files.waveshare.com/wiki/ESP32-S3-Touch-LCD-4.3/demo/ESP32-S3-Touch-LCD-4.3_Code.zip

Lzw655, 2024-12-24:
> Hi @Lzw655 sorry for the late reply. I was facing some challenges through the libraries, so I contacted the Wavehare development team, and they gave me the following demo file for the screen. In [...]

No worries. Thank you for your feedback!

closed 2025-02-18 by Lzw655