esp-arduino-libs/ESP32_Display_Panel #140: ESP32_Display_Panel error: 'ESP_Panel' does not name a type; did you mean 'ESP_PanelLcd'?  ESP_Panel *panel = nullptr;

ArmandsSilitis, 2024-12-20:
Hello,

I can't fix the error error: 'ESP_Panel' does not name a type; did you mean 'ESP_PanelLcd'?
ESP_Panel *panel = nullptr;

I've done everything as instructed in the README and in the library. I still get this error. I tried the suggestions of others who had such an error, but it didn't help either. I admit that there is some nuance that prevents access to the previously defined "ESP_Panel"

I use Arduino IDE, tried all versions of both ESP_Panel_Library, arduini, and LVGL. I used the code examples "PanelTest"

The display itself is ST7262 with touch GT911. I've edited ESP_PAnel_Board_Custom.h, the code with the RGB example works and the screen works, but I want to add my UI from EEZ studio.

I hope there are suggestions where the problem could be.

Thanks in advance!

I added files Arduino_code and Set up for ESP_Panel_Board_Custom.

[Arduino_code.txt](https://github.com/user-attachments/files/18210808/Arduino_code.txt)
[ESP_Panel_Board_Custom.txt](https://github.com/user-attachments/files/18210823/ESP_Panel_Board_Custom.txt)

Lzw655, 2024-12-23:
Hi @ArmandsSilitis, 

I'm sorry that you encountered this issue. Could you please tell me which directory the attached *ESP_Panel_Board_Custom.h* file is located in? The library searches for this file in the following order: `Current Project Directory` > `Arduino Library Directory` > `ESP32_Display_Panel Directory`. Please check if there are other *ESP_Panel_Board_Custom.h* files in these locations in order.

ArmandsSilitis, 2025-01-03:
Hello,

Thanks for your involvement.

I managed to sort out the file order, it turns out that ESP_Panel_Board_Custom had to be placed in the src folder.

Although the screen didn't even work with this library, in the end I managed to run this screen with Arduino_GFX_Library, but I had to write a new driver library.

Thanks.