esp-arduino-libs/ESP32_Display_Panel #141: Problem with Initialization of ESP32_Display_Panel with ESP32-S3-WROOM-1 N8R8 and  Waveshare 1.69inch Touch Round LCD Display

hamidnpc, 2024-12-21:
Hello all,
I am using ESP32-S3-WROOM-1 N8R8 with  Waveshare 1.69inch Touch Round LCD Display (based on ST7789V2), and using this [guide](https://www.waveshare.com/wiki/1.69inch_Touch_LCD_Module#ESP32S3_User_Guide), I am having problem with starting display module.

Whenever I am connecting I2C (TP_SDA ,TP_SCL) to board, I will get this error.  I tried to change the TP_SDA ,TP_SCL to some other pins, but still I will get the same when I am just using this. I tested this on two esp32, the problem is still there.

```
E (528) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(128): i2c transaction failed
E (529) CST816S: read_id(190): I2C read failed
E (529) CST816S: esp_lcd_touch_new_i2c_cst816s(100): Read version failed
E (535) CST816S: Initialization failed!
E (538) CST816S: [CST816S.cpp:41] begin(): Check error -1 (ESP_FAIL)
```

Thanks!

Lzw655, 2024-12-23:
Hi @hamidnpc, 

Are you using the latest version of the library or the offline version provided by Waveshare? I checked the latter, and it appears to be a very old version of the library, which I cannot support. Once release v1.0 is available, I will contact Waveshare's official developers to update the library in their Arduino examples.

hamidnpc, 2024-12-23:
No. You are right. I am using these versions and these [libraries](https://files.waveshare.com/wiki/1.69inch-Touch-LCD-Module/1.69inch_Touch_LCD_ESP32S3.zip). 
ESP32_Display_Panel v0.0.2
ESP32_IO_Expander	>= v0.0.1
arduino-esp32 >= v2.0.9，< 3.0.0

Is there anything I can do meanwhile? 
Here is the ESP_Panel_Conf.h file:

```
/*
 * SPDX-FileCopyrightText: 2023 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */
#ifndef ESP_PANEL_CONF_H
#define ESP_PANEL_CONF_H

// *INDENT-OFF*
/* Set to 0 if use a custom board */
#define ESP_PANEL_USE_SUPPORTED_BOARD   (0)

#define ESP_PANEL_USE_ST7262_BOARD   (0)
#define ESP_PANEL_USE_GC9A01_BOARD   (0)
#define ESP_PANEL_USE_ST7789_BOARD   (1)

#if ESP_PANEL_USE_SUPPORTED_BOARD
/*
 * Supported Boards.
 *  - ESP32-C3-LCDkit: https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32c3/esp32-c3-lcdkit/index.html
 *  - ESP32-S3-Box: https://github.com/espressif/esp-box/tree/master
 *  - ESP32-S3-Box-3: https://github.com/espressif/esp-box/tree/master
 *  - ESP32-S3-Box-3(beta): https://github.com/espressif/esp-box/tree/c4c954888e11250423f083df0067d99e22d59fbe
 *  - ESP32-S3-Box-Lite: https://github.com/espressif/esp-box/tree/master
 *  - ESP32-S3-EYE: https://github.com/espressif/esp-who/blob/master/docs/en/get-started/ESP32-S3-EYE_Getting_Started_Guide.md
 *  - ESP32-S3-Korvo-2: https://docs.espressif.com/projects/esp-adf/en/latest/design-guide/dev-boards/user-guide-esp32-s3-korvo-2.html
 *  - ESP32-S3-LCD-EV-Board: https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-lcd-ev-board/index.html
 *  - ESP32-S3-LCD-EV-Board-2: https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-lcd-ev-board/index.html
 *  - ESP32-S3-USB-OTG: https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-usb-otg/index.html
 *
 * Uncomment one of the following macros to select an supported development board.
 */
// #define ESP_PANEL_BOARD_ESP32_C3_LCDKIT
// #define ESP_PANEL_BOARD_ESP32_S3_BOX
// #define ESP_PANEL_BOARD_ESP32_S3_BOX_3
// #define ESP_PANEL_BOARD_ESP32_S3_BOX_3_BETA
// #define ESP_PANEL_BOARD_ESP32_S3_BOX_LITE
// #define ESP_PANEL_BOARD_ESP32_S3_EYE
// #define ESP_PANEL_BOARD_ESP32_S3_KORVO_2
// #define ESP_PANEL_BOARD_ESP32_S3_LCD_EV_BOARD
// #define ESP_PANEL_BOARD_ESP32_S3_LCD_EV_BOARD_2
// #define ESP_PANEL_BOARD_ESP32_S3_USB_OTG

#else
/*-------------------------------- LCD Related --------------------------------*/
/* Set to 0 if not using LCD */
#define ESP_PANEL_USE_LCD           (1)
#if ESP_PANEL_USE_LCD
/**
 * LCD IC name. Choose one of the following:
 *      - ILI9341
 *      - GC9503, GC9A01
 *      - ST7262, ST7789, ST7796
 */
#if ESP_PANEL_USE_ST7262_BOARD

#define ESP_PANEL_LCD_NAME          ST7262

#elif ESP_PANEL_USE_GC9A01_BOARD

#define ESP_PANEL_LCD_NAME          GC9A01
[... 272 lines cut ...]

/* Set to 1 if use PWM for backlight brightness control. */
#define ESP_PANEL_LCD_BL_USE_PWM            (1)
#if ESP_PANEL_LCD_BL_USE_PWM
/**
 *  Backlight LEDC Parameters.
 *
 *  Please refer to https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/ledc.html for details.
 */
    #define ESP_PANEL_LCD_BL_PWM_TIMER          (0)
    #define ESP_PANEL_LCD_BL_PWM_CHANNEL        (0)
    #define ESP_PANEL_LCD_BL_PWM_RESOLUTION     (10)
    #define ESP_PANEL_LCD_BL_PWM_FREQ_HZ        (5000)
#endif /* ESP_PANEL_LCD_BL_USE_PWM */
#endif /* ESP_PANEL_USE_BL */

/*-------------------------------- Others --------------------------------*/
/* Assert on error. Otherwise return error code */
#define ESP_PANEL_CHECK_RESULT_ASSERT       (0)

#endif /* ESP_PANEL_USE_SUPPORTED_BOARD */
// *INDENT-OFF*

#endif
```

Y1hsiaochunnn, 2024-12-23:
Hi @hamidnpc ,
 Well, I don't know if you have to use this library, in fact the 1.69 screen you describe should be a [1.69inch Touch LCD Module](https://www.waveshare.com/1.69inch-touch-lcd-module.htm). So you should refer to the product's wiki, https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-1.69, maybe you just want to light it.

After pre-released v1.0.0 of the library, I will do compatible. Includes the BSP of ESP32-S3-Touch-LCD-1.69

Lzw655, 2024-12-24:
Sorry, I can't see any problem with this file. There may be bugs in the older version of the library.