esp-arduino-libs/ESP32_Display_Panel #144: UnknownPackageError: Could not find the package with 'espressif/toolchain-riscv32-esp @ 13.2.0+20240530' requirements for your system 'darwin_arm64'

Kanzll, 2024-12-24:
Hello,

Unable to compile the platformIO example, with all versions (@5.1 and even the latest).
here is the complete OUTPUT:
```
Resolving ESP-LCD dependencies...
Tool Manager: Installing espressif/toolchain-riscv32-esp @ 13.2.0+20240530
UnknownPackageError: Could not find the package with 'espressif/toolchain-riscv32-esp @ 13.2.0+20240530' requirements for your system 'darwin_arm64'
```
 I am on vscode with PlatformIO on os x (15.1.1)

Thank you

Lzw655, 2024-12-25:
Hi @Kanzll, 

I tried compiling this project on Linux, and apart from dependency version errors causing the build to fail, I did not encounter the `UnknownPackageError`. I will attempt to compile it on MacOS next.

Kanzll, 2024-12-25:
Hello @Lzw655,
It's been a few days, that I am not able to compile a single project plaformIO for the Waveshare 5inch lcd board (1024x600)

Lzw655, 2024-12-25:
I was able to reproduce this issue. May I confirm if you are intending to use Arduino as the framework? Also, could you specify the target ESP chip model?

Kanzll, 2024-12-25:
Yes, I use Arduino framework, the chip model is esp32s3.  (Board: Waveshare ESP32-S3-Touch-LCD-5B )

Lzw655, 2024-12-26:
Hi @Kanzll,

Please try to modify the *platformio.ini* file as follows, then the project can be compiled success in my test. 
(BTW, note that the *lv_conf.h* should be enabled and placed to the *src* folder)
```
[env:ESP-LCD]
platform = espressif32
board = ESP-LCD
framework = arduino
platform_packages =
        platformio/framework-arduinoespressif32@https://github.com/esp-arduino-libs/arduino-esp32.git#high_perf/v3.0.3
        platformio/framework-arduinoespressif32-libs@https://github.com/esp-arduino-libs/arduino-esp32-sdk.git#high_perf/v3.0.3
upload_speed = 921600
monitor_speed = 115200
build_flags =
        -DBOARD_HAS_PSRAM
        -DLV_CONF_INCLUDE_SIMPLE
        -DDISABLE_ALL_LIBRARY_WARNINGS
        -DARDUINO_USB_CDC_ON_BOOT=1
        -DCORE_DEBUG_LEVEL=1
        -DLV_LVGL_H_INCLUDE_SIMPLE
        -I src
lib_deps =
        https://github.com/esp-arduino-libs/ESP32_Display_Panel.git
        https://github.com/esp-arduino-libs/ESP32_IO_Expander.git#v0.1.0
    https://github.com/lvgl/lvgl.git#release/v8.3
```

Kanzll, 2024-12-26:
Hello @Lzw655,
Thank you for your update, but I have another compilation error :
```
Resolving esp32s3box dependencies...
Error! Failed to extract upstream toolchainconfigurations:
Extra data: line 1 column 4 (char 3)
You can disable this feature via the `board_build.arduino.upstream_packages = no` setting in your `platformio.ini` file.
```
Other error with  `board_build.arduino.upstream_packages = no` :
```

Resolving esp32s3box dependencies...
Tool Manager: Installing git+https://github.com/esp-arduino-libs/arduino-esp32.git#high_perf/v3.0.3
git version 2.18.0
Cloning into '/Users/kiman/.platformio/.cache/tmp/pkg-installing-r58oopd4'...

fatal: could not read Username for 'https://github.com': Device not configured

VCSBaseException: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', '--branch', 'high_perf/v3.0.3', 'https://github.com/esp-arduino-libs/arduino-esp32.git', '/Users/kiman/.platformio/.cache/tmp/pkg-installing-r58oopd4']
```
Thank you !

Lzw655, 2024-12-27:
Sorry, I can't reproduce this issue now; it seems related to Git and network conditions.

Kanzll, 2025-01-01:
Hello @Lzw655,
I got 404 error when try 1st url in browser :
[image]
is it an authentication problem ? Should I add an access token or something like that?
Thank you for your help !

Lzw655, 2025-01-02:
This is not a standard URL. The 'high_perf/v3.0.3' is a branch's name and the section is a format for PlatformIO.

Kanzll, 2025-01-03:
With this URL I got always  `fatal: could not read Username for 'https://github.com': Device not configured`. 
I replaced by this one `platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.3`
and project compiles successfully. 
there is difference with this twos URL ?

Lzw655, 2025-01-06:
Oh, I'm really sorry for that. The repository `esp-arduino-libs/arduino-esp32.git` seems to have been accidentally deleted. However, it is identical to `espressif/arduino-esp32.git`. You can directly use the latter as a replacement.

Kanzll, 2025-01-07:
Thank you for your help @Lzw655 !

Lzw655, 2025-01-08:
You are welcome! Can your board work well?

Kanzll, 2025-01-09:
Hello @Lzw655,
Sorry for the late reply, yes it worked for firsts moment,  now it compile and upload, but I had a kernel panic, I don't know if it's related with this configuration change, here is the kernel panic message :
```
ELF file SHA256: 24b24cc75c147297

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037c22e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29e4
entry 0x403c9880
LVGL porting example start
Initialize panel device
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400556c0  PS      : 0x00060030  A0      : 0x820c2a7e  A1      : 0x3fceb800  
A2      : 0x0000031e  A3      : 0x0000031c  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x820ce644  A9      : 0x3fceb7c0  
A10     : 0x00000000  A11     : 0x3c0f0138  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3c0f5ef4  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000031c  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffc  

Backtrace: 0x400556bd:0x3fceb800 0x420c2a7b:0x3fceb810 0x420ca48a:0x3fcebb30 0x420ca4c6:0x3fcebbc0 0x420139d2:0x3fcebc00 0x4205c2db:0x3fcebc40 0x420033ce:0x3fcebcc0 0x42003428:0x3fcebcf0 0x420038b2:0x3fcebd10 0x420144c4:0x3fcebd60 0x4038256e:0x3fcebd80
```

Thank you for your help !

Kanzll, 2025-01-09:
Here are logs :
`Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037b52a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x508
load:0x403c9700,len:0x4
load:0x403c9704,len:0xad0
load:0x403cc700,len:0x29e4
entry 0x403c9880
[   391][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x4202e0d4
[   402][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x4202e0a0
[   414][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x4202e06c
[   425][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x4202e038
[   436][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x4202e0d4
[   448][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x4202e0a0
[   459][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x4202e06c
[   470][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x4202e038
[   482][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x4202e0d4
[   493][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x4202e0a0
[   504][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x4202e06c
[   516][V][esp32-hal-periman.c:235] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x4202e038
[   533][I][esp32-hal-psram.c:90] psramInit(): PSRAM enabled
[   542][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 44 successfully set to type UART_RX (2) with bus 0x3fc96aac
[   552][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 43 successfully set to type UART_TX (3) with bus 0x3fc96aac
=========== Before Setup Start ===========
Chip Info:
------------------------------------------
  Model             : ESP32-S3
  Package           : 0
  Revision          : 2
  Cores             : 2
  CPU Frequency     : 240 MHz
  XTAL Frequency    : 40 MHz
  Embedded Flash    : No
  Embedded PSRAM    : No
  2.4GHz WiFi       : Yes
  Classic BT        : No
  BT Low Energy     : Yes
  IEEE 802.15.4     : No
------------------------------------------
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   303444 B ( 296.3 KB)
  Free Bytes        :   276684 B ( 270.2 KB)
  Allocated Bytes   :    22800 B (  22.3 KB)
  Minimum Free Bytes:   271376 B ( 265.0 KB)
  Largest Free Block:   237556 B ( 232.0 KB)
------------------------------------------
SPIRAM Memory Info:
------------------------------------------
  Total Size        :  6815744 B (6656.0 KB)
  Free Bytes        :  6812808 B (6653.1 KB)
  Allocated Bytes   :      576 B (   0.6 KB)
  Minimum Free Bytes:  6812808 B (6653.1 KB)
  Largest Free Block:  6684660 B (6528.0 KB)
  Bus Mode          : OPI
------------------------------------------
Flash Info:
------------------------------------------
  Chip Size         : 16777216 B (16 MB)
  Block Size        :    65536 B (  64.0 KB)
  Sector Size       :     4096 B (   4.0 KB)
  Page Size         :      256 B (   0.2 KB)
  Bus Speed         : 80 MHz
  Bus Mode          : QIO
------------------------------------------
[... 46 lines cut ...]
D (461) ESP_PanelLcd: Begin start
D (462) ESP_PanelLcd: Begin end
W (462) ESP_PanelLcd: Display on/off function is disabled
D (465) ESP_Panel: Begin touch
D (468) ESP_PanelBus_I2C: Panel IO @0x3fcecf70 created
D (473) ESP_PanelTouch: Disable interruption
D (477) GT911_CPP: Use default GT911 driver data(address: 0x5d)
I (483) GT911: LCD touch panel create success, version: 1.1.1
W (488) GT911: Unable to initialize the I2C address
I (493) GT911: TouchPad_ID:0x39,0x31,0x31
I (496) GT911: TouchPad_Config_Version:82
D (500) ESP_Panel: Panel begin end
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400556c0  PS      : 0x00060030  A0      : 0x820debde  A1      : 0x3fceb7f0  
A2      : 0x0000031e  A3      : 0x0000031c  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x820ea7d4  A9      : 0x3fceb7b0  
A10     : 0x00000000  A11     : 0x3c1101e0  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3c11ce58  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000031c  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffc  

Backtrace: 0x400556bd:0x3fceb7f0 0x420debdb:0x3fceb800 0x420e661a:0x3fcebb20 0x420e6656:0x3fcebbb0 0x4202db5e:0x3fcebbf0 0x42076ebf:0x3fcebc30 0x42002ca2:0x3fcebcb0 0x42002cfc:0x3fcebce0 0x420036e2:0x3fcebd00 0x4202ef5d:0x3fcebd60 0x40380fd2:0x3fcebd80

ELF file SHA256: 1ceaf6ad2c2584ce`

Lzw655, 2025-01-09:
The panel initialization is working well. There might be some wrong operations later. Please share your `app.cpp` file. Besides, you can also try to comment all LVGL-related codes and run the following codes:

```c
        ...
        panel->begin();

        Serial.println("Draw color bar from top to bottom, the order is B - G - R");
        panel->getLcd()->colorBarTest(panel->getLcdWidth(), panel->getLcdHeight());
```

Kanzll, 2025-01-09:
Effectively, kernel panic doesn't occur when comment lvgl-related codes in setup function.
here is my main.app :
```
/*
 * SPDX-FileCopyrightText: 2024 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 */

#include <Arduino.h>
#include <ESP_Panel_Library.h>
#include <lvgl.h>
#include "lvgl_port_v8.h"
#include <ui.h>
#include <EEPROM.h>
#include "WiFi.h"
#include <vector>
#include "time.h"

//EEPROM
#define EEPROM_SIZE 128
#define EEPROM_ADDR_WIFI_FLAG 0
#define EEPROM_ADDR_WIFI_CREDENTIAL 4

const char *ntpServer = "pool.ntp.org";
const long gmtOffset_sec = -5 * 60 * 60;  // Set your timezone here
const int daylightOffset_sec = 0;

//my custom for wifi
typedef enum {
  NONE,
  NETWORK_SEARCHING,
  NETWORK_CONNECTED_MANUAL, //connected after user entered password
  NETWORK_CONNECTED_AUTO, //connected after config loaded from EEPROM
  NETWORK_CONNECTED,
  NETWORK_CONNECT_FAILED
} Network_Status_t;

Network_Status_t networkStatus = NONE;

//timer refresh network list
static lv_timer_t *timer;

static int foundNetworks = 0;
unsigned long networkTimeout = 10 * 1000;
String ssidName, ssidPW;

TaskHandle_t ntScanTaskHandler, ntConnectTaskHandler;
std::vector<String> foundWifiList;

//déclaration des fonctions pour l'ordre 
static void beginWIFITask(void *pvParameters);
static void scanWIFITask(void *pvParameters);

static void popupMsgBox(String title, String msg) {

  lv_label_set_text(ui_WifiConnectinfoLbl1,title.c_str());
  lv_label_set_text(ui_WifiConnectinfoLbl2,msg.c_str());

  //affiche la fenetre
  lv_obj_clear_flag(ui_WifiConnectInfoPanel, LV_OBJ_FLAG_HIDDEN);
  lv_obj_move_foreground(ui_WifiConnectInfoPanel);
}

static void networkScanner() {
  xTaskCreate(scanWIFITask,
              "ScanWIFITask",
              4096,
              NULL,
              1,
[... 252 lines cut ...]
    Serial.println("Draw color bar from top to bottom, the order is B - G - R");
    panel->getLcd()->colorBarTest(panel->getLcdWidth(), panel->getLcdHeight());

    /* Lock the mutex due to the LVGL APIs are not thread-safe */
    lvgl_port_lock(-1);
    ui_init();
    /* Release the mutex */
    lvgl_port_unlock();
    //Load wifi setup
    tryPreviousNetwork(); 
    Serial.println(title + " end");
}

void loop()
{
  lv_timer_handler();
  delay(5);
  if (lv_scr_act()==ui_HomeScreen) {
    updateLocalTime();
    update_StatusBar();
    delay(1000);
  }
}

```

Lzw655, 2025-01-09:
I think you can try the following codes:

```c
void setup()
{
    String title = "LVGL porting example";

    Serial.begin(115200);
    Serial.println(title + " start");
    Serial.println("Initialize panel device");
    
    // Should initialize LVGL by calling `lvgl_port_init()`
    // /* Initialize LVGL core */
    // lv_init();

    ESP_Panel *panel = new ESP_Panel();
    panel->init();

#if LVGL_PORT_AVOID_TEAR
    // When avoid tearing function is enabled, configure the RGB bus according to the LVGL configuration
    ESP_PanelBus_RGB *rgb_bus = static_cast<ESP_PanelBus_RGB *>(panel->getLcd()->getBus());
    rgb_bus->configRgbFrameBufferNumber(LVGL_PORT_DISP_BUFFER_NUM);
    rgb_bus->configRgbBounceBufferSize(LVGL_PORT_RGB_BOUNCE_BUFFER_SIZE);
#endif
    panel->begin();

    /* 
    Serial.println("Draw color bar from top to bottom, the order is B - G - R");
    panel->getLcd()->colorBarTest(panel->getLcdWidth(), panel->getLcdHeight());
    */

    Serial.println("Initialize LVGL");
    lvgl_port_init(panel->getLcd(), panel->getTouch());

    /* Lock the mutex due to the LVGL APIs are not thread-safe */
    lvgl_port_lock(-1);
    ui_init();
    /* Release the mutex */
    lvgl_port_unlock();
    //Load wifi setup
    tryPreviousNetwork(); 
    Serial.println(title + " end");
}
```

Kanzll, 2025-01-09:
Yeah it's works ! Thanks a lot @Lzw655 !
Just to know, I never used lvgl_port_init function in the beginning and it worked, no kernel panic ?

Lzw655, 2025-01-10:
Thank you for pointing out another issue! I found a bug in the `ESP_PANEL_OTHER_CHECK_LOGE()` macro used by the `lvgl_port_lock()` function. It will trigger kernel panic when checking the error. I will fix it now.

closed 2025-01-10 by Lzw655