esp-arduino-libs/ESP32_Display_Panel #145: pio example not working?

B4stl3r, 2025-01-04:
Hi,

I'm using a Waveshare ESP32-S3-Touch-LCD-5 (B) with 1024x600 px.

Still struggling to get anything drawn on the display with the provided pio example.

I checked the boards/ESP-LCD.json, I think it should be compatible with this Display ?

In the Board_Supported.h File I changed the "use supported" to 1 and uncommented:
#define BOARD_WAVESHARE_ESP32_S3_Touch_LCD_5_B

(I also tried the Custom config from the Waveshare Wiki Example, Link see below).

With the provided platformio.ini I just receive:
*** missing SConscript file 'C:\\Users\\User\\.platformio\\packages\\framework-arduinoespressif32\\tools\\platformio-build.py'

When I uncomment the platform_packages compilation is NOT possible, as the latest version of ESP32_IO_Expander has some major changes and therefore is not working.

So I tried:
lib_deps =
	https://github.com/esp-arduino-libs/ESP32_Display_Panel.git#v0.1.4 ;multiple ones here, 0.2.1 
	https://github.com/esp-arduino-libs/ESP32_IO_Expander.git#v0.0.4 ;multiple ones here, till 0.1.0..
        https://github.com/lvgl/lvgl.git#release/v8.3

With these versions, it can be compiled and flashed, but display stays black without anything drawn.

Also tried to get the lv demo widgets to get running, but no luck.
I don't have any lv_conf.h as in the example there's none.. assume as the buildflags define the simple, it's not needed?

Do you have any input / idea's? 
Here's the Waveshare Wiki of the display https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-5 
Thanks in advance!

Lzw655, 2025-01-06:
Hi @B4stl3r,

I'm sorry, I don't have this development board to test with. @Y1hsiaochunnn, do you have any thoughts or suggestions regarding this issue?

Y1hsiaochunnn, 2025-01-06:
@B4stl3r ,Can you share your project?

B4stl3r, 2025-01-07:
Hi @Y1hsiaochunnn , 

it's the example here:
https://github.com/esp-arduino-libs/ESP32_Display_Panel/tree/master/examples/PlatformIO

and the waveshare demos 8/9 from here:
https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-5#Resources
"ported" to Platformio.. (just thrown the files in a copy of the example above)..

Enough for you to troubleshoot?

Lzw655, 2025-01-09:
Hi @B4stl3r, 

I think I found the issue. Could you please test it with the [bugfix/fix_board_esp32_s3_touch_lcd_4_3](https://github.com/esp-arduino-libs/ESP32_Display_Panel/tree/bugfix/fix_board_esp32_s3_touch_lcd_4_3) branch?

B4stl3r, 2025-01-09:
Hi @Lzw655 ,
thx for your change and effort.
Just tried the new pio example.. the example does not compile for myself:

Linking .pio\build\ESP-LCD\firmware.elf
C:/Users/user/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:/Users/user/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/lib/esp32s3/no-rtti\libc.a(libc_a-getentropyr.o):(.literal+0x4): warning: _getentropy is not implemented and will always fail
C:/Users/user/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:/Users/user/.platformio/packages/framework-arduinoespressif32-libs/esp32s3/qio_opi\libesp_system.a(startup.c.obj):(.literal.__esp_system_init_fn_init_components0+0x8): undefined reference to `_Unwind_SetEnableExceptionFdeSorting'
C:/Users/user/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:/Users/user/.platformio/packages/framework-arduinoespressif32-libs/esp32s3/qio_opi\libesp_system.a(startup.c.obj): in function `__esp_system_init_fn_init_components0':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/startup.c:497:(.text.__esp_system_init_fn_init_components0+0x1e): undefined reference to `_Unwind_SetEnableExceptionFdeSorting'
C:/Users/user/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:/Users/user/.platformio/packages/framework-arduinoespressif32-libs/esp32s3/lib\libnewlib.a(reent_init.c.obj):(.iram1.0.literal+0x4): undefined reference to `_cleanup_r'
C:/Users/user/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:/Users/user/.platformio/packages/framework-arduinoespressif32-libs/esp32s3/lib\libnewlib.a(newlib_init.c.obj):(.data.s_stub_table+0x98): undefined reference to `_cleanup_r'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\ESP-LCD\firmware.elf] Error 1

For my old / customized example (based on your's, but with the 0.2.2 lib now) the display also stays black and nothing is drawn.

I checked your PR and have seen, that you changed also the touch init.
The touch itself was working proberly, at least a 1 point touch. a multiple point touch was not working properly (what I can remember). So not sure, if THIS change was required..

Regarding the display itself:
I tried also to use the displayparameters
#if ESP_PANEL_USE_1024_600_LCD
    #define ESP_PANEL_LCD_RGB_CLK_HZ            (21 * 1000 * 1000)
    #define ESP_PANEL_LCD_RGB_HPW               (88)
    #define ESP_PANEL_LCD_RGB_HBP               (188)
    #define ESP_PANEL_LCD_RGB_HFP               (44)
    #define ESP_PANEL_LCD_RGB_VPW               (6)
    #define ESP_PANEL_LCD_RGB_VBP               (16)
    #define ESP_PANEL_LCD_RGB_VFP               (3)
which are directly from waveshare example, but still just a black display.

So unfortunately still not working :(

Lzw655, 2025-01-10:
Sorry, I can't reproduce this issue on win10 with the newest version. Could you please delete your toolchains (C:/Users/user/.platformio/packages/toolchain-*) and reinstall them?

BTW, I think the reason for a black display is the backlight is not powered. Here are the key changes: https://github.com/esp-arduino-libs/ESP32_Display_Panel/commit/2eee5f8f72ebcb60b07a45807f36364b10fabd88#diff-e8c68dae75f27be2b686ed1598e767fbacf9cc17ffc727dced199f4569a248b7R269-R276

B4stl3r, 2025-01-10:
Hi, 
the backlight **is** turned on.
I've also thought on this topic and have a modified example, which is toggling the backlight every second.
It can also be controlled via PIN2.. the backlight control of this lib does also currently not support it, but that's a different topic.
First it would be great, if the display is showing anything.

the relvant changes for my display (5B) are:
https://github.com/esp-arduino-libs/ESP32_Display_Panel/commit/2eee5f8f72ebcb60b07a45807f36364b10fabd88#diff-541f4e8f74b8589af663293d0e78ac7fdc9a63241298d11a42fcccca374d0a3b

if've checked them already, yes.

As mentiond, nothing on the display, nor the label, or the widget demo when I change the example.

Lzw655, 2025-01-10:
Hi @H-sw123 @Y1hsiaochunnn, any ideas?

B4stl3r, 2025-01-11:
... in desperation I installed Arduino IDE and flashed the waveshare examples..
Everything worked..
Tried to port it again, did not work..
Switched to another machine, installed vs code and pio fresh, downloaded your example, flashed, working..

So my original machine pio setup is somehow broken...

Sorry for the confusion. Regarding the Backlight of Waveshare 5B I will create a seperate topic in the next days.

Thanks for your support!

closed 2025-01-11 by B4stl3r

Lzw655, 2025-01-13:
No worries. Feel free to ask.