esp-arduino-libs/ESP32_Display_Panel #147: Waveshare ESP32-S3-Touch-LCD-4 - Unable to use touch and I2C sensor at the same time

dinisfc, 2025-01-09:
Currently testing a prototype built from the Squareline Porting example. This board uses the GT911 Touch controller which uses the same channel as any other I2C external device. I'm using a SEN66 sensor, using its library which requires the use of the Wire library. When I do this, panel->init() will fail and show something like this:

> E (1506) i2c: i2c driver install error
> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
> 
> Core  1 register dump:
> PC      : 0x42058aab  PS      : 0x00060330  A0      : 0x82001d2e  A1      : 0x3fcaf8b0  
> A2      : 0x00000000  A3      : 0x3c060164  A4      : 0x00000010  A5      : 0x3fcaaf94  
> A6      : 0x00000024  A7      : 0x0000000a  A8      : 0x8202fd80  A9      : 0x3fcaf890  
> A10     : 0x00000002  A11     : 0x3c060164  A12     : 0x3fcaf6d8  A13     : 0x3fcaf798  
> A14     : 0x3fcaf7e8  A15     : 0x3fcab140  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
> EXCVADDR: 0x00000008  LBEG    : 0x40056f08  LEND    : 0x40056f12  LCOUNT  : 0x00000000  
> 
> 
> Backtrace: 0x42058aa8:0x3fcaf8b0 0x42001d2b:0x3fcaf8d0 0x42033681:0x3fcaf8f0 0x4037d062:0x3fcaf910

And then get stuck in a loop doing the same. 

I'm using version 0.1.8 per the Waveshare instructions. I'm using an ESP IOExpander library, v0.0.4, altered by Waveshare, so updating any of this libraries breaks the code. I'm also using lv_conf.h by Waveshare. Being a dev board the connections can't be altered.

Would like to know if there's a way to not have the I2C error and if, maybe, making the updated library (0.2.2 as of posting) work be worth it. 

Note: the LCD-4 and LCD-4.3 are not the same! For LCD-4, the custom board configurations are used.

Lzw655, 2025-01-09:
Hi @dinisfc,

1. For the conflict between this library and the `Wire` library on I2C, here's a solution: set `ESP_PANEL_TOUCH_BUS_SKIP_INIT_HOST` to `1` in `ESP_Panel_Board_Custom.h`. This way, the driver will not initialize the I2C host internally. You will then need to manually initialize the I2C host using `Wire` in your `.ino` file before initializing the panel.
2. Since the [Release v1.0.0](https://github.com/esp-arduino-libs/ESP32_Display_Panel/pull/137) is in progress, I think it's better to update your project after it's ready.
3. BTW, we can support the `ESP32-S3-Touch-LCD-4` board in the library if it's helpful for you.

dinisfc, 2025-01-09:
Thanks for the answer.
Setting ESP_PANEL_TOUCH_BUS_SKIP_INIT_HOST to 1 and having Wire.begin before anything else worked.

dinisfc, 2025-02-06:
I've since updated to 0.2.2 and I'm now having errors again, even with ESP_PANEL_TOUCH_BUS_SKIP_INIT_HOST at 1.
I'm getting:

> E (5056) i2c: i2c driver install error
> E (5057) ESP_PanelHost: [ESP_FAIL] begin(153): I2C[0] install driver failed
> E (5057) ESP_Panel: init(470): Initialize host failed
> E (5061) ESP_Panel: begin(489): Panel is not initialized
> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
> 
> Core  1 register dump:
> PC      : 0x420273cb  PS      : 0x00060130  A0      : 0x82027445  A1      : 0x3fcaf8e0  
> A2      : 0x00000000  A3      : 0x3fcaf908  A4      : 0x3fcaf90c  A5      : 0x3c071251  
> A6      : 0x0000001c  A7      : 0x00000003  A8      : 0x0000001c  A9      : 0x3fcaf8c0  
> A10     : 0x00000005  A11     : 0x3fcaf908  A12     : 0x3fcaf90c  A13     : 0x00000000  
> A14     : 0x3fcaf790  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
> EXCVADDR: 0x00000020  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffe  
> 
> 
> Backtrace: 0x420273c8:0x3fcaf8e0 0x42027442:0x3fcaf900 0x4202751d:0x3fcaf930 0x420257a5:0x3fcaf950 0x4201eaec:0x3fcaf970 0x420329ba:0x3fcaf990 0x42001cef:0x3fcaf9b0 0x4203846d:0x3fcaf9d0 0x4037cfc6:0x3fcaf9f0

Wire.begin and the sensor's reading start function are both before the screen. Running only the sensor or screen does not cause this error. While the system enters a loop showing this error repeatedly, the sensor remains on and running.

Lzw655, 2025-02-08:
Hi @dinisfc,

Could you share me the whole `ESP_Panel_Board_Custom.h` file?

dinisfc, 2025-02-10:
Changing `ESP_PANEL_EXPANDER_SKIP_INIT_HOST` to 0 fixed it. Thanks.

betamoojw, 2025-03-12:
> Hi [@dinisfc](https://github.com/dinisfc),
>
> 1. For the conflict between this library and the `Wire` library on I2C, here's a solution: set `ESP_PANEL_TOUCH_BUS_SKIP_INIT_HOST` to `1` in [...]

I was wondering when we could use the `ESP32-S3-Touch-LCD-4` board in the library. It will be helpful for the waveshare ESP32-S3-Touch-LCD-4 user without complex additional configuration.

Lzw655, 2025-03-12:
> I was wondering when we could use the `ESP32-S3-Touch-LCD-4` board in the library. It will be helpful for the waveshare ESP32-S3-Touch-LCD-4 user without complex additional configuration.

Hi @Y1hsiaochunnn, could you help us add support for `ESP32-S3-Touch-LCD-4` in the library?

Lzw655, 2025-05-26:
Hi @waszee,

I'm sorry you've encountered this problem. Currently, both `Wire` and this library use the old version of the I2C driver. It is likely that some of your other drivers (such as BME680) use the new version of the I2C driver, which leads to conflicts.

waszee, 2025-05-26:
Thank you for the comments.  I am going to monitor posts for solutions.  Instead of the i2c BME680 sensor I connected a simpler DHT_11 sensor which uses the GPIO6 pin on the Waveshare board and does not use I2C.  That is working and I can read temperature and humidity.  I am not very happy with the DHT_11 response and precision as it fails to show small changes due to HVAC that the BME680 does detect and I like to monitor.  I am now trying to configure the unit to add a MSQTT client post while maintaining the local sensor's visual lcd charting display.  First attempt resulted in display flashes followed by reboots. After not trying to run a MSQTT client loop and making the WIFI connection early in setup, I am getting a stable display that is charting data and the publish command is at least not crashing.  I do not have my "homeassistant" side configured to accept the data so still work in progress. The publish command is triggered from the add data timer event inside the lvgl loop. I tried the Uncle Rus BME680 drivers with it but like the Adafruit they work without LVGL but gave the i2c Conflict when I tried to link to the existing I2C used to setup the touch panel.  Could easily be a simple mistake in coding that I am making.

I have tried a couple of other approaches.  I used Platformio and the Espressif IDE to make some displays using LGVL version 8.3.11.  That worked but the compiler gives a flag that I need to update from the old i2c driver.  

In addition, both Platformio and Arduinio IDE constantly try to get me to update from LGVL version 8 to version 9.2.  I have tried and think the actual chart graphic commands will changed very little.  I have not been able to figure out the port to link the hardware display panels to LVGL.  The arduino simple port example uses a file lvgl_V8_port,h but I have not found enough examples to show how to create the lvgl_V9_port.h and lvg_V9_port.c.    
If any of you have that please post and let me know where to look for it.  Found some LVGL 9 examples using "cheap yellow" displays and Espressif has posted some examples for their boards but I got lost trying to correlate to the Waveshare-esp32-s3-touch-lcd-4.3 that I have.

Lzw655, 2025-05-27:
Hi @Y1hsiaochunnn 

I'm truly sorry that this matter has been postponed continuously. I've been extremely busy recently. I promise that as soon as I have free time, the first thing I'll do is to transplant the new version of the i2c driver.

Lzw655, 2025-05-27:
Hi @waszee,

LVGL V8 is indeed gradually being phased out. The library will support the V9 port in the future, but I think this will have to wait until the i2c driver is updated. LVGL V9 is used by default in the [LCD examples](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/lcd/rgb_panel/main/rgb_lcd_example_main.c) of ESP - IDF, and you can refer to them.

Y1hsiaochunnn, 2025-05-27:
> Hi [@Y1hsiaochunnn](https://github.com/Y1hsiaochunnn)
> 
> I'm truly sorry that this matter has been postponed continuously. I've been extremely busy recently. I promise that as soon as I have free time, the first thing I'll do is to transplant the new version of the i2c driver.

To tell the truth, I am busy too. I can understand you. I will try my best to help you if there is anything I need